espressif/esp-idf #12454: esp32c6 device is unreachable (bcn_timeout) if running in STA mode with WIFI_PS_MIN_MODEM

AxelLin, 2023-10-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1-678-g8c6114d0d2

### Espressif SoC revision.

esp32c6

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32c6 custom board

### Power Supply used.

USB

### What is the expected behavior?

The device should join AP, got IP and network should be ok.

### What is the actual behavior?

After update esp-idf v5.1 branch to 8c6114d0d2, I notice the esp32c6 in station mode is not reachable even it got IP address if esp_wifi_set_ps(WIFI_PS_MIN_MODEM) is set.
The device just repeat got IP and then got WIFI_EVENT_STA_BEACON_TIMEOUT, re-try join AP and go IP then got WIFI_EVENT_STA_BEACON_TIMEOUT.
It just repeat above behavior.
I try to ping the device after it got IP, but it is unreachable.

Below is the log with WIFI_PS_MIN_MODEM, many bcn_timeout.
```
I (2397) pp: pp rom version: 5b8dcfa
I (2397) net80211: net80211 rom version: 5b8dcfa
I (2417) wifi:wifi driver task: 40833744, prio:23, stack:6656, core=0
I (2417) wifi:wifi firmware version: 0bd9738
I (2417) wifi:wifi certification version: v7.0
I (2417) wifi:config NVS flash: enabled
I (2417) wifi:config nano formating: disabled
I (2427) wifi:mac_version:HAL_MAC_ESP32AX_761,ut_version:N
I (2427) wifi:Init data frame dynamic rx buffer num: 32
I (2437) wifi:Init static rx mgmt buffer num: 5
I (2437) wifi:Init management short buffer num: 32
I (2447) wifi:Init dynamic tx buffer num: 32
I (2447) wifi:Init static tx FG buffer num: 2
I (2447) wifi:Init static rx buffer size: 1700
I (2457) wifi:Init static rx buffer num: 10
I (2457) wifi:Init dynamic rx buffer num: 32
I (2467) wifi_init: rx ba win: 6
I (2467) wifi_init: tcpip mbox: 32
I (2467) wifi_init: udp mbox: 6
I (2477) wifi_init: tcp mbox: 6
I (2477) wifi_init: tcp tx win: 5744
I (2487) wifi_init: tcp rx win: 5744
I (2487) wifi_init: tcp mss: 1440
[... 295 lines cut ...]

I (19017) wifi:pm start, type:0, aid:0x4, trans-BSSID:18:a6:f7:ab:ca:66, BSSID[5]:0x66, mbssid(max-indicator:0, index:0), he:0
I (19027) wifi:set rx beacon pti, rx_bcn_pti: 10, bcn_timeout: 25000, mt_pti: 10, mt_time: 10000
I (19037) wifi:[ADDBA]TX addba request, tid:0, dialogtoken:4, bufsize:64, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (19047) wifi:[ADDBA]TX addba request, tid:7, dialogtoken:5, bufsize:64, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x20)
I (19057) wifi:[ADDBA]TX addba request, tid:5, dialogtoken:6, bufsize:64, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (19067) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (19087) wifi:[ADDBA]RX addba response, status:0, tid:0/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:64
I (19097) wifi:[ADDBA]RX addba response, status:0, tid:7/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:64
I (19097) wifi:[ADDBA]RX addba response, status:0, tid:5/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:64
W (19727) wifi:<ba-add>idx:0, ifx:0, tid:0, TAHI:0x10066ca, TALO:0xabf7a618, (ssn:7, win:64, cur_ssn:7), CONF:0xc0000005
I (20577) esp_netif_handlers: sta ip: 192.168.0.104, mask: 255.255.255.0, gw: 192.168.0.1
I (20617) wifi:Set ps type: 0, coexist: 0

I (20617) WIFI-SENSOR: WIFI POWER SAVE MODE: 0
I (20627) WIFI-SENSOR: local server is already started
I (20627) WIFI-SENSOR: internet service is already started

```

### Steps to reproduce.

I just make device join AP.

Note, I didn't have such issue with v5.1.1-588-gc1c843f5e2.

AxelLin, 2023-10-24:
1. I try the same application code on esp32c3, it works without this issue. So this is esp32c6 specific issue.
2. If I call esp_wifi_set_ps(WIFI_PS_NONE); before esp_wifi_start() it works.
   So this issue is related to WIFI_PS_MIN_MODEM on C6.
3. If I don't call esp_wifi_set_ps() at all.
   I still observe the "wifi:bcn_timeout,ap_probe_send_start".
   This make me wondering if the default setting is WIFI_PS_MIN_MODEM? (if without calling esp_wifi_set_ps() at all).

AxelLin, 2023-10-24:
Now I can reproduce the issue with examples/wifi/getting_started/station running on ESP32-C6-DevKitC-1.
The only thing you need to change is to set "CONFIG_COMPILER_OPTIMIZATION_SIZE=y"

ps. If you build with CONFIG_COMPILER_OPTIMIZATION_DEFAULT=y, you cannot see the symptom.

AxelLin, 2023-10-25:
@Espressif-liuuuu 
Can you take a look at this beacon timeout issue? https://github.com/espressif/esp-idf/issues/12454#issuecomment-1777395948
This is a big problem since the issue is in release branch now.

Espressif-liuuuu, 2023-10-25:
Hi @AxelLin, this is a known issue and being fixed, please hold on and it would be synced to github recently. 
please keep updated with @xuxiao111

AxelLin, 2023-10-27:
> Hi @AxelLin, this is a known issue and being fixed, please hold on and it would be synced to github recently. please keep updated with @xuxiao111

If it just happens on master, it's ok to hold on and wait.
But it happens on release branch, this really needs an urgent fix.

Sherry616, 2023-10-30:
Yes, we are currently working on an urgent fix to this issue.

closed 2023-11-01 by AxelLin