espressif/esp-idf #12455: Can't build with Enable Enterprise Option turned off

tkircher, 2023-10-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-3318-g8fc8f3f479

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Build without errors.

### What is the actual behavior?

The build fails as shown in the included build log.

### Steps to reproduce.

$ idf.py menuconfig

Component config > Wi-Fi > [ ] Enable enterprise option

### Build or installation Logs.

```plain
esp_sdk/esp-idf/components/wpa_supplicant/src/rsn_supp/wpa.c: In function 'wpa_supplicant_process_1_of_4':
esp_sdk/esp-idf/components/wpa_supplicant/src/rsn_supp/wpa.c:658:9: error: implicit declaration of function 'is_wpa2_enterprise_connection' [-Werror=implicit-function-declaration]
  658 |     if (is_wpa2_enterprise_connection()) {
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

### More Information.

With the Enterprise option selected the build completes successfully. If there are other build options which need to change in order to support this configuration then there is at least a missing dependency.

listout, 2023-10-27:
@tkircher Can you try this patch?

```diff
diff --git a/components/wpa_supplicant/src/rsn_supp/wpa.c b/components/wpa_supplicant/src/rsn_supp/wpa.c
index 06f564e084..54aec9ac42 100644
--- a/components/wpa_supplicant/src/rsn_supp/wpa.c
+++ b/components/wpa_supplicant/src/rsn_supp/wpa.c
@@ -655,6 +655,7 @@ void wpa_supplicant_process_1_of_4(struct wpa_sm *sm,
     u8 *kde, *kde_buf = NULL;
     size_t kde_len;
 
+#ifdef CONFIG_ESP_WIFI_ENTERPRISE_SUPPORT
     if (is_wpa2_enterprise_connection()) {
         wpa2_ent_eap_state_t state = eap_client_get_eap_state();
         if (state == WPA2_ENT_EAP_STATE_IN_PROGRESS) {
@@ -663,6 +664,7 @@ void wpa_supplicant_process_1_of_4(struct wpa_sm *sm,
             return;
         }
     }
+#endif
     wpa_sm_set_state(WPA_FIRST_HALF_4WAY_HANDSHAKE);
 
     wpa_printf(MSG_DEBUG, "WPA 1/4-Way Handshake");
```

@kapilkedawat What do you think?

kapilkedawat, 2023-11-16:
This is fixed in https://github.com/espressif/esp-idf/commit/53939de34ede197b12fd44ae8bda26627fc04867

closed 2023-11-16 by tkircher