espressif/esp-idf #12456: ESP32 Log Level Issue Halting execution of Firmware

veneno-529, 2023-10-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

I am working with Nimble. My firmware used to work fine with ESP - IDF v5.0 but after I upgraded to ESP - IDF 5.1.1 the code gets stuck while reading data from NVS. The code freezes, esp doesn't restart it just freezes. I have attached the log output after first boot up for both verbose and debug log levels.
Note that I have set the log level to verbose. 
Here is the catch, if I set the log level to Debug the firmware runs just fine, it doesn't halt midway. The firmware still works fine on ESP-IDF 5.0 with verbose log level.

[debug_log_output.txt](https://github.com/espressif/esp-idf/files/13160786/debug_log_output.txt)
[verbose_log_output.txt](https://github.com/espressif/esp-idf/files/13160787/verbose_log_output.txt)

rahult-github, 2023-10-27:
Hi @veneno-529 ,

I tried to compile IDF examples for both the mentioned debug levels and i don't observe the behaviour you mentioned. To have a better understanding of the issue can you :

1. Please share which chipset are you using. 
2. Do you observe this issue with any IDF examples ? 
3.  Is it possible to share your application code so that i can reproduce the issue OR any simulation modification in existing examples equivalent to steps beings done at your end that can be tried ?
4. Please share the sdkconfig being used at your end to confirm the settings being done. 

Also, how exactly is the log debug levels being modified. This is because ,when you mentioned that DEBUG level is set, we still don't see the HCI command logs for stack bring up during init time. Something like below should be observed in the console log that are missing in the log you provided

D (2504) NimBLE: registering characteristic 33333333-2222-2222-1111-111100000000 with def_handle=24 val_handle=25

D (2514) NimBLE: registering descriptor 34343434-2323-2323-1212-121201010101 with handle=27

D (2524) NimBLE: ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0003 len=0

D (2534) NimBLE: 0x03
D (2534) NimBLE: 0x0c
D (2534) NimBLE: 0x00
D (2544) NimBLE:

I (2544) uart: queue free spaces: 8
V (2544) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args
V (2554) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting flags 0xE
D (D (2564) intr_alloc: Connected src 34 to int 3 (cpu 1)
2544) NimBLE: ble_hs_hci_cmd_send: ogf=0x04 ocf=0x0001 len=0

D (2574) NimBLE: 0x01
D (2574) NimBLE: 0x10
D (2574) NimBLE: 0x00
D (2584) NimBLE:

I (2584) main_task: Returned from app_main()
D (2584) NimBLE: ble_hs_hci_cmd_send: ogf=0x04 ocf=0x0003 len=0

D (2594) NimBLE: 0x03
D (2594) NimBLE: 0x10
D (2594) NimBLE: 0x00
D (2604) NimBLE:

The usual way of setting Debug log level in Nimble is to set both the options below: 

Component Config  --> Log Output --> Default Log Verbosity --> Debug
Component Config --> Bluetooth --> NimBLE Options --> NimBLE Host log verbosity --> Debug Logs

veneno-529, 2023-11-08:
Hello,

I tried doing what you had suggested still I'm finding the same issue.
Everytime it halts at 
`D (2740) NIMBLE_NVS: peer_dev_rec restored 0 records
V (2750) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (2750) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (2760) intr_alloc: Connected src 49 to int 3 (cpu 1)`

Post which there is no output.
Unfortunately I can't share my complete application. 
But I'm using nimble bleprph example with RPA enabled and I'm storing bond keys in NVS.

I'm using ESP32-WROOM-32E 8MB flash. 
Note that I'm using Platformio v3.3.1 with ESP IDF v5.1.1.
PFA sdk file 
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/13292361/sdkconfig.zip)

Also, I did try running the bleprph example from githup repo of esp-idf it works fine without any issues but when I ran that application I didn't get any logs stating NIMBLE_NVS: peer_dev_rec restored 0 records.

veneno-529, 2024-01-11:
Hello,

Can someone help me with this issue. Sometimes the main task returns successfully from app_main() other times it doesn't.

rahult-github, 2024-04-18:
Hi @veneno-529 , 

Can you share any simplified application that can be used at my end to reproduce this issue ?

veneno-529, 2024-05-09:
Hi, 

I was able to resolve this issue. The only change I made is I called my functions and created tasks before running the nimble_port_freertos_init(host_task). Thus, closing the issue.