espressif/esp-idf #12457: "Secure Boot Check Fail " error in ESP32 when using Secure Boot and Flash Encryption.

hgwc, 2023-10-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
I was trying to use Secure Boot and Flash Encryption.
- using esp-idf 4.3
- Chip is ESP32-D0WD-V3 (revision 3)
I executed it with the following procedure:
1) modify menuconfig
Security features
`CONFIG_SECURE_SIGNED_ON_BOOT=y
CONFIG_SECURE_SIGNED_ON_BOOT=y
CONFIG_SECURE_SIGNED_ON_UPDATE=y
CONFIG_SECURE_SIGNED_APPS=y
CONFIG_SECURE_SIGNED_APPS_ECDSA_SCHEME=y
CONFIG_SECURE_BOOT=y
CONFIG_SECURE_BOOT_V1_ENABLED=y
CONFIG_SECURE_BOOTLOADER_ONE_TIME_FLASH is not set
CONFIG_SECURE_BOOTLOADER_REFLASHABLE=y
CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES=y
CONFIG_SECURE_BOOT_SIGNING_KEY="secure_boot_signing_key.pem"
CONFIG_SECURE_BOOTLOADER_KEY_ENCODING_256BIT=y
CONFIG_SECURE_BOOTLOADER_KEY_ENCODING_192BIT is not set
CONFIG_SECURE_BOOT_INSECURE is not set
CONFIG_SECURE_FLASH_ENC_ENABLED=y
CONFIG_SECURE_FLASH_ENCRYPTION_MODE_DEVELOPMENT=y
CONFIG_SECURE_FLASH_ENCRYPTION_MODE_RELEASE is not set`

Partition Table
`CONFIG_PARTITION_TABLE_SINGLE_APP is not set
CONFIG_PARTITION_TABLE_TWO_OTA is not set
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_SINGLE_APP_ENCRYPTED_NVS is not set
CONFIG_PARTITION_TABLE_TWO_OTA_ENCRYPTED_NVS is not set
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions.2MB.csv"
CONFIG_PARTITION_TABLE_FILENAME="partitions.2MB.csv"
CONFIG_PARTITION_TABLE_OFFSET=0x10000
CONFIG_PARTITION_TABLE_MD5=y`

2) python espsecure.py generate_signing_key secure_boot_signing_key.pem
3) python idf.py bootloader 
4) python espefuse.py --port /dev/ttyUSB0 burn_key secure_boot_v1 secure-bootloader-key-256.bin
5) python espefuse.py --port /dev/ttyUSB0 burn_efuse ABS_DONE_0
6) python esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x00 bootloader-reflash-digest.bin
7) python esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 bootloader.bin
8) python idf.py flash monitor

partition table:
`*******************************************************************************
ESP-IDF Partition Table

Name, Type, SubType, Offset, Size, Flags

stnv,data,nvs,0x11000,16K,

nvs,data,nvs,0x15000,16K,

otadata,data,ota,0x19000,8K,

phy_init,data,phy,0x1b000,4K,

nvs_key,data,nvs_keys,0x1c000,4K,encrypted

ota_0,app,ota_0,0x20000,1M,

ota_1,app,ota_1,0x120000,1M,
*******************************************************************************`

efuses:
`KEY_STATUS (BLOCK0):                     Usage of efuse block 3 (reserved)                  = False R/W (0b0)

Identity fuses:
MAC (BLOCK0):                            Factory MAC Address                               
   = 48:e7:29:a2:51:78 (CRC 0x8c OK) R/W 
MAC_CRC (BLOCK0):                        CRC8 for factory MAC address                       = 140 R/W (0x8c)
CHIP_VER_REV1 (BLOCK0):                  Silicon Revision 1                                 = True R/W (0b1)
CHIP_VER_REV2 (BLOCK0):                  Silicon Revision 2                                 = True R/W (0b1)
CHIP_VERSION (BLOCK0):                   Reserved for future chip versions                  = 2 R/W (0b10)
CHIP_PACKAGE (BLOCK0):                   Chip package identifier                            = 1 R/W (0b001)
MAC_VERSION (BLOCK3):                    Version of the MAC field                           = 0 R/W (0x00)

Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                Flash encryption mode counter                      = 1 R/W (0b0000001)
UART_DOWNLOAD_DIS (BLOCK0):              Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):             Flash encryption config (key tweak bits)           = 15 R/W (0xf)
CONSOLE_DEBUG_DISABLE (BLOCK0):          Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                     Secure boot V1 is enabled for bootloader image     = True R/W (0b1)
ABS_DONE_1 (BLOCK0):                     Secure boot V2 is enabled for bootloader image     = False R/W (0b0)
JTAG_DISABLE (BLOCK0):                   Disable JTAG                                       = True R/W (0b1)
DISABLE_DL_ENCRYPT (BLOCK0):             Disable flash encryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_DECRYPT (BLOCK0):             Disable flash decryption in UART bootloader        = True R/W (0b1)
DISABLE_DL_CACHE (BLOCK0):               Disable flash cache in UART bootloader             = True R/W (0b1)
BLOCK1 (BLOCK1):                         Flash encryption key                              
   = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/- 
BLOCK2 (BLOCK2):                         Secure boot key                                   
   = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/- 
BLOCK3 (BLOCK3):                         Variable Block 3                                  
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).
`

execution result:
[image]

AdityaHPatwardhan, 2023-10-25:
@hgwc I think there is a small issue in the following commands

>
python esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x00 bootloader-reflash-digest.bin
python esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 bootloader.bin

For both of the these commands I think you need to add the --encrypt option that would encrypt the bootloader and the digest before adding it to the flash. I have added the updated commands below

5) python esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x00 bootloader-reflash-digest.bin --encrypt
6) python esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 bootloader.bin --encrypt

Can you please try that and check if this fixes your issue?

Thanks,
Aditya

AdityaHPatwardhan, 2023-10-25:
Additionally,
As you already have the ESP32-ECO3 chip
then I would recommend you to use [Secure Boot V2](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v2.html) instead of the older Secure Boot V1.

You  may follow [these steps](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/host-based-security-workflows.html#enable-secure-boot-v2-externally)
to enable secure boot v2 on the device.

hgwc, 2023-10-26:
Thanks for your answer.

I performed the above process on a new esp32.
However, '5 python espefuse.py --port /dev/ttyUSB0 burn_efuse ABS_DONE_0' was excluded.
I did 2,3,4,6,7,8
I cannot determine whether flash encryption and secure boot were successful.
I am attaching the first boot log and efuse, please review.
Best Regards,

file:///home/hgwc/efuse_summary.txt
file:///home/hgwc/first_booting_log.txt

AdityaHPatwardhan, 2023-10-26:
@hgwc 
I dont think the files have been properly attached. I only see the text name

Also, sorry for the confusion earlier

`5. python espefuse.py --port /dev/ttyUSB0 burn_efuse ABS_DONE_0`

This step is indeed necessary.

But steps 
>6 . python esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x00 bootloader-reflash-digest.bin
7 . python esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 bootloader.bin

Can be clubber together. ( I have edited my earlier comment to reflect the correct index)

1) why have you chosen to use Secure Boot V1 on a ECO3 device, is there any particular reason?
2) The detailed steps to enable Secure Boot V2 are documented clearly at this [link](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/host-based-security-workflows.html#enable-secure-boot-v2-externally), I would recommend going once through the steps to enable flash encryption and secure boot together for more details.

hgwc, 2023-10-26:
Sorry
I have attached 2 files.

There is no particular reason to use secure boot v1. 
v2 is currently in progress.

[efuse_summary.txt](https://github.com/espressif/esp-idf/files/13173442/efuse_summary.txt)
[first_booting_log.txt](https://github.com/espressif/esp-idf/files/13173444/first_booting_log.txt)

AdityaHPatwardhan, 2023-10-26:
This looks like the correct log for secure boot V1.

hgwc, 2023-10-26:
Is flash encryption performed normally?

AdityaHPatwardhan, 2023-10-31:
Yes, the flash encryption logs look good as well.
You can find more details at [this link](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html)
I am closing this issue, since the main problem seems to be resolved. Feel free to re-open if required.
Thanks,
Aditya

closed 2023-10-31 by AdityaHPatwardhan