espressif/esp-idf #12458: BLE protocol uses 2.4G function to communicate with other brands on 2.4G

zenghi, 2023-10-25:
### Is your feature request related to a problem?

The RF PHY of CH582 is based on BLE transceiver, which means that the modulation and demodulation parameters of the transceiver include frequency, deviation, symbol_ Rate, packet_ Handlers (preparable, syncword, length, crc, whitening) are all compliant with the underlying specifications of Bluetooth.

This means that the other party conforms to the packet format of ble and can directly call the underlying layer, theoretically communicating with each other.
[image]
Leading packet 1 byte,
Access address 4 bytes,
PDU contains 1 byte header, 1 byte length, and 0-251 bytes of data
CRC checks 3 bytes.

### Additional context.

Due to the high power consumption of ESP and the need to replace it with external chips for ultra-low power consumption, this leads to difficulty in chip communication

zhp0406, 2023-11-30:
@zenghi 
what should be done on the ESP32 side？

zenghi, 2023-12-01:
> @zenghi what should be done on the ESP32 side？
HI，The image is the concept of the other party's data packet format, and we need to control the content, channel, and rate of the ble packet. It can also be used as a receiving concept, purely for RHY 2.4G transmission and reception functions.
[image]

zenghi, 2023-12-14:
> maybe,you can refer to the example below to implement your functionality: https://github.com/espressif/esp-idf/tree/master/examples/phy/cert_test

HI，This example only has commands, and currently I don't know how to send PDU data esp_ Ble_ Tx - p 1- n 7- l 37- t 2- s 0x71764129- r 0- m 0 Currently, the 1M protocol sends commands like this, but I don't know how to send specific data, and 582 cannot receive messages. 582 channel has been set with 7 addresses, and the same is true for all 71764129 addresses

zhp0406, 2023-12-15:
@zenghi 
I'm not sure what 'specific data' means. In cert_test example, only these [payload types](https://gitlab.espressif.cn:6688/espressif/esp-idf/-/blob/master/components/esp_phy/include/esp_phy_cert_test.h/#L57) are supported.

zenghi, 2023-12-16:
@zhp0406 
At present, we need to send 123 data through this PHY interface, but this Cert_ The test example only sees esp_ble_ tx-p 1-n 7-l 37-t2-s 0x71764129-r 0-m 0 How to send a string like 123 for interfaces like this? Currently, esp_ ble_ rx How does RX receive data.
Is there an example of sending and receiving through this Phyble interface

zhp0406, 2023-12-18:
@zenghi 
We do not support custom data.

zenghi, 2023-12-18:
@zhp0406  Can you help create a custom sending and receiving function through the BLE PHY interface。

zenghi, 2023-12-28:
@zhp0406  
[image]
Hi , we can receive the other party's phy ble data. Can you help us create a callback function? When the data is received, we can call this callback function in real time and pass the received data to this callback function.

zhp0406, 2024-01-02:
@zenghi 
In esp-idf/components/esp_phy/src/phy_callback.c, we do have a callback funciton that can stop RX after some time and return received info. The parameter g_rf_cmdstop and func esp_phy_cmdstop_callback in phy_callback.c should be used.

happytm, 2024-01-31:
@zenghi Did you figure out how to achieve communication between CH582 and ESP using 125k phy ?

Can you share code you are using for CH582 please?

Thanks.

Alvin1Zhang, 2024-03-11:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates.

closed 2024-03-11 by Alvin1Zhang

zenghi, 2024-03-13:
@happytm I apologize for not being able to implement this feature