espressif/esp-idf #12463: SPI flash driving not working for Winbond NAND flash W25N01GVZEIG

DDANPEAL, 2023-10-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**Env:** ESP32 chip and idf-esp 5.1.1

We're using NAND W25N01GVZEIG with custom driver to store vehicle data in it.
For backend requirements, data needs to be sent in a file format ( currently sent in binary packets).

We want to setup a file system in this flash (128MB) like FAT or SPIFFS to create/read/write files.

I tried to use the available spi flash APIs available in esp-idf 5.1.1

```
	const spi_bus_config_t bus_config = {
		.mosi_io_num = CONFIG_SPI_MOSI,
		.miso_io_num = CONFIG_SPI_MISO,
		.sclk_io_num = CONFIG_SPI_CLK,
		.quadhd_io_num = -1,
		.quadwp_io_num = -1,
	};

	const esp_flash_spi_device_config_t device_config = {
		.host_id = HOST_ID,
		.cs_id = 0,
		.cs_io_num = CONFIG_SPI_CS,
		.io_mode = SPI_FLASH_DIO,
		.freq_mhz = ESP_FLASH_40MHZ,
	};

	// Initialize the SPI bus
	ESP_LOGI(TAG, "DMA CHANNEL: %d", SPI_DMA_CHAN);
	ESP_ERROR_CHECK(spi_bus_initialize(HOST_ID, &bus_config, SPI_DMA_CHAN));

	// Add device to the SPI bus
	esp_flash_t* ext_flash;
	ESP_ERROR_CHECK(spi_bus_add_flash_device(&ext_flash, &device_config));

	// Probe the Flash chip and initialize it
	esp_err_t err = esp_flash_init(ext_flash);
	if (err != ESP_OK) {
		ESP_LOGE(TAG, "Failed to initialize external Flash: %s (0x%x)", esp_err_to_name(err), err);
		return NULL;
	}
```

But the ID read command `memspi_host_read_id_hs` is not compatible with this flash.
I added the following changes

```
diff --git a/components/spi_flash/memspi_host_driver.c b/components/spi_flash/memspi_host_driver.c
index 0668c9a65b..87b127bd2e 100644
--- a/components/spi_flash/memspi_host_driver.c
+++ b/components/spi_flash/memspi_host_driver.c
@@ -89,13 +89,20 @@ esp_err_t memspi_host_read_id_hs(spi_flash_host_inst_t *host, uint32_t *id)
     uint32_t id_buf = 0;
     spi_flash_trans_t t = {
         .command = CMD_RDID,
-        .miso_len = 3,
+        .miso_len = 4,
         .miso_data = ((uint8_t*) &id_buf),
     };
     host->driver->common_command(host, &t);
-
-    uint32_t raw_flash_id = id_buf;
-    ESP_EARLY_LOGV(TAG, "raw_chip_id: %X\n", raw_flash_id);
+    uint32_t raw_flash_id = 0;
+    if(id_buf == 0x21AAEFFF)
+    {
+        raw_flash_id = id_buf>>8;
+    }
+    else
+    {
+        raw_flash_id = id_buf;
+    }
+    ESP_EARLY_LOGE(TAG, "raw_chip_id: %X\n", raw_flash_id);
     if (raw_flash_id == 0xFFFFFF || raw_flash_id == 0) {
         ESP_EARLY_LOGE(TAG, "no response\n");
         return ESP_ERR_FLASH_NO_RESPONSE;
@@ -104,7 +111,7 @@ esp_err_t memspi_host_read_id_hs(spi_flash_host_inst_t *host, uint32_t *id)
     uint8_t mfg_id = raw_flash_id & 0xFF;
     uint16_t flash_id = (raw_flash_id >> 16) | (raw_flash_id & 0xFF00);
     *id = ((uint32_t)mfg_id << 16) | flash_id;
-    ESP_EARLY_LOGV(TAG, "chip_id: %X\n", *id);
+    ESP_EARLY_LOGE(TAG, "chip_id: %X\n", *id);
     return ESP_OK;
 }

```
when trying to mounting the FAT system it starts

```
E (1044) wl_partition: erase_range - start_address=0x07efd000, size=0x00081000, result=0x00000107
E (1044) wl_flash: initSections(286): result = 0x00000107
E (1044) wl_flash: init(179): result = 0x00000107
E (1054) wl_ext_safe: init(65): result = 0x00000107
E (1064) wear_levelling: wl_mount: init instance=0x00000000, result=0x107
E (1064) vfs_fat_spiflash: esp_vfs_fat_spiflash_mount_rw_wl(123): failed to mount wear levelling layer. ret = 0
E (1074) FTP: Failed to mount FATFS (ESP_ERR_TIMEOUT)
```

I tried to change the block size to 1920 but it didn't help.

RathiSonika, 2023-10-26:
Hello @DDANPEAL,  As of the moment, we do not provide support for NAND flash. However, development efforts are currently in progress, and this feature will be available in the near future. If you're interested, you can go through the [pull request (PR)](https://github.com/espressif/esp-idf/pull/9422) that introduces NAND flash support. Thank you!

chaabanemalki, 2023-10-30:
Thank you for the update. I'll follow this topic in the PR

DDANPEAL, 2024-03-11:
Now we have a flash driver working an available.

closed 2024-03-11 by DDANPEAL