espressif/esp-idf #12466: Guru Meditation Error when calling 'adc_oneshot_read' function

marco711, 2023-10-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**General issue report**

**IDF version:**
ESP-IDF v5.1-dev-3710-gacac972f70-dirty 2nd stage bootloader

**SDK config file:**
Attached.
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/13175457/sdkconfig.zip)

**Issue:**
Hello,
I'm using an ESP32-S3 on a custom board where I'm using the following task for reading the ADC value every 100 ms:

```
void adc_task(void *pvParameter)
{
	//-------------ADC1 Init---------------//
	ESP_ERROR_CHECK(adc_oneshot_new_unit(&init_config1, &adc1_handle));

	//-------------ADC1 Config---------------//
	ESP_ERROR_CHECK(adc_oneshot_config_channel(adc1_handle, ADC1_CHAN3, &config));

	//-------------ADC1 Calibration Init---------------//
	adc_cali_handle_t adc1_cali_handle = NULL;
	bool do_calibration1 = example_adc_calibration_init(ADC_UNIT_1, EXAMPLE_ADC_ATTEN, &adc1_cali_handle);

	while(1)
	{
		ESP_ERROR_CHECK(adc_oneshot_read(adc1_handle, ADC1_CHAN3, &adc_raw));

		if (do_calibration1) {
			ESP_ERROR_CHECK(adc_cali_raw_to_voltage(adc1_cali_handle, adc_raw, &adc_voltage));
		}
				
		vTaskDelay(pdMS_TO_TICKS(100));
	}
}
```

The problem I have is that when I move the board abruptly I get a "Software reset due to exception/panic" and the board resets. In some cases in the terminal I get the following result just before the reset:

```
[2023-10-26_10:47:46:46.000076] OK Msg ID, evt_id: 156 152
[2023-10-26_10:47:46:46.799916] OK Msg ID, evt_id: 158 154
[2023-10-26_10:47:46:46.960088] Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[2023-10-26_10:47:46:46.960243] 
[2023-10-26_10:47:46:46.960318] Core  0 register dump:
[2023-10-26_10:47:46:46.960381] PC      : 0x4207727d  PS      : 0x00060430  A0      : 0x8200cad7  A1      : 0x3fcd8fe0  
0x4207727d: adc_oneshot_read at /home/ms/esp/esp-idf/components/esp_adc/adc_oneshot.c:165 (discriminator 5)

[2023-10-26_10:47:46:46.981831] A2      : 0x122b6b8b  A3      : 0x00000003  A4      : 0x3fca03dc  A5      : 0x00000009  
[2023-10-26_10:47:46:46.981934] A6      : 0x00000004  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000  
[2023-10-26_10:47:46:46.992491] A10     : 0x00000000  A11     : 0x3fc9f4d8  A12     : 0x3fc9f778  A13     : 0x000030e3  
[2023-10-26_10:47:46:46.992724] A14     : 0xb33fffff  A15     : 0xb33fffff  SAR     : 0x00000012  EXCCAUSE: 0x0000001c  
[2023-10-26_10:47:47:47.008278] EXCVADDR: 0x122b6bef  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  
0x400570e8: memset in ROM

0x400570f3: memset in ROM

[2023-10-26_10:47:47:47.017928] 
[2023-10-26_10:47:47:47.018008] 
[2023-10-26_10:47:47:47.018051] Backtrace: 0x4207727a:0x3fcd8fe0 0x4200cad4:0x3fcd9010 0x40382ecd:0x3fcd9040
0x4207727a: adc_oneshot_read at /home/ms/esp/esp-idf/components/esp_adc/adc_oneshot.c:165 (discriminator 2)

0x4200cad4: adc_task at /home/ms/esp/test_blufi_iot/mqtt-tcp/main/app_main.c:2336 (discriminator 2)

0x40382ecd: vPortTaskWrapper at /home/ms/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

According to the documentation as the `EXCVADDR: 0x122b6bef` is garbage value there might be a pointer not initialized or corrupted. I have checked the line 165 of the `adc_oneshot.c` file and there is the following line:

`ESP_RETURN_ON_FALSE(chan < SOC_ADC_CHANNEL_NUM(handle->unit_id), ESP_ERR_INVALID_ARG, TAG, "invalid channel");`

I don't understand which could be the pointer not initialized or corrupted.

In some other cases I get the following output just before the board resets (after abruptly moving it):

```
[2023-10-26_11:42:15:15.801233] OK Msg ID, evt_id: 34 178
[2023-10-26_11:42:15:15.928909] 
[2023-10-26_11:42:15:15.945951] assert failed: adc_oneshot_ll_set_output_bits /IDF/components/hal/esp32s3/include/hal/adc_ll.h:847 (bits == ADC_BITWIDTH_12 || bits == ADC_BITWIDTH_DEFAULT)
[2023-10-26_11:42:15:15.946230] 
[2023-10-26_11:42:15:15.946373] 
[2023-10-26_11:42:15:15.961916] Backtrace: 0x40375a8e:0x3fcd88d0 0x403804cd:0x3fcd88f0 0x40386fd5:0x3fcd8910 0x420b98ff:0x3fcd8a30 0x420b99b5:0x3fcd8a50 0x42077299:0x3fcd8a80 0x4200cad4:0x3fcd8ab0 0x40382ecd:0x3fcd8ae0
0x40375a8e: panic_abort at /home/ms/esp/esp-idf/components/esp_system/panic.c:451

0x403804cd: esp_system_abort at /home/ms/esp/esp-idf/components/esp_system/port/esp_system_chip.c:78

0x40386fd5: __assert_func at /home/ms/esp/esp-idf/components/newlib/assert.c:81

0x420b98ff: adc_hal_onetime_start at adc_oneshot_hal.c:?

0x420b99b5: adc_oneshot_ll_output_invert at /home/ms/esp/esp-idf/components/hal/esp32s3/include/hal/adc_ll.h:986
 (inlined by) adc_oneshot_hal_setup at /home/ms/esp/esp-idf/components/hal/adc_oneshot_hal.c:70

0x42077299: adc_oneshot_read at /home/ms/esp/esp-idf/components/esp_adc/adc_oneshot.c:165 (discriminator 5)

0x4200cad4: adc_task at /home/ms/esp/test_blufi_iot/mqtt-tcp/main/app_main.c:2336 (discriminator 2)

0x40382ecd: vPortTaskWrapper at /home/ms/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

Is there something I'm not correctly doing with the ADC?

Bruce297, 2023-10-31:
It could not be reproduce on my board, could you provide more information about "move the board abruptly", such as: what else peripheral do you used?
Also, you can enable coredump to save information to flash when a fatal error occurs. Look [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/core_dump.html)

marco711, 2023-10-31:
I'm using the I2C and Wi-Fi. The Bluetooth is also configured but is not being in use at the moment of the board reset. Additionally, I'm using the MQTT which is permanently transmitting data read from the I2C and the ADC.

AxelLin, 2023-12-16:
> According to the documentation as the `EXCVADDR: 0x122b6bef` is garbage value there might be a pointer not initialized or corrupted. I have checked the line 165 of the `adc_oneshot.c` file and there [...]
What is the value of ADC1_CHAN3 in your code ?

> [2023-10-26_11:42:15:15.945951] assert failed: adc_oneshot_ll_set_output_bits /IDF/components/hal/esp32s3/include/hal/adc_ll.h:847 (bits == ADC_BITWIDTH_12 || bits == ADC_BITWIDTH_DEFAULT)

Do you check what is the value of 'bits' here?