espressif/esp-idf #12469: ESP32-C6 UART1 wake-up modes

msia3dev, 2023-10-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

First of all, sorry for posting the same things here and on the forum as well. I didn't receive any feedback on a forum so I may have more luck getting answers here.

While researching how I can wake up HP core from auto light sleep using UART1, I stumbled upon a section in the Technical Reference Manual, related to different UART wake-up modes -- "26.4.8 Wake­up, page 637". Of 4 modes, the only one that seems to be implemented in the ESP-IDF is mode 0, allowing [wake-up if a certain number of RX signal edges are detected](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/peripherals/uart.html#_CPPv425uart_set_wakeup_threshold11uart_port_ti).

I've tried changing the mode to the other 3 by adjusting the register value, but none of them seem to work. I can confirm that mode 0 does work, enabled either from ESP-IDF or by manually setting the correct register value. I've also confirmed that the register value is not being changed during light sleep entry/exit. Is there anything that I'm missing here?

For testing purposes, I'm using ESP32-C6-DevKitC-1 v1.2. UART1 is connected to GPIO4 (RX) and GPIO5 (TX) and uses XTAL as a clk source. "Power down Digital Peripheral in light sleep" and "Disable all GPIO when chip at sleep" are both turned off. This was tested on the master branch as well as the v5.1.1 release.
The reason for even trying to change modes is to use the ability of "mode 1" which states that it keeps the UART core working and is able to fill RX FIFO. With mode 0, the first packet gets cut or there is some garbage received which makes the system unusable. Before resorting to LP core I wanted to see if I can utilize mode 1 or 2 to be able to wake HP core but also get all data that was sent to UART.

Also, based on the technical reference manual, wake-up should be possible from LP UART as well. I've tried using it from HP Core but `esp_sleep_enable_uart_wakeup` seems to handle only UART 1 and 0 as wakeup triggers, and since whole section about Power Management Unit is missing from manual, it's a blackbox for configuring.

To simplify, a simple answer to whether it's even possible to have HP core into a light sleep but any of 3 UARTS active to collect initial data and as well wake up HP core so we get all the data, will be just enough. Having to wake up on UART as is now does not seem to be a good solution for us because we can't afford to lose data when waking up.

Let me know if there is any additional info I can provide.

Also, do you have a rough estimate of when we can expect an updated Technical Reference Manual with a "Low-Power Management" section?

Thanks in advance!

hbler99, 2025-03-17:
Hello! Apologies for the delayed response.
The master branch of the IDF has recently added support for wakeup modes 1, 2, and 3 of LP UART. If you wish to utilize these different LP UART wakeup modes, kindly refer to the following source files: [uart_wakeup.c](https://github.com/espressif/esp-idf/blob/master/components/esp_driver_uart/src/uart_wakeup.c) and [test_lp_core_uart_wakeup.c](https://github.com/espressif/esp-idf/blob/master/components/ulp/test_apps/lp_core/lp_core_basic_tests/main/test_lp_core_uart_wakeup.c).
Furthermore, support for all wakeup modes of the HP UARTs will be available in the near future.

Alvin1Zhang, 2025-07-02:
Thanks for reporting, fix is available at https://github.com/espressif/esp-idf/commit/d115a3daac20343a1ceba58ed27e9c5ee615c91a. Feel free to reopen.

closed 2025-07-02 by Alvin1Zhang