espressif/esp-idf #12470: Development of an official DNP3 library for Espressif SoCs

shukrhu, 2023-10-27:
### Describe the solution you'd like.

Develop the DNP3 library to be used in the Espressif SoCs based in the Standard as the same way as it was developed the Lib for Modbus

### Describe alternatives you've considered.

Instead to write down the protocol in code, by using a library which could be easily and widdely used for Utilities Automation integration with supervisories (SCADA) 

### Additional context.

DNP3 protocol is very widely used inside the utilities as it has a more powerfull concept than Modbus, and aloows more precise and accurate data exchange between automation devices.

david-cermak, 2023-11-28:
@shukrhu We will not support this library in ESP-IDF, as this is not a widely used IoT protocol. Besides esp_modbus, there might be some future plans on other industrial protocol support, but not DNP3.

We'd leave this open for a bit, to see if there's any other comments.

david-cermak, 2024-03-11:
Closing as there's no recent activity.
Assessment: will not support DNP3 in IDF.

closed 2024-03-11 by david-cermak