espressif/esp-idf #12472: Ninja does not support ARM64 (NVIDIA Jetson Orin Nano)

AlfonGio, 2023-10-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1-dirty

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The program build and flash successfully from my VSCode to ESP32-S3 board via USB Bridge

### What is the actual behavior?

[image]

Different architecture, I have looked at tools.json ninja is only declared linux_amd64 while the arm64 only for macos.
Is there a way that I can install it by myself? or what should I do?

### Steps to reproduce.

1. Step 1: Create new project using hello world template in ESP-IDF extension for VSCode
2. Step 2: Configure the device which is ESP32-S3 and serial connection /dev/ttyACM0
3. Step 3: Use the Build, Flash, and Monitor command pallet
...

### Build or installation Logs.

```plain
CMake Error at /home/orin/esp/esp-idf/tools/cmake/project.cmake:448 (__project):
  Running

   '/home/orin/.espressif/tools/ninja/1.10.2/ninja' '--version'

  failed with:

   /home/orin/.espressif/tools/ninja/1.10.2/ninja: 1: ELF: not found

  /home/orin/.espressif/tools/ninja/1.10.2/ninja: 2: Syntax error: word
  unexpected (expecting ")")
```

### More Information.

I tried on ARM64 Linux (Ubuntu 22.04) using VSCode extension and it works.

dobairoland, 2023-10-30:
Hi @AlfonGio.

Could you please make an experiment? Could you try to run the executable from `ninja-linux.zip` downloadable from https://github.com/ninja-build/ninja/releases/tag/v1.11.1? Please share the output if you encounter any issue.

Note: After extracting the executable from the above zip probably you will have to set the executable bit file permission for it.

dobairoland, 2023-10-30:
One more suggestion I've got from @fhrbata. (Thanks!)

If you follow the guide and install the packages as suggested in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html#for-linux-users then `ninja` should already been installed on your computer (for the right architecture).

But this doesn't change the fact that we still could have add it to our `tools.json`.

mfialaf, 2023-10-31:
> Hi @AlfonGio.
>
> Could you please make an experiment? Could you try to run the executable from `ninja-linux.zip` downloadable from https://github.com/ninja-build/ninja/releases/tag/v1.11.1? Please [...]

Just a note for the experiment
Trying the ninja from the URL working with esp-idf would require esp-idf version >= 5.2. 
If the ninja would be executed standalone, like `ninja --version`, then it's independent to esp-idf.

mfialaf, 2023-11-02:
Hi, @AlfonGio. The official release on GitHub by ninja-build, does not seem to support arm64 architecture for linux. But you can download it from debian package here: https://packages.debian.org/bullseye/ninja-build

dobairoland, 2023-11-02:
> I tried on ARM64 Linux (Ubuntu 22.04)

Ubuntu has it as well: https://packages.ubuntu.com/jammy/ninja-build

So it is enough to follow the official getting started guide and install the prerequisite packages.