espressif/esp-idf #12473: E (12345) wifi:NAN WiFi stop

kriegste, 2023-10-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP32
IDF 5.1.1

After esp_restart(), just before restarting,

> E (12345) wifi:NAN WiFi stop

appears in the log. What does this error mean? How can I prevent it? It did not appear in IDF 4.4.x.

mscott370, 2023-11-04:
Hi @kriegste 
I'm also seeing the error message when i use Arduino-ESP32 v3.0.0 (based on ESP-IDF 5.1).
   "E (23862) wifi:NAN WiFi stop"

- The error message is logged when i call WiFi.disconnect() and did not appear using Arduino-ESP32 v2.0.14 (ESP-IDF v4.4.6). The only way i can avoid the message is to suppress wifi logging:
  esp_log_level_set("wifi", ESP_LOG_NONE) 
- not really a desirable solution as i normally would want to see wifi errors.
-  I am not using any WiFi-Aware (NAN) functions, so should not be seeing the NAN error message, and if i try to avoid the error by terminating NAN before calling WiFi.disconnect() with:
  esp_err_t ret = esp_wifi_nan_stop(); 
... that just produces  a "NAN isn't started" message. So if NAN isn't started, why do we see this new message?

kriegste, 2023-11-04:
Oh, I was not aware NAN stands for WiFi-Aware... I checked all my floating point code since nan normally is a floating point issue... That is an unfortunate name for an error!

Anyway, I do not use WiFi-Aware either. It is disabled in sdkconfig (which seems to be default).

ShyamalKhachane, 2024-04-08:
Hi @kriegste @mscott370, this is not an error, it's supposed to be an informative print. We will suppress it in upcoming commits.

mscott370, 2024-08-30:
Hi @ShyamalKhachane - has the commit for this issue occurred yet?

bjuberchaub, 2025-02-21:
@ShyamalKhachane It looks like an error due to the log level:

[image]

closed 2025-03-24 by espressif-bot