espressif/esp-idf #12474: TWAI giving broken messages via twai_receive()

giga993, 2023-10-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Espressif SoC revision.

ESP32 Wrover E ECO3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

`twai_receive()` does only deliver correct messages to application.

### What is the actual behavior?

`twai_receive()` does very rarely deliver a corrupt messages to application.

We encountered this problem by a once happened random undefined behavior and couldn't localize it at first. After narrowing it down by extensive error logging we found CAN-messages in our log, which never have been sent by anyone (the structure of the data in the message does not match the ID). 

Further observation by a PCAN-Adapter in listen only mode confirmed, that all messages which have been sent over the bus had the correct format (also no BUS-Errors).
 
=> Either the message got corrupted on its way from the transceiver to the application or this message was never received, and an never existent message got randomly delivered to the application.

The occurrence just by observation is very very rare. With a system running 24/7 we can not reproduce the issue within days.

Though the pattern of the issue is always the same: ID and DLC is correct, and the data is almost correct, it is just shifted by some bytes. E.g. the byte 0 of the correct sent message gets byte 2-4 and so on of the received message by the application (we never saw it shifted by more than 4 bytes). So a message with ID 0x10000 dlc: 8 originally data: 0xAB 0xCD 0xEF 0x01 0x23 0x45 0x67 0x89 gets: `Message ID: 0x10000 (65536), len: 8, data: 45 67 89 AB CD EF 01 23`

### Steps to reproduce.

That's the hard part. 

Though, we could reproduce the problem by inducing CAN-errors by cutting the data lines and a modified ESP IDF TWAI Example.

Steps to Reproduce:
1) Flash [gist](https://gist.github.com/giga993/b931b858e09ed1860c56c56ca389812a) to an ESP32 Wrover E ECO3 16MB flash 8MB PSRAM with CAN-Transceiver
2) Connect an PCAN Transceiver to the ESP32 CAN-Bus (point to point, no real bus) with following config:
[image]
3) Open terminal and start ESP-IDF monitor of the ESP32
4) Simulate a bad CAN-Bus connection by plugging and un-plugging the CAN-Bus connector between PCAN-Adapter and ESP32 transceiver
5) See many many `TWAI_ALERT_BUS_ERROR` and `TWAI_ALERT_ERR_PASS` or `TWAI_ALERT_PERIPH_RESET` errors
6) Eventually spot one rare corrupted messages `Message ID: 0x10000 (65536), len: 8, data: 45 67 89 AB CD EF 01 23, msg tickcount: 686471ms` between many many errors.

It takes me around 2-15mins of fiddling with the data lines to get this error. 
The PCAN on the other side does never receive a corrupted message in the application layer (I guess they are all filtered out by the CAN-Errors/CRC-Check).

[... 1378 lines cut ...]
W (1068553) TWAI Alert and Recovery: TWAI State: 1
        msgs_to_tx: 1
        msgs_to_rx: 0
        tx_error_counter: 135
        rx_error_counter: 0
        tx_failed_count: 85
        rx_missed_count: 0
        rx_overrun_count: 0
        arb_lost_count: 21
        bus_error_count: 44366
E (1068565) TWAI Alert and Recovery: !!! ALERT !!!: ERR_PASS (1000/1000)
W (1068762) TWAI Alert and Recovery: TWAI State: 1
        msgs_to_tx: 1
        msgs_to_rx: 0
        tx_error_counter: 130
        rx_error_counter: 0
        tx_failed_count: 85
        rx_missed_count: 0
        rx_overrun_count: 0
        arb_lost_count: 21
        bus_error_count: 44415
E (1068774) TWAI Alert and Recovery: !!! ALERT !!!: ERR_PASS (1000/1000)
</details>

Note: Around 50% of the time I could reproduce the error, the corrupted message is really really close to a `TWAI_ALERT_BUS_ERROR`, like within 1ms between the alert and the corrupted message being delivered.

giga993, 2023-10-27:
Addition: 
I saw #10004 with similar symptoms but on my side I can not see the `TWAI_ALERT_ARB_LOST` as a cause.

Dazza0, 2023-10-30:
@giga993 Could you double check that all of the TWAI errata workarounds are enabled via menuconfig?

giga993, 2023-10-30:
@Dazza0 Hi, thanks for the quick response. 

We tested it with 2 different hardware devices (all got the ESP32 Wrover E Eco 3 module) and 3 different software projects (one software project per hardware device (which I can not share) and additionally the [gist](https://gist.github.com/giga993/b931b858e09ed1860c56c56ca389812a)/idf example). 

On all 3 software projects idf version is v5.1 and the sdkconfig is as following:

```
#
# TWAI Configuration
#
CONFIG_TWAI_ISR_IN_IRAM=y
CONFIG_TWAI_ERRATA_FIX_BUS_OFF_REC=y
CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST=y
CONFIG_TWAI_ERRATA_FIX_RX_FRAME_INVALID=y
CONFIG_TWAI_ERRATA_FIX_RX_FIFO_CORRUPT=y
CONFIG_TWAI_ERRATA_FIX_LISTEN_ONLY_DOM=y
# end of TWAI Configuration
```

We also tested independently: 
- ISR not in IRAM and `twai_general_config_t .intr_flags = ESP_INTR_FLAG_LEVEL1`
- With and without all errata fixes (all together disabled, we did not test every single variation of errata fixes enabled or disabled)
This did not change the outcome. We could reproduce the issue with any TWAI sdkconfig we tested.

giga993, 2023-10-30:
Strange thing is: the complete TWAI messages are saved in FIFO and a queue. So ID, length and flags are stored right next to the data itself. Tough, the application layer never received a message with wrong ID, length or flags.
Every time we reproduced the problem, it was always the data field. And so far it was never random bytes, but the bytes were shifted by some byte number. (Byte 0 was not Byte 0 anymore but Byte 3 or so).

giga993, 2023-11-09:
Hi @Dazza0,
any idea what we could further test? 
Receiving invalid CAN messages on the application layer is quite a problem.

giga993, 2024-02-21:
@Dazza0 Any updates on this one?

energizer77, 2024-07-11:
We're also seeing this behaviour, where the data is shifted, especially in noisy environments where we have CAN errors.
This is happening on ~1k devices we deployed, but we don't have this on the other devices present on the buses (stm32s with integrated controlers), is there something doable to have reliable data ? Thanks!

giga993, 2024-07-17:
@Dazza0 any updates on this one? 
Having a CAN-Interface randomly making up payload data is a major problem in almost all CAN applications (as long as you rely on the message integrity on the application side).

andrew-elder, 2024-07-22:
How do we get an issue elevated within espressif?

wanckl, 2024-07-24:
Dazza0 is pause to maintian twai driver, I'll trying on this issue, sorry for block long time, I may need some time try to reproduct ,,

giga993, 2024-09-13:
Hi @wanckl,
any progress on this?

giga993, 2025-01-07:
Hi @wanckl, were you able to reproduce the issue?
Like @energizer77 described, the issue has to do with a bad connection / noisy environment / interference and thus broken CAN frames. Other devices on the bus just filter them out by the modul (e.g. my PCAN adapter just sees an error frame) but the ESP is sometimes forwarding, some message which has never been sent, to the application.

wanckl, 2025-04-14:
@giga993 Hi, I think I'm now reproduce this issue, it caused by a frame is interrupted during a frame receiving, then the next received frame will broken data.

enable or disable all erratas don't mind the issue, it myabe another hardware issue.

But I can say now other chips except esp32 don't have this issue.

I'll now try to fix it.

wanckl, 2025-04-14:
> I think I'm now reproduce this issue

I'm NOT correctly reproduce the issue, I de something like disable the config `CONFIG_TWAI_ERRATA_FIX_RX_FRAME_INVALID` then it get samilar issue,,,

After actually enable this errata workaround and using the way mention above, I can NOT reproduce issue again.

But I geuss I'm more close to the issue, @giga993 Could you help appling bellow changes then check what log printed before your data error, 🤝 .

```
diff --git a/components/hal/twai_hal_iram.c b/components/hal/twai_hal_iram.c
index 922701926e..421305cf28 100644
--- a/components/hal/twai_hal_iram.c
+++ b/components/hal/twai_hal_iram.c
@@ -9,6 +9,7 @@
 #include "sdkconfig.h"
 #include "esp_compiler.h"
 #include "hal/twai_hal.h"
+#include "esp_rom_sys.h"

 #ifdef CONFIG_TWAI_ERRATA_FIX_RX_FIFO_CORRUPT
 //Errata condition occurs at 64 messages. Threshold set to 62 to prevent the chance of failing to detect errata condition.
@@ -116,10 +117,12 @@ uint32_t twai_hal_get_events(twai_hal_context_t *hal_ctx)
         twai_ll_err_seg_t seg;
         twai_ll_parse_err_code_cap(hal_ctx->dev, &type, &dir, &seg);    //Decode error interrupt
         //Check for errata condition (RX message has bus error at particular segments)
+        esp_rom_printf("type %d dir %d seg %d\n", type, dir, seg);
         if (dir == TWAI_LL_ERR_DIR_RX &&
             ((seg == TWAI_LL_ERR_SEG_DATA || seg == TWAI_LL_ERR_SEG_CRC_SEQ) ||
              (seg == TWAI_LL_ERR_SEG_ACK_DELIM && type == TWAI_LL_ERR_OTHER))) {
             TWAI_HAL_SET_BITS(events, TWAI_HAL_EVENT_NEED_PERIPH_RESET);
+            esp_rom_printf("errata_invalid\n");
         }
 #endif
         twai_ll_clear_err_code_cap(hal_ctx->dev);

```

energizer77, 2025-05-06:
@wanckl here some logs with your changes :
```
type 1 dir 0 seg 26
BUS_ERROR-ID0309
type 2 dir 1 seg 10
errata_invalid
BUS_ERROR-ID0309
ARB_LOST-ID0309
type 1 dir 0 seg 26
BUS_ERROR-ID0309
type 2 dir 1 seg 10
errata_invalid
BUS_ERROR-ID0309
```

wanckl, 2025-05-07:
@energizer77 Hi, Thank you for your reply

the log is resonable and indicate the workaround is corect, But what I want is the log when your messages broken, can you mark it

giga993, 2025-05-16:
Hi @wanckl 

I ve redone the test with the changes.

I've setup a full project (not just the gist) so you can reproduce it easily:

Hardware setup is same as before: 
ESP32 Wrover E Eco3 16MB Flash, 8MB PSRAM + CAN Transceiver
PEAK-PCAN USB Adapter and PCAN-View
Directly wired together.

Busspeed is at 125kbit/s.

This is the PCAN Message send by the PC: 

`ID=1,Type=D,Length=8,Data=0123456789ABCDEF,CycleTime=5,Paused=1,IDFormat=hex,Comment="TWAI-Test"`

At first you see the:
```
I (334) TWAI Alert and Recovery: Driver started
I (339) TWAI Alert and Recovery: Starting transmissions
E (1310) TWAI Alert and Recovery: CAN receive timed out
E (2310) TWAI Alert and Recovery: CAN receive timed out
E (3310) TWAI Alert and Recovery: CAN receive timed out
E (4310) TWAI Alert and Recovery: CAN receive timed out
E (5310) TWAI Alert and Recovery: CAN receive timed out
```
And it won't appear after the PCAN Message is activated from the PC.

Then I just disconnect the CAN-Plug 1-3 times and after your prints there is a message handled to the application which was never send on the bus:

`E (53013) TWAI Alert and Recovery:      Message ID: 0x1 (1), len: 8, data: 89 AB CD EF 01 23 45 67, msg cnt: 9407`

It is hard to say when this happens, as your logs are printed from the ISR, so with higher priority. 

Here the full log:

<details>

First Try, 3x disconnect:

```
--- Warning: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM3 instead...
--- esp-idf-monitor 1.5.0 on \\.\COM3 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7176
load:0x40078000,len:15576
load:0x40080400,len:4
--- 0x40080400: _init at ??:?

load:0x40080404,len:3904
entry 0x40080640
I (29) boot: ESP-IDF v5.3.1-6-gd9ecbca0b1-dirty 2nd stage bootloader
I (29) boot: compile time May 15 2025 15:05:46
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v3.0
I (39) boot.esp32: SPI Speed      : 80MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 16MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
[... 2205 lines cut ...]
type 0 dir 0 seg 0
type 0 dir 0 seg 0
type 0 dir 0 seg 0
type 0 dir 0 seg 0
type 3 dir 0 seg 25
type 3 dir 0 seg 25
type 3 dir 0 seg 25
type 3 dir 0 seg 25
type 3 dir 0 seg 25
type 3 dir 0 seg 25
type 2 dir 1 seg 6
type 0 dir 0 seg 22
type 0 dir 0 seg 22
type 2 dir 0 seg 6
type 0 dir 1 seg 22
type 0 dir 0 seg 0
type 0 dir 0 seg 0
type 0 dir 0 seg 0
type 0 dir 0 seg 0
type 0 dir 0 seg 0
type 0 dir 0 seg 0
type 0 dir 0 seg 0
```

<\details>

warren-velocity, 2025-06-17:
Hi @wanckl is there any update to this issue? I'm seeing it frequently and it's causing serious issues with my project and customers. thanks

wanckl, 2025-06-18:
@giga993 @warren-velocity Hi, I'm able to reproduce the issue correctly, I need other team to help debugging the issue. 

For now, don't have update.

But what can be confirmed is other chips which support TWAI like ESP32S3 ..don't have this issue. Even the new chip ESP32C5 support TWAI-FD feature. I suggest considering other chips

warren-velocity, 2025-06-18:
Thanks @wanckl . Can you please confirm exactly which chips do have the issue?

giga993, 2026-02-04:
When can you provide a soft fix?

giga993, 2026-09-11:
This issue is open since almost 3 years now. 
Can you provide a fix?
@wanckl

wanckl, 2026-09-11:
@giga993 I'll look on this issue again today, ,sorry

franz-hoepfinger-oscat, 2026-09-11:
I can confirm: the new CAN Driver still has issues. 

- https://gist.github.com/diplfranzhoepfinger/772b9c92d3d5ccbcfc1c2870f1aa5f3e 
- https://gist.github.com/diplfranzhoepfinger/939819be24c3fd17816c71f5fa8e0f94
- https://gist.github.com/diplfranzhoepfinger/885e5ccdb613dc16c1832340d3bfa78c

more coming.

franz-ms-muc, 2026-09-12:
> We're also seeing this behaviour, where the data is shifted, especially in noisy environments where we have CAN errors. This is happening on ~1k devices we deployed, but we don't have this on the [...]

Also old TWAI driver does not have it

diplfranzhoepfinger, 2026-09-12:
https://gist.github.com/diplfranzhoepfinger/4a9f7c8944b7398fe97399b1c4954e4f

`E (18939) TWAI1 Master: CAN1 Status: TEC:16 BusErr:3 StuffErr:1 AckErr:2`

diplfranzhoepfinger, 2026-09-12:
switching to the OLD TWAI Driver on a ESP32-P4: 

https://gist.github.com/diplfranzhoepfinger/7be6ab5a8201659bbea68f16d75aecce

working. 

i was simulating like this: 

150m Cable, 250kbit/s, Simulated Busload with PCAN-View of about 80% Peak, 50% Median. 

screenshot coming.

diplfranzhoepfinger, 2026-09-12:
https://github.com/espressif/esp-idf/issues/18084#issuecomment-5648317415

wanckl, 2026-09-18:
@giga993 @andrew-elder  Hi, For the data error issue, it seems esp32 only hardware issue

From my temporary test, we may need expend the workaround condition:
https://github.com/espressif/esp-idf/blob/master/components/esp_hal_twai/twai_hal_v1.c#L309-L311

Change to
```c
        if ((seg == TWAI_LL_ERR_SEG_DATA || seg == TWAI_LL_ERR_SEG_CRC_SEQ) ||
            (seg == TWAI_LL_ERR_SEG_ACK_DELIM && type == TWAI_LL_ERR_OTHER)) {
```

It remove the `DIR` check. and will bring more `hw reset`, but still better then give broken data silently.

You can try if it is able to fix the issue. and I'll simultaneously check/confirm with designer to find stable solution.

Sorry for long time on this issue.

wanckl, 2026-09-23:
Hi @giga993 @timoxd7 @andrew-elder 

Design team confirmed that it is a hardware issue similar as https://docs.espressif.com/projects/esp-chip-errata/zh_CN/latest/esp32/03-errata-description/esp32/twai-invalid-received-data-frame.html

And only exist on esp32.

Fix code is right as I mentioned above: remove the DIR check for `reset` condition.

We'll update the errata description then and fix in IDF

closed 2026-10-10 by espressif-bot