espressif/esp-idf #12475: ESP32-C3 `gpio_wakeup_enable` does not accept `GPIO_INTR_*EDGE`

cyberhuman, 2023-10-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello.

In my custom board, I have a push button attached to GPIO9 (active low level). I have configured the pin to generate an interrupt when the button is pressed or released:
```
#define BUTTON_PIN GPIO_NUM_9
...
    gpio_config_t io_conf = {};
    io_conf.intr_type = GPIO_INTR_ANYEDGE;
    io_conf.mode = GPIO_MODE_INPUT;
    io_conf.pin_bit_mask = BIT64(BUTTON_PIN);
    io_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
    io_conf.pull_up_en = GPIO_PULLUP_ENABLE;
    gpio_config(&io_conf);

    gpio_install_isr_service(0);
    gpio_isr_handler_add(BUTTON_PIN, button_isr_handler, NULL);
```
So far so good... but when the controller goes into the light sleep mode, the interrupt does not fire, as expected.
Hence I enable the pin to wake up the controller from sleep by calling `gpio_wakeup_enable` for the pin and then `esp_sleep_enable_gpio_wakeup` to enable wakeup by GPIO:
```
    gpio_sleep_sel_dis(BUTTON_PIN);
    gpio_wakeup_enable(BUTTON_PIN, GPIO_INTR_LOW_LEVEL);
    esp_sleep_enable_gpio_wakeup();
```
The problem is, `gpio_wakeup_enable` only accepts `GPIO_INTR_LOW_LEVEL` or `GPIO_INTR_HIGH_LEVEL`!
I looked around in the ESP-IDF code and in the TRM and noticed that the wakeup condition is controlled by the same bits (`GPIO_PINn_INT_TYPE`) in the same register (`GPIO_PINn_REG`) as the interrupt type in `gpio_config` (or, rather, `gpio_set_intr_type`). So I changed the order of the calls:
```
    gpio_sleep_sel_dis(BUTTON_PIN);
    gpio_wakeup_enable(BUTTON_PIN, GPIO_INTR_LOW_LEVEL);
    esp_sleep_enable_gpio_wakeup();

    gpio_config_t io_conf = {};
    io_conf.intr_type = GPIO_INTR_ANYEDGE;
    io_conf.mode = GPIO_MODE_INPUT;
    io_conf.pin_bit_mask = BIT64(BUTTON_PIN);
    io_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
    io_conf.pull_up_en = GPIO_PULLUP_ENABLE;
    gpio_config(&io_conf);

    gpio_install_isr_service(0);
    gpio_isr_handler_add(BUTTON_PIN, button_isr_handler, NULL);
```
and surely enough, the controller now wakes up from the light sleep on any edge change as desired!
So my question is, does it work by chance, or is there an omission in the framework? I didn't manage to find any limitation of the function in the TRM, and since it works, I think it may be because the condition in `gpio_wakeup_enable` was copied from `gpio_deep_sleep_wakeup_enable` and not updated accordingly. Or am I missing something?

esp-lis, 2023-10-30:
> ### Answers checklist.
> * [x]  I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
> * [x]  I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
> * [x]  I have searched the issue tracker for a similar issue and not found a similar issue.
> 
> ### General issue report
> Hello.
> 
> In my custom ESP32-C3-based board, I have a push button attached to GPIO9 (active low level). I have configured the pin to generate an interrupt when the button is pressed or released:
> 
> ```
> #define BUTTON_PIN GPIO_NUM_9
> ...
>     gpio_config_t io_conf = {};
>     io_conf.intr_type = GPIO_INTR_ANYEDGE;
>     io_conf.mode = GPIO_MODE_INPUT;
>     io_conf.pin_bit_mask = BIT64(BUTTON_PIN);
>     io_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
>     io_conf.pull_up_en = GPIO_PULLUP_ENABLE;
>     gpio_config(&io_conf);
> 
>     gpio_install_isr_service(0);
>     gpio_isr_handler_add(BUTTON_PIN, button_isr_handler, NULL);
> ```
> 
> So far so good... but when the controller goes into the light sleep mode, the interrupt does not fire, as expected. Hence I enable the pin to wake up the controller from sleep by calling `gpio_wakeup_enable` for the pin and then `esp_sleep_enable_gpio_wakeup` to enable wakeup by GPIO:
> 
> ```
>     gpio_sleep_sel_dis(BUTTON_PIN);
>     gpio_wakeup_enable(BUTTON_PIN, GPIO_INTR_LOW_LEVEL);
>     esp_sleep_enable_gpio_wakeup();
> ```
> 
> The problem is, `gpio_wakeup_enable` only accepts `GPIO_INTR_LOW_LEVEL` or `GPIO_INTR_HIGH_LEVEL`! I looked around in the ESP-IDF code and in the TRM and noticed that the wakeup condition is controlled by the same bits (`GPIO_PINn_INT_TYPE`) in the same register (`GPIO_PINn_REG`) as the interrupt type in `gpio_config` (or, rather, `gpio_set_intr_type`). So I changed the order of the calls:
> 
> ```
>     gpio_sleep_sel_dis(BUTTON_PIN);
>     gpio_wakeup_enable(BUTTON_PIN, GPIO_INTR_LOW_LEVEL);
>     esp_sleep_enable_gpio_wakeup();
> 
>     gpio_config_t io_conf = {};
>     io_conf.intr_type = GPIO_INTR_ANYEDGE;
>     io_conf.mode = GPIO_MODE_INPUT;
>     io_conf.pin_bit_mask = BIT64(BUTTON_PIN);
>     io_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
>     io_conf.pull_up_en = GPIO_PULLUP_ENABLE;
>     gpio_config(&io_conf);
> 
>     gpio_install_isr_service(0);
>     gpio_isr_handler_add(BUTTON_PIN, button_isr_handler, NULL);
> ```
> 
> and surely enough, the controller now wakes up from the light sleep on any edge change as desired! So my question is, does it work by chance, or is there an omission in the framework? I didn't manage to find any limitation of the function in the TRM, and since it works, I think it may be because the condition in `gpio_wakeup_enable` was copied from `gpio_deep_sleep_wakeup_enable` and not updated accordingly. Or am I missing something?

GPIO wakeup signals only support level-type, edge-type is not supported in the design (edge-type wakeup requires clock support, and there is no available clock after the hardware enters light sleep).

cyberhuman, 2023-11-03:
@esp-lis thank you. Does it matter that my board has an external 32KHz crystal, or it should not be working still?
I really wonder how it works for me.

esp-lis, 2023-11-06:
@cyberhuman You can refer to this option, it's located at (Top) -> Component config -> Hardware Settings -> RTC Clock Config -> RTC clock source -> External 32kHz crystal.

cyberhuman, 2023-11-06:
@esp-lis thank you, I know about this option.  I use BLE and automatic light sleep with 32kHz crystal to save power.
My question is whether having the external crystal oscillator helps with light sleep wakeup on edge change. You said there is no available clock, but the slow clock is available. Is it used by the chip in this case?

esp-lis, 2024-03-04:
@cyberhuman refer to the figure below (It is described in section **6.2.2 Architectural Overview** of [esp32-c3_technical_reference_manual_en](https://www.espressif.com.cn/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf)), the green dotted line marks the working clock path of the RTC module (the working clock drives the count of the RTC Timer counter), and the red dotted line marks the logical clock path of the RTC module (the logic clock drives the circuit logic of the RTC module, and __the hardware will automatically gate or power off this part of the clock during sleep to save power__). The wake-up of edge type belongs to the circuit logic, which needs to get the level state of at least two clock cycles to determine whether it is a pos-edge or a neg-edge.
[image]

esp-lis, 2024-03-04:
@cyberhuman If you have automatic light sleep enabled on your software system and need to use GPIO wake up, and use GPIO interrupt as an event source for some program actions, you can refer to the example in [esp-idf!12126](https://github.com/espressif/esp-idf/pull/12126), It provides a simple message-based event-driven logic that you can modify to support more features.

cyberhuman, 2024-04-15:
Thank you for the explanation!

closed 2024-04-15 by cyberhuman