espressif/esp-idf #12477: The W5500 + SPI_Slave driver does not work stably (IDFGH-11328)

VasylenkoYuriy, 2023-10-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1

### Espressif SoC revision.

ESP32 Wroon 32

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP-32 WROOM WIFI + BLUETOOTH ESP32 NODEMCU

### Power Supply used.

USB

### What is the expected behavior?

Sorry for my bad English.

My device needs to receive data via SPI every 1ms and send it via Eth W5500. Using WiFi instead of W5500, the project works stably.
I have already tested the operation of the W5500 + SPIslave(DMA) for two weeks, so I am providing a short code to find bugs, and not a real project.

### What is the actual behavior?

LWIP is not running, the socket is not created, only the Eth driver is running and SPI_Slave works and receives messages every 1 ms. From a personal computer via a local network I send UDP packets to the IP address of my ESP32 at any random port. After some time a failure occurs. If you turn off SPI everything works fine.

### Steps to reproduce.

My main code:

```c
void app_main(void)
{
    esp_eth_handle_t myEth = MyInitEth();
    
    esp_netif_init(); // Initialize TCP/IP network interface (should be called only once in application)
     
    nnetif(myEth);
    
    vTaskDelay(7000);
   
    xTaskCreate(My_SPI_Slave_task, "My_SPI_Slave", 4096, NULL, 5, NULL);

    //xTaskCreate(My_udp_server_task, "My_udp_server", 4096,NULL, 5, NULL);
}
```

My task SPI code:

```c
#define GPIO_MOSI 13
[... 273 lines cut ...]
0x4008d182: vPortTaskWrapper at C:/Users/yuriy/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 0928843e63d183e2

Rebooting...
 

When increasing rx_task_stack_size = 2048 to rx_task_stack_size = 8192 errors:
```typescript
E (120665) w5500.mac: emac_w5500_alloc_recv_buf(535): invalid frame length 3
E (120665) w5500.mac: no mem for receive buffer
E (122156) w5500.mac: emac_w5500_alloc_recv_buf(535): invalid frame length 3
E (122156) w5500.mac: no mem for receive buffer
E (122966) w5500.mac: emac_w5500_alloc_recv_buf(535): invalid frame length 3
E (122966) w5500.mac: no mem for receive buffer
E (123003) w5500.mac: emac_w5500_alloc_recv_buf(535): invalid frame length 0
E (123004) w5500.mac: emac_w5500_alloc_recv_buf(535): invalid frame length 0
E (123008) w5500.mac: emac_w5500_alloc_recv_buf(535): invalid frame length 0
``` 
and so on ad infinitum, the work is very unstable.
```

### More Information.

Accidentally assigned the wrong CS pin to my ESP32 spi_slave, which was always in the "0" state. The data was taken due to buffer overflow, there was no error. If you use WiFi instead of W5500, the work is stable and there are no errors. If you remove the Ethernet cable or ensure that no packets are exchanged, there is no error

wanckl, 2023-10-31:
@VasylenkoYuriy Sorry for letting you suffer issues, can you provide the commit id you are working on? thank you

VasylenkoYuriy, 2023-10-31:
> @VasylenkoYuriy Sorry for letting you suffer issues, can you provide the commit id you are working on? thank you

Hi @wanckl!
I'm sorry for my bad english

If you asked which version of esp-idf I am working with, I am attaching a picture of where and how I downloaded it. esp-idf 5.1. Downloaded from Esspresif.
If the question is about my code, then here is the project commit
[](https://github.com/VasylenkoYuriy/mcu.git)
[image]

Thank you!

VasylenkoYuriy, 2023-10-31:
> @VasylenkoYuriy Sorry for letting you suffer issues, can you provide the commit id you are working on? thank you

@wanckl, The W5500 driver does not work together with receiving data via SPI_slave. Provided that there is input traffic (any packets to any port) on the W5500. error "E (120665) w5500.mac: emac_w5500_alloc_recv_buf(535): invalid frame length 3
E (120665) w5500.mac: no mem for receive buffer" or ```
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception).
Debug exception reason: Stack canary watchpoint triggered (w5500_tsk) 
Core  1 register dump:
PC      : 0x4008847f  PS      : 0x00060736  A0      : 0x8008d350  A1      : 0x3ffb8ec0  
0x4008847f: esp_cpu_compare_and_set at C:/Users/yuriy/esp/esp-idf/components/esp_hw_support/cpu.c:410

A2      : 0x3ffb0084  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00000058  
A6      : 0x0000000a  A7      : 0xff000000  A8      : 0x3ffb22e4  A9      : 0x5a97e5cc  
A10     : 0x3ffb0234  A11     : 0x3ffb22e4  A12     : 0x3ffb93f4  A13     : 0x00000005  
A14     : 0x3ffb93bc  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff9  
0x400014fd: strlen in ROM

0x4000150d: strlen in ROM

```

kostaond, 2023-11-03:
@VasylenkoYuriy, how is your W5500 connected to ESP32? Do you use some W5500 devkit? If so, which one?

Could you please try to use `ethernet\basic` example and try to run it on your setup? Just try to ping ESP32.

kostaond, 2024-04-19:
I must close due to lack of feedback... Feel free to re-open and please provide additional info as requested.

closed 2024-04-19 by kostaond