espressif/esp-idf #12478: example/wifi/softap_sta not working on ESP32-C6-DevKitC1

kojibuta, 2023-10-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.2-dev-3775-gb4268c874a

### Espressif SoC revision.

ESP32-C6 (QFN40) (revision v0.0)

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

The ESP32-C6 should connect to the Wi-Fi router and obtain an IP address but it doesn't.

### What is the actual behavior?

The ESP32-C6 (apparently) connectes to the Wi-Fi router but it never obtains an IP address.

### Steps to reproduce.

1. goto folder examples/wifi/softap_sta
2. configure ESP_WIFI_REMOTE_AP_SSID and ESP_WIFI_REMOTE_AP_PASSWORD using "idf.py menuconfig"
3. compile using "idf.py build"
4. flash to ESP32-C6-DevKitC-1 using "idf.py flash"

### Debug Logs.

```plain
--- esp-idf-monitor 1.3.3 on /dev/cu.usbmodem1101 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (239) esp_image: segment 4: paddr=000b4170 vaddr=40814890 size=034a8h ( 13480) load
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001a36e
0x4001a36e: ets_sha_process in ROM

SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1804
load:0x4086c410,len:0xde0
load:0x4086e610,len:0x2dfc
entry 0x4086c41a
I (23) boot: ESP-IDF v5.2-dev-3775-gb4268c874a 2nd stage bootloader
I (23) boot: compile time Oct 29 2023 20:18:35
I (24) boot: chip revision: v0.0
[... 122 lines cut ...]
I (3983) wifi:pm start, type: 1, itwt_start:0

I (3983) wifi:pm start, type:1, aid:0xd, trans-BSSID:36:02:db:cd:a8:8e, BSSID[5]:0x8e, mbssid(max-indicator:0, index:0), he:1
I (3983) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (3993) wifi:set rx beacon pti, rx_bcn_pti: 10, bcn_timeout: 25000, mt_pti: 10, mt_time: 10000
I (4003) wifi:[ADDBA]TX addba request, tid:0, dialogtoken:1, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (4013) wifi:[ADDBA]TX addba request, tid:7, dialogtoken:2, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x20)
I (4023) wifi:[ADDBA]TX addba request, tid:5, dialogtoken:3, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (4033) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (4043) wifi:(opr)len:7, TWT Required:0, VHT Operation Present:0, 6GHz Info Present:0, Co-Hosted BSS:0, Basic MCS and NSS:0xfffc
I (4053) wifi:(opr)len:7, Default PE Duration:7, TXOP RTS Threshold:12(384 us), ER-SU-Disable:1
I (4063) wifi:(opr)len:7, BSS Color:34, disabled:0, Partial BSS Color:0
I (4063) wifi:[ADDBA]RX addba response, status:0, tid:0/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:32
I (4073) wifi:[ADDBA]RX addba response, status:0, tid:7/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:32
I (4083) wifi:[ADDBA]RX addba response, status:0, tid:5/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:32
W (6223) wifi:<ba-add>idx:0, ifx:0, tid:0, TAHI:0x1008ea8, TALO:0xcddb0236, (ssn:0, win:64, cur_ssn:0), CONF:0xc0000005
```

### More Information.

The ESP32-C6 connectes to remote Wi-Fi router and obtains an IP address only forcing WIFI_MODE_STA. It does not work in WIFI_MODE_APSTA mode.
When running in WIFI_MODE_APSTA mode it connects to remote Wi-Fi router but it does not obtain an IP address.
It remains stuck forever.
If later, a client connects to the ESP32-C6 SoftAP, as soon as the event WIFI_EVENT_AP_STACONNECTED is triggered, the ESP32-C6 immediately obtains and IP address from the remote Wi-Fi router too.
Please see events starting at time 53863 in this log [log001.txt](https://github.com/espressif/esp-idf/files/13199079/log001.txt)

AxelLin, 2023-10-30:
https://github.com/espressif/esp-idf/issues/12454

kojibuta, 2023-11-08:
Hi @AxelLin, do you have any update omn this topic? I see https://github.com/espressif/esp-idf/issues/12454 is closed but I still cannot run the softap_sta example successfully on ESP32-C6.
Thanks

AxelLin, 2023-11-09:
@kojibuta  
#12454 is fixed in v5.1 branch, I'm not sure if v5.2 branch includes the fix.
Maybe you can update your v5.2 branch (v5.2-dev-3903-g66992aca7a) and check again.

xuxiao111, 2023-11-09:
Hi @kojibuta @AxelLin,
this issue has been fixed in this commit: https://github.com/espressif/esp-idf/commit/462d57e18f0ab32e9fd84d0bd4d9b24f748026b8, you can update your v5.2 branch and have a try.

kojibuta, 2023-11-09:
Hi @AxelLin @xuxiao111, I updated to the last commit and the example now works as expected.
Thank you.

closed 2023-11-09 by kojibuta