espressif/esp-idf #12479: Blufi例程烧录后使用EspBluFI APP无法配网

TkoSky, 2023-10-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Blufi例程烧录后使用EspBluFI APP无法配网，使用模组为ESP32-C3-WROOM-02

TkoSky, 2023-10-30:
[image]
[image]

TkoSky, 2023-10-30:
同样的例程代码我使用S3和C6都是能够正常配网的，C3好像是点击连接然后就马上断开了

TkoSky, 2023-10-30:
新情况，C3的这份代码使用安卓无法配网，但是使用ios是可以配网的
ESP-IDF版本为esp-idf-v5.1

zhp0406, 2023-10-31:
@TkoSky 
感谢你报告问题，你这个问题是必现的吗？
我这面没有复现出来该问题，下面是我的完整的  log：
I (30) boot: ESP-IDF v5.1 2nd stage bootloader
I (30) boot: compile time Oct 31 2023 19:34:02
I (30) boot: chip revision: v0.3
I (33) boot.esp32c3: SPI Speed      : 80MHz
I (38) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 2MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00177000
I (86) boot: End of partition table
I (90) esp_image: segment 0: paddr=00010020 vaddr=3c0d0020 size=2ebb0h (191408) map
I (129) esp_image: segment 1: paddr=0003ebd8 vaddr=3fc93600 size=01440h (  5184) load
I (130) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=cde58h (843352) map
I (269) esp_image: segment 3: paddr=0010de80 vaddr=3fc94a40 size=01cc4h (  7364) load
I (271) esp_image: segment 4: paddr=0010fb4c vaddr=40380000 size=13480h ( 78976) load
I (296) boot: Loaded app from partition at offset 0x10000
I (296) boot: Disabling RNG early entropy source...
I (308) cpu_start: Unicore app
I (308) cpu_start: Pro cpu up.
I (316) cpu_start: Pro cpu start user code
I (316) cpu_start: cpu freq: 160000000 Hz
I (317) cpu_start: Application information:
I (319) cpu_start: Project name:     blufi_demo
I (325) cpu_start: App version:      v5.1
I (329) cpu_start: Compile time:     Oct 31 2023 19:33:55
I (335) cpu_start: ELF file SHA256:  50d68ac1...
I (341) cpu_start: ESP-IDF:          v5.1
I (345) cpu_start: Min chip rev:     v0.3
I (350) cpu_start: Max chip rev:     v0.99 
I (355) cpu_start: Chip rev:         v0.3
I (360) heap_init: Initializing. RAM available for dynamic allocation:
I (367) heap_init: At 3FC9D3B0 len 0003F360 (252 KiB): DRAM
I (373) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (380) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (387) spi_flash: detected chip: generic
I (391) spi_flash: flash io: dio
W (395) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (408) sleep: Configure to isolate all GPIO pins in sleep state
I (415) sleep: Enable automatic switching of GPIO sleep configuration
I (422) coexist: coex firmware version: ebddf30
I (427) coexist: coexist rom version 9387209
I (432) app_start: Starting scheduler on CPU0
I (437) main_task: Started on CPU0
I (437) main_task: Calling app_main()
I (447) pp: pp rom version: 9387209
I (447) net80211: net80211 rom version: 9387209
I (457) wifi:wifi driver task: 3fca6104, prio:23, stack:6656, core=0
I (457) wifi:wifi firmware version: b2f1f86
I (457) wifi:wifi certification version: v7.0
I (467) wifi:config NVS flash: enabled
I (467) wifi:config nano formating: disabled
I (467) wifi:Init data frame dynamic rx buffer num: 32
I (477) wifi:Init management frame dynamic rx buffer num: 32
I (477) wifi:Init management short buffer num: 32
I (487) wifi:Init dynamic tx buffer num: 32
I (487) wifi:Init static tx FG buffer num: 2
I (497) wifi:Init static rx buffer size: 1600
I (497) wifi:Init static rx buffer num: 10
I (497) wifi:Init dynamic rx buffer num: 32
I (507) wifi_init: rx ba win: 6
I (507) wifi_init: tcpip mbox: 32
I (517) wifi_init: udp mbox: 6
I (517) wifi_init: tcp mbox: 6
I (517) wifi_init: tcp tx win: 5744
I (527) wifi_init: tcp rx win: 5744
I (527) wifi_init: tcp mss: 1440
I (537) wifi_init: WiFi IRAM OP enabled
I (537) wifi_init: WiFi RX IRAM OP enabled
I (547) phy_init: phy_version 970,1856f88,May 10 2023,17:44:12
I (587) wifi:mode : sta (68:67:25:06:ef:a4)
I (587) wifi:enable tsf
I (597) BLE_INIT: BT controller compile version [85b425c]
I (597) BLE_INIT: Bluetooth MAC: 68:67:25:06:ef:a6

I (607) BLUFI_EXAMPLE: BD ADDR: 68:67:25:06:ef:a6

I (607) BLUFI_EXAMPLE: BLUFI init finish

I (617) BLUFI_EXAMPLE: BLUFI VERSION 0103

I (617) main_task: Returned from app_main()
I (17117) BLUFI_EXAMPLE: BLUFI ble connect

zhp0406, 2023-10-31:
@TkoSky 
麻烦多再测试几次看看能不能配网成功，如果还是有问题，请把完整的 log 给我，谢谢。

TkoSky, 2023-11-01:
必现的

TkoSky, 2023-11-01:
I (30) boot: ESP-IDF v5.1.1 2nd stage bootloader
I (30) boot: compile time Nov  1 2023 09:19:19
I (30) boot: chip revision: v0.4
I (33) boot.esp32c3: SPI Speed      : 80MHz
I (38) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00177000
I (86) boot: End of partition table
I (90) esp_image: segment 0: paddr=00010020 vaddr=3c0d0020 size=2ed78h (191864) map
I (129) esp_image: segment 1: paddr=0003eda0 vaddr=3fc93800 size=01278h (  4728) load
I (130) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=ce7e0h (845792) map
I (270) esp_image: segment 3: paddr=0010e808 vaddr=3fc94a78 size=01e3ch (  7740) load
I (272) esp_image: segment 4: paddr=0011064c vaddr=40380000 size=136b8h ( 79544) load
I (297) boot: Loaded app from partition at offset 0x10000
I (297) boot: Disabling RNG early entropy source...
I (308) cpu_start: Unicore app
I (309) cpu_start: Pro cpu up.
I (317) cpu_start: Pro cpu start user code
I (317) cpu_start: cpu freq: 160000000 Hz
I (317) cpu_start: Application information:
I (320) cpu_start: Project name:     blufi_demo
I (325) cpu_start: App version:      1
I (330) cpu_start: Compile time:     Nov  1 2023 09:18:51
I (336) cpu_start: ELF file SHA256:  b5bac8e7cc7edf76...
I (342) cpu_start: ESP-IDF:          v5.1.1
I (347) cpu_start: Min chip rev:     v0.3
I (351) cpu_start: Max chip rev:     v0.99 
I (356) cpu_start: Chip rev:         v0.4
I (361) heap_init: Initializing. RAM available for dynamic allocation:
I (368) heap_init: At 3FC9D590 len 0003F180 (252 KiB): DRAM
I (374) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (381) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (388) spi_flash: detected chip: generic
I (392) spi_flash: flash io: dio
I (396) sleep: Configure to isolate all GPIO pins in sleep state
I (403) sleep: Enable automatic switching of GPIO sleep configuration
I (410) coexist: coex firmware version: 80b0d89
I (415) coexist: coexist rom version 9387209
I (420) app_start: Starting scheduler on CPU0
I (425) main_task: Started on CPU0
I (425) main_task: Calling app_main()
I (435) pp: pp rom version: 9387209
I (435) net80211: net80211 rom version: 9387209
I (445) wifi:wifi driver task: 3fca6260, prio:23, stack:6656, core=0
I (445) wifi:wifi firmware version: ce9244d
I (445) wifi:wifi certification version: v7.0
I (455) wifi:config NVS flash: enabled
I (455) wifi:config nano formating: disabled
I (455) wifi:Init data frame dynamic rx buffer num: 32
I (465) wifi:Init management frame dynamic rx buffer num: 32
I (465) wifi:Init management short buffer num: 32
I (475) wifi:Init dynamic tx buffer num: 32
I (475) wifi:Init static tx FG buffer num: 2
I (485) wifi:Init static rx buffer size: 1600
I (485) wifi:Init static rx buffer num: 10
I (485) wifi:Init dynamic rx buffer num: 32
I (495) wifi_init: rx ba win: 6
I (495) wifi_init: tcpip mbox: 32
I (505) wifi_init: udp mbox: 6
I (505) wifi_init: tcp mbox: 6
I (505) wifi_init: tcp tx win: 5744
I (515) wifi_init: tcp rx win: 5744
I (515) wifi_init: tcp mss: 1440
I (525) wifi_init: WiFi IRAM OP enabled
I (525) wifi_init: WiFi RX IRAM OP enabled
I (535) phy_init: phy_version 970,1856f88,May 10 2023,17:44:12
I (575) wifi:mode : sta (48:31:b7:39:7f:8c)
I (575) wifi:enable tsf
I (575) BLE_INIT: BT controller compile version [963cad4]
I (575) BLE_INIT: Bluetooth MAC: 48:31:b7:39:7f:8e

I (595) BLUFI_EXAMPLE: BD ADDR: 48:31:b7:39:7f:8e

I (595) BLUFI_EXAMPLE: BLUFI init finish

I (595) BLUFI_EXAMPLE: BLUFI VERSION 0103

I (595) main_task: Returned from app_main()
I (12145) BLUFI_EXAMPLE: BLUFI ble connect

W (23245) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x8
I (23245) BLUFI_EXAMPLE: BLUFI ble disconnect

TkoSky, 2023-11-01:
必现的，手机型号为红米K60 @zhp0406

zhp0406, 2023-11-01:
@TkoSky 
我测试了 v5.1.1 版本（根据你的 log ），因为我没有 K60 设备。我用 Oneplus 进行了测试，未能复现问题。我怀疑问题可能只存在于红米设备上。你可以尝试在其他设备上进行测试吗？

另外，我已经记录了一些 HCI 日志。你能使用这个补丁，在 K60 设备上再测试一下吗？
[Uploading 0001-fix-bt-Display-hci-data-including-up-and-down.patch…]()

zhp0406, 2023-11-01:
[0001-fix-bt-Display-hci-data-including-up-and-down (1).patch](https://github.com/espressif/esp-idf/files/13223388/0001-fix-bt-Display-hci-data-including-up-and-down.1.patch)

zhp0406, 2023-11-01:
@TkoSky 
如果你有抓包卡的话，能提供一下抓包的数据更好。

TkoSky, 2023-11-01:
我这边没有抓包卡，我这边验证了一下确实是红米设备不行，我这边用了一台三星手机是可以配网的，其他品牌设备不确定，上面这个补丁我应该怎么使用呢？ @zhp0406

zhp0406, 2023-11-01:
你需要进入 esp--idf 的目录，然后在该目录下应用 patch：

**应用补丁**：它读取补丁文件并将补丁描述的更改应用到你的代码库中，下面是:

```
git am < [patch-file]
```

zhp0406, 2023-11-01:
@TkoSky 
你这面都测试了哪些红米设备？我这面尽可能多找一些红米手机来做测试。

TkoSky, 2023-11-01:
目前我只测试了红米的K60，有可能是这个型号的问题？

TkoSky, 2023-11-01:
host_recv_pkt_cb,544 ^^^^04 3e 1f 0a 00 01 00 01 01 42 b2 50 7a 3b 5f 00 00 00 00 00 00 00 00 00 00 00 00 18 00 00 00 f4 01 05
transmit_command-->1d 04 02 01 00
host_recv_pkt_cb,544 ^^^^04 0f 04 00 05 1d 04
host_recv_pkt_cb,544 ^^^^04 0c 08 00 01 00 0c 1d 00 61 4b
transmit_command-->16 20 02 01 00
host_recv_pkt_cb,544 ^^^^04 0f 04 00 05 16 20
host_recv_pkt_cb,544 ^^^^04 3e 0c 04 00 01 00 ff f9 01 f7 8e 00 00 00
transmit_command-->22 20 06 01 00 fb 00 48 08
host_recv_pkt_cb,544 ^^^^04 0e 06 05 22 20 00 01 00
I (34405) BLUFI_EXAMPLE: BLUFI ble connect

transmit_command-->0a 20 01 00
host_recv_pkt_cb,544 ^^^^04 0e 04 05 0a 20 00
host_recv_pkt_cb,544 ^^^^04 3e 0b 07 01 00 fb 00 48 08 1b 00 48 01
host_recv_pkt_cb,544 ^^^^04 3e 0d 05 01 00 00 00 00 00 00 00 00 00 00 00
host_recv_pkt_cb,544 ^^^^04 05 04 00 01 00 08 
W (45491) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x8
I (45492) BLUFI_EXAMPLE: BLUFI ble disconnect

transmit_command-->08 20 20 1e 02 01 06 0d 09 42 4c 55 46 49 5f 44 45 56 49 43 45 02 0a 09 03 03 ff ff 05 12 06 00 10 00 00 
host_recv_pkt_cb,544 ^^^^04 0e 04 05 08 20 00 
transmit_command-->0a 20 01 00 
host_recv_pkt_cb,544 ^^^^04 0e 04 05 0a 20 00 
transmit_command-->06 20 0f 00 01 00 01 00 00 00 00 00 00 00 00 00 07 00 
host_recv_pkt_cb,544 ^^^^04 0e 04 05 06 20 00 
transmit_command-->0a 20 01 00 
host_recv_pkt_cb,544 ^^^^04 0e 04 05 0a 20 00 
transmit_command-->0a 20 01 01 
host_recv_pkt_cb,544 ^^^^04 0e 04 05 0a 20 00 
host_recv_pkt_cb,544 ^^^^04 3e 1f 0a 00 01 00 01 01 53 e5 d5 91 e8 7b 00 00 00 00 00 00 00 00 00 00 00 00 18 00 00 00 f4 01 05 
transmit_command-->1d 04 02 01 00 
host_recv_pkt_cb,544 ^^^^04 0f 04 00 05 1d 04 
host_recv_pkt_cb,544 ^^^^04 0c 08 00 01 00 0c 1d 00 61 4b 
transmit_command-->16 20 02 01 00
host_recv_pkt_cb,544 ^^^^04 0f 04 00 05 16 20
host_recv_pkt_cb,544 ^^^^04 3e 0c 04 00 01 00 ff f9 01 f7 8e 00 00 00
transmit_command-->22 20 06 01 00 fb 00 48 08 
host_recv_pkt_cb,544 ^^^^04 0e 06 05 22 20 00 01 00
I (56186) BLUFI_EXAMPLE: BLUFI ble connect

transmit_command-->0a 20 01 00
host_recv_pkt_cb,544 ^^^^04 0e 04 05 0a 20 00 
host_recv_pkt_cb,544 ^^^^04 3e 0b 07 01 00 fb 00 48 08 1b 00 48 01
host_recv_pkt_cb,544 ^^^^04 3e 0d 05 01 00 00 00 00 00 00 00 00 00 00 00 
host_recv_pkt_cb,544 ^^^^04 05 04 00 01 00 08 
W (67272) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x8
I (67272) BLUFI_EXAMPLE: BLUFI ble disconnect

transmit_command-->08 20 20 1e 02 01 06 0d 09 42 4c 55 46 49 5f 44 45 56 49 43 45 02 0a 09 03 03 ff ff 05 12 06 00 10 00 00
host_recv_pkt_cb,544 ^^^^04 0e 04 05 08 20 00 
transmit_command-->0a 20 01 00
host_recv_pkt_cb,544 ^^^^04 0e 04 05 0a 20 00
transmit_command-->06 20 0f 00 01 00 01 00 00 00 00 00 00 00 00 00 07 00
host_recv_pkt_cb,544 ^^^^04 0e 04 05 06 20 00 
transmit_command-->0a 20 01 00
host_recv_pkt_cb,544 ^^^^04 0e 04 05 0a 20 00
transmit_command-->0a 20 01 01
host_recv_pkt_cb,544 ^^^^04 0e 04 05 0a 20 00

TkoSky, 2023-11-01:
上面是更新了补丁之后的log @zhp0406

zhp0406, 2023-11-02:
@TkoSky 
目前怀疑只有红米 k60 有问题。

zhp0406, 2023-11-02:
@TkoSky 
你把下面代码注释掉一下看看问题是否还存在：
components/bt/host/bluedroid/stack/hcic/hciblecmds.c
btsnd_hcic_ble_set_data_length
[image]

zhp0406, 2023-11-08:
@TkoSky 
测试有结果吗？

TkoSky, 2023-11-16:
最近有点忙，还没时间测试这个，等我有时间测试同步一下结果给你 @zhp0406

TkoSky, 2023-11-18:
你好按照你的说法在1010行添加了return TRUE,但是还是会出现这个问题 @zhp0406

zhp0406, 2023-11-20:
@TkoSky 
好的，目前看来还是只跟 K60 手机有关系，我这面使用其他设备一直没有复现出来问题。

zhp0406, 2023-11-22:
我刚试了一下 k50 (至尊版)，也是正常的。

zhp0406, 2023-12-08:
我用 k60，blufi 使用的是 ESP32-C3 v5.1.1 ，AP 使用的是 Iphone13（17.1.1），但是，没有成功复现你的问题，
这是测试流程录屏：
https://github.com/espressif/esp-idf/assets/131946088/25b3d0c5-e49c-467a-b4fd-da5f893f7696
这是手机版本信息：
[image]
不确定是不是跟 K60 的版本有关系。

---
使用之前清除手机和 ESP32-C3 的绑定信息。可以再测试一下吗？
其中 ESP32-C3 的绑定信息你可以使用下面命令：
idf.py erase-flash -p  /dev/ttyS0
 /dev/ttyS0 是端口信息，可以实际修改。
 
 如果还有问题，请把你的 K60 版本告诉我，并告诉我你使用的 AP 的信息。

zhp0406, 2023-12-20:
@TkoSky 
你可以给我一下你具体使用的 commit 吗？

TkoSky, 2023-12-24:
[image]
抱歉最近没看信息，这个是我的手机参数信息，看看是否能给到你帮助 @zhp0406

zhp0406, 2023-12-25:
@TkoSky 
我使用的 k60 版本和你一样，但是没有复现出来问题。
你那面可以换一个 ESP32-C3 测试一下吗？排除一下与芯片无关系。

TkoSky, 2024-02-01:
我这边试了一下，换了一颗C3的芯片也是不能配网，应该不像芯片的问题，因为使用其他型号手机测试都是可以的，或者方便你那边编译一个固件我烧录进去帮你测试一下？

TkoSky, 2024-02-01:
对了，我现在手机升级了小米新的那个HyperOS1.0.4版本测试也是无法配网， @zhp0406

zhp0406, 2024-02-04:
@TkoSky 
在你的 log 中报告了 0x08,0x08 代表超时断开：
[image]
既： ESP32-C3 在没有在规定时间内收到对方的 ack

我这面不太能确定是小米手机没有回 ack，还是回了 ack 没有收到。如果你能提供一下抓包文件，那该问题就比较清晰了。
使用相同的芯片和手机，我这面的没有复现你的问题，不确定我们之间的差异在哪里。
考虑到我们最近也修复了一些兼容性 bug，你能否使用再使用一下最新的 master 分支进行测试？
如果还是有问题，下面方法粗略确定一下是 host 问题还是 controller 问题，
1-使用 ESP32-C2 ESP32-C6 或者 ESP32 芯片进行一下测试，该芯片具有不同的 controller
2-使用 nimble host
更改 (Top) → Component config → Bluetooth → Bluetooth → Host 为 (X) NimBLE - BLE only

zhp0406, 2024-02-28:
Hi，有进行测试吗？

closed 2024-03-11 by Alvin1Zhang