espressif/esp-idf #12483: ESP32-C3 Load Access Fault

jortega11, 2023-10-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello! I am working on a project using a `ESP32-WROOM-32U` and I wanted to migrate it to a `Beetle ESP32-C3` from the company `DFRobot`. My project is working fine, but when I changed it to the new board I started getting a **fatal error Load Access Fault**.

How could I solve the problem or know what is causing the issue?

Thank you very much!

o-marshmallow, 2023-10-31:
Hello @jortega11 ,

We will need more information to help you:
* Which version of ESP-IDF are you using?
* Which ESP32-C3 revision are you using?
* How did you build the project?
* What are the logs actually showing?
* What is you project configuration?
* What is your project actually using/doing? For example, the ESP32 has two cores, two I2C ports whereas the C3 has a single core and one I2C port, this can lead to issues.
* Are you using IDF monitor? If yes, what is the backtrace shown by it when the panic occurs?

jortega11, 2023-11-10:
I was able to find the error. It was caused because in a print statement a pointer was being passed and that was causing the error, which makes sense because the Load Access Fault fatal error was related to pointers.

closed 2023-11-10 by jortega11