espressif/esp-idf #12484: esp-idf-v5.1-rc2/install.sh esp32s3 results in error.

kikura, 2023-10-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have tried the compiling and flashing instructions for Tulip CC here:

https://github.com/bwhitman/tulipcc/blob/main/docs/tulip_flashing.md

I got as far as:

esp-idf-v5.1-rc2/install.sh esp32s3

but get the following output and error message:

Detecting the Python interpreter
Checking "python3" ...
Python 3.10.12
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Current system platform: linux-amd64
Selected targets are: esp32s3
Installing tools: xtensa-esp-elf-gdb, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs
Installing xtensa-esp-elf-gdb@12.1_20221002
Downloading https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v12.1_20221002/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz
Destination: /home/nick/.espressif/dist/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz.tmp
80%WARNING: file size mismatch for /home/nick/.espressif/dist/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz.tmp, expected 32006939, got 25735652
WARNING: Download failure: <urlopen error retrieval incomplete: got only 25735652 out of 32006939 bytes>
WARNING: Failed to download https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v12.1_20221002/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz to /home/nick/.espressif/dist/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz.tmp
Downloading https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v12.1_20221002/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz
Destination: /home/nick/.espressif/dist/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz.tmp
89%WARNING: file size mismatch for /home/nick/.espressif/dist/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz.tmp, expected 32006939, got 28766692
WARNING: Download failure: <urlopen error retrieval incomplete: got only 28766692 out of 32006939 bytes>
WARNING: Failed to download https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v12.1_20221002/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz to /home/nick/.espressif/dist/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz.tmp
Downloading https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v12.1_20221002/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz
Destination: /home/nick/.espressif/dist/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz.tmp
92%WARNING: file size mismatch for /home/nick/.espressif/dist/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz.tmp, expected 32006939, got 29721336
WARNING: Download failure: <urlopen error retrieval incomplete: got only 29721336 out of 32006939 bytes>
WARNING: Failed to download https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v12.1_20221002/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz to /home/nick/.espressif/dist/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz.tmp
ERROR: Failed to download, and retry count has expired
Please make sure you have a working Internet connection.

This is on the latest version of Ubuntu.

It has been suggested I try here for a solution.

Any ideas? Thanks in advance.

dobairoland, 2023-10-30:
As the output suggests, you probably have some issue with your Internet connection because it is not able to completely download the files. You could try to workaround the issue by manually downloading those files and placing them into the  /home/nick/.espressif/dist directory. 

For example https://aria2.github.io/ might be one of the solutions. The simplest one could be your Web browser. Please let me know if this helps and we might consider implementing something similar.

kikura, 2023-10-30:
Thanks. The weird thing is downloads for everything else works fine. I will try the manual download though.

closed 2023-11-13 by dobairoland