espressif/esp-idf #12487: Timed GPIO Interrupt

jrl290, 2023-10-31:
### Is your feature request related to a problem?

I'm looking for a way to grab a timer value upon triggering an interrupt on a GPIO pin

Unfortunately the standard gpio interrupt hooks are not high enough priority to activate on time during other computations

I attempted to write a high priority interrupt in assembly, and it did work, but it also interferes with the standard C gpio interrupt functions (gpio_install_isr_service). And I certainly can't write the rest of my gpio interrupt code in assembly

I'm working with Seeed Studio Xiao ESP32-S3 and ESP32-C3 for my use case, but I barely know enough to program it for the ESP32-S3

Here is the test C code (ESP32-S3):
`/*
 *
	This tests the GPIO interrupt latency using a HLI
	that directly sets the register

*/
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "spi_flash_mmap.h"
#include "driver/gpio.h"
#include "esp_intr_alloc.h"
#include "driver/timer.h"
#include "driver/uart.h"
#include <string.h>
#include "soc/rtc.h"
#include <sys/time.h>
#include <string.h>

#include "esp_log.h"

#include "rom/ets_sys.h"
#include "soc/periph_defs.h"

#include "soc/timer_group_reg.h"

unsigned int * TIMR_ADDR_LO = (unsigned int*)TIMG_T0LO_REG(0); 
unsigned int * TIMR_ADDR_HI = (unsigned int*)TIMG_T0HI_REG(0); 
unsigned int * UPDATE_ADDR = (unsigned int*)TIMG_T0UPDATE_REG(0);

#define GPIO_INPUT_IO_1     7
#define GPIO_INPUT_PIN_SEL1  (1ULL<<GPIO_INPUT_IO_1)

#define INTR_NUM 31
#define CPU_CORE 0

volatile uint64_t timer0_value = 0;
 
static void gpio_task_example(void* arg) {
	while(1) {
		*UPDATE_ADDR = 1;
		uint64_t lastValue = *TIMR_ADDR_LO;
		printf("LOW TIMER ADDR: %X\n", (unsigned int)TIMR_ADDR_LO);
		fprintf(stdout, "GPIO Reg value = %lli - %lli\n", lastValue, timer0_value);
		vTaskDelay(pdMS_TO_TICKS(1000));
	}
}

void app_main() {
    timer_config_t config = {
        .divider = 80,
        .counter_dir = TIMER_COUNT_UP,
        .counter_en = TIMER_START,
        .alarm_en = TIMER_ALARM_DIS,
        .auto_reload = TIMER_AUTORELOAD_DIS,
        .clk_src = TIMER_SRC_CLK_APB
    };
    timer_init(TIMER_GROUP_0, TIMER_0, &config);

	gpio_config_t io_conf;
	//interrupt of rising edge
	io_conf.intr_type = GPIO_INTR_POSEDGE;
	//bit mask of the pins, use GPIO4/5 here
	io_conf.pin_bit_mask = GPIO_INPUT_PIN_SEL1;
	//set as input mode
	io_conf.mode = GPIO_MODE_INPUT;
	//enable pull-up mode
	io_conf.pull_up_en = 0;
	io_conf.pull_down_en = 1;
	ESP_ERROR_CHECK(gpio_config(&io_conf));
	
	gpio_set_level(GPIO_INPUT_PIN_SEL1, 0);

	ESP_INTR_DISABLE(INTR_NUM);
	intr_matrix_set(CPU_CORE, ETS_GPIO_INTR_SOURCE, INTR_NUM);
	ESP_INTR_ENABLE(INTR_NUM);
    // fprintf(stdout, "This tests the interrupt latency, plug in a fungen to pin 17 and observe on pin 4!\n");

    xTaskCreatePinnedToCore(gpio_task_example, "gpio_task_example", 2048, NULL, 10, NULL, 0);

	while(1) {
    	// fprintf(stdout, "This tests the interrupt latency, plug in a fungen to pin 17 and observe on pin 4!\n");
		vTaskDelay(pdMS_TO_TICKS(1000));
	}
}
`

Here is the assembly:
`#include <xtensa/coreasm.h>
#include <xtensa/corebits.h>
#include <xtensa/config/system.h>
#include "freertos/xtensa_context.h"
#include "esp_private/panic_reason.h"
#include "sdkconfig.h"
#include "soc/soc.h"
#include "soc/gpio_reg.h"
#include "soc/dport_reg.h"

#define TIMR_ADDR_LO 0x6001F004
#define TIMR_ADDR_HI 0x6001F008
#define TIMR_ADDR_UPD 0x6001F00c
#define GPIO_PIN 7

#define L5_INTR_STACK_SIZE  12
#define LX_INTR_A15_OFFSET 0
#define LX_INTR_A14_OFFSET 4
#define LX_INTR_A13_OFFSET 8
    .data
_l5_intr_stack:
    .space      L5_INTR_STACK_SIZE

    .section .iram1,"ax"
    .global     xt_highint5
    .type       xt_highint5,@function
    .align      4
xt_highint5:

	movi    a0, _l5_intr_stack // assuming a0 saved restored by EXCSAVE_5 at the end of this
    s32i    a15, a0, LX_INTR_A15_OFFSET
    s32i    a14, a0, LX_INTR_A14_OFFSET
    s32i    a13, a0, LX_INTR_A13_OFFSET

    /* Clear interupt on GPIO17 */
    movi    a14, GPIO_STATUS_W1TC_REG
    movi    a15, (1 << GPIO_PIN)
    s32i    a15, a14, 0
    memw
    
    movi a14, TIMR_ADDR_UPD    //Timer 0 Update address into a10
    s32i a0, a14, 0         //A write instruction copies the timer to the timer registers
    movi a14, timer0_value  //timer0_value address into a11
    movi a15, TIMR_ADDR_LO    //Timer 0 lower 32 bits address into a15
    l32i a13, a15, 0
    s32i a13, a14, 0        //Store lower 32 bits to timer0_value
    movi a15, TIMR_ADDR_HI    //Timer 0 higher 32 bits address into a15
    l32i a13, a15, 0
    s32i a13, a14, 4        //Store lower 32 bits to timer0_value
    memw

   	l32i    a15, a0, LX_INTR_A15_OFFSET
    l32i    a14, a0, LX_INTR_A14_OFFSET
    l32i    a13, a0, LX_INTR_A13_OFFSET

    rsr     a0, EXCSAVE_5 // restore a0
    rfi     5

    .global ld_include_xt_highint5
ld_include_xt_highint5:`

### Describe the solution you'd like.

A high level library that captures a timer upon triggering a gpio interrupt at highest priority

### Describe alternatives you've considered.

I did try using the PWM capture library, but it too returned substantial variation in latency when the CPU is loaded. Also it's not available on the ESP32-C3

suda-morris, 2023-10-31:
> I did try using the PWM capture library, but it too returned substantial variation in latency when the CPU is loaded.

The CPU latency should not affect the **captured** timestamp, because it's done by hardware.

If you have the chance to use esp32c6, it has a new peripheral called "Event Task Matrix", which can also do the same job. We can use any GPIO event to trigger the timer's task (in this case, the task is "capture the timestamp").

jrl290, 2023-10-31:
I shall check my PWM code and get back to you. Thank you for the response

igrr, 2023-11-01:
Also, if you need to capture a PWM waveform, this can be done pretty accurately using the RMT peripheral.

jrl290, 2023-11-08:
PWM is definitely worked as needed with the ESP32S3

I shall check out the RMT peripheral on the ESP32C3 and report back

closed 2023-11-13 by jrl290