espressif/esp-idf #12488: "wifi:m f null" issue occurs when the device is running

Focalcrest-Madi, 2023-10-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

After my ESP32C3 device was running for a period of time, a "wifi:m f null" problem occurred and the WiFi could not connect to the router. After restarting the device, it returned to normal.
My device is using IDF version v4.4.2

AxelLin, 2023-12-17:
https://github.com/espressif/esp-idf/issues/11615#issuecomment-1822030053

hansw123, 2024-03-28:
feel free to reopen @Focalcrest-Madi

closed 2024-04-04 by Indastri