espressif/esp-idf #12489: esp32c3 efuse codeing error

gl-chenxiaosheng, 2023-10-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

IDF version: release v5.1.1
target: esp32c3

I want to write some device infomation in BLOCK3. But it does not work as normal. There some bytes is writed mistake.
So i try again in Block4/Block5/Block6/Block7. And i see the error log as follows:

`[0;32mI (4900) efuse: BURN BLOCK7[0m`
`[0;33mW (4900) efuse: BLOCK7 got a coding error[0m`
`[0;31mE (4900) efuse: BURN BLOCK7 - ERROR (written bits != read bits)[0m`
`[0;33mW (4900) efuse: BLOCK7: next retry to fix an error [1/3]...[0m`
`[0;32mI (4910) efuse: BURN BLOCK7[0m`
`[0;33mW (4910) efuse: BLOCK7 got a coding error[0m`
`[0;31mE (4910) efuse: BURN BLOCK7 - ERROR (written bits != read bits)[0m`
`[0;33mW (4920) efuse: BLOCK7: next retry to fix an error [2/3]...[0m`
`[0;32mI (4930) efuse: BURN BLOCK7[0m`
`[0;33mW (4930) efuse: BLOCK7 got a coding error[0m`
`[0;31mE (4930) efuse: BURN BLOCK7 - ERROR (written bits != read bits)[0m`
`[0;33mW (4940) efuse: BLOCK7: next retry to fix an error [3/3]...[0m`
`[0;33mW (4950) efuse: Coding error was not fixed[0m`
`[0;31mE (4950) efuse: Written data are incorrect[0m`
`[0;32mI (4960) efuse: Batch mode. Prepared fields are committed[0m`
`[1;31m../main/src/gl_producttest.c[506]: esp_efuse_batch_write_commit error[0m`

And I use the espefuse.py to check the efuse summary as follows:
`Error(s) in BLOCK3 [ERRORS:1 FAIL:0]`
`Error(s) in BLOCK4 [ERRORS:1 FAIL:0]`
`Error(s) in BLOCK5 [ERRORS:1 FAIL:1]`
`Error(s) in BLOCK6 [ERRORS:1 FAIL:1]`
`Error(s) in BLOCK7 [ERRORS:1 FAIL:1]`
`BLOCK0          (                ) [0 ] err__regs: 00000000 00000000 00000000 00000000 00000000 00000000`
`EFUSE_RD_RS_ERR0_REG        0x89911100`
`EFUSE_RD_RS_ERR1_REG        0x00000000`
`WARNING: Coding scheme has encoding bit error warnings`

How to resolve this error?

gl-chenxiaosheng, 2023-10-31:
By the way, I have ensured that block4~block7 are empty and have not been written before.

KonstantinKondrashov, 2023-10-31:
Hi @gl-chenxiaosheng!
Could you provide the summary of the whole efuses? Make sure that the efuse BLOCK7 is not write-protected and read-protected.

After a burn operation efuses are read back to compare whether it was burned correctly or not. This logs says that it is not, probably due to write or read protection bits.

gl-chenxiaosheng, 2023-10-31:
Thank to your reply. 

The data that i wan to write is "61cceb487a606433" , "776a3039386239", "9483C4A098B9". And I try it from Block 3 to Block7, the result show the "9483C4A098B9" mistask written to "9483C4A044B9".

There is the the summary of the whole efuses.
```c
espefuse.py v4.7.dev2

Connecting....

Detecting chip type... ESP32-C3

Error(s) in BLOCK3 [ERRORS:1 FAIL:0]

Error(s) in BLOCK4 [ERRORS:1 FAIL:0]

Error(s) in BLOCK5 [ERRORS:1 FAIL:1]

Error(s) in BLOCK6 [ERRORS:1 FAIL:1]

Error(s) in BLOCK7 [ERRORS:1 FAIL:1]

BLOCK0          (                ) [0 ] err__regs: 00000000 00000000 00000000 00000000 00000000 00000000

EFUSE_RD_RS_ERR0_REG        0x89911100

EFUSE_RD_RS_ERR1_REG        0x00000000

=== Run "summary" command ===

EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)

----------------------------------------------------------------------------------------

Calibration fuses:

K_RTC_LDO (BLOCK1)                                 BLOCK1 K_RTC_LDO                                   = 36 R/W (0b0001001)

K_DIG_LDO (BLOCK1)                                 BLOCK1 K_DIG_LDO                                   = 4 R/W (0b0000001)

V_RTC_DBIAS20 (BLOCK1)                             BLOCK1 voltage of rtc dbias20                      = 76 R/W (0x13)

V_DIG_DBIAS20 (BLOCK1)                             BLOCK1 voltage of digital dbias20                  = 16 R/W (0x04)

DIG_DBIAS_HVT (BLOCK1)                             BLOCK1 digital dbias when hvt                      = -12 R/W (0b10011)

THRES_HVT (BLOCK1)                                 BLOCK1 pvt threshold when hvt                      = 1800 R/W (0b0111000010)

TEMP_CALIB (BLOCK2)                                Temperature calibration data                       = -9.3 R/W (0b101011101)

OCODE (BLOCK2)                                     ADC OCode                                          = 104 R/W (0x68)

ADC1_INIT_CODE_ATTEN0 (BLOCK2)                     ADC1 init code at atten0                           = 1544 R/W (0b0110000010)

ADC1_INIT_CODE_ATTEN1 (BLOCK2)                     ADC1 init code at atten1                           = -72 R/W (0b1000010010)

ADC1_INIT_CODE_ATTEN2 (BLOCK2)                     ADC1 init code at atten2                           = -156 R/W (0b1000100111)

ADC1_INIT_CODE_ATTEN3 (BLOCK2)                     ADC1 init code at atten3                           = -616 R/W (0b1010011010)

ADC1_CAL_VOL_ATTEN0 (BLOCK2)                       ADC1 calibration voltage at atten0                 = -276 R/W (0b1001000101)

ADC1_CAL_VOL_ATTEN1 (BLOCK2)                       ADC1 calibration voltage at atten1                 = -36 R/W (0b1000001001)

ADC1_CAL_VOL_ATTEN2 (BLOCK2)                       ADC1 calibration voltage at atten2                 = -228 R/W (0b1000111001)

ADC1_CAL_VOL_ATTEN3 (BLOCK2)                       ADC1 calibration voltage at atten3                 = -332 R/W (0b1001010011)

Config fuses:

[... 231 lines cut ...]

Wdt fuses:

WDT_DELAY_SEL (BLOCK0)                             RTC watchdog timeout threshold; in unit of slow cl = 40000 R/W (0b00)

                                                   ock cycle                                         

Error(s) in BLOCK3 [ERRORS:1 FAIL:0]

Error(s) in BLOCK4 [ERRORS:1 FAIL:0]

Error(s) in BLOCK5 [ERRORS:1 FAIL:1]

Error(s) in BLOCK6 [ERRORS:1 FAIL:1]

Error(s) in BLOCK7 [ERRORS:1 FAIL:1]

BLOCK0          (                ) [0 ] err__regs: 00000000 00000000 00000000 00000000 00000000 00000000

EFUSE_RD_RS_ERR0_REG        0x89911100

EFUSE_RD_RS_ERR1_REG        0x00000000

WARNING: Coding scheme has encoding bit error warnings
```

KonstantinKondrashov, 2023-10-31:
@gl-chenxiaosheng
It makes the picture more or less clear for me. I would like to take a look at how you burn these data  ("61cceb487a606433" , "776a3039386239", "9483C4A098B9"), separately (3 burn operations) or by one batch.
Efuse Blocks (1-max) have the RS coding scheme. This means that in addition to the user data, encoded data is also burned. If you burn separately then the following burn operation breaks the coding data. These coding data are used by the efuse controller to recover user data if an error is detected according to the RS algorithm, but due to it being broken then unexpected wrong data can occur.

You need to use these APIs https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/efuse.html#_CPPv427esp_efuse_batch_write_beginv. See the below example to have a look how to burn efuse data into one efuse block at once.

```c
esp_efuse_batch_write_begin();
  esp_efuse_write_field_blob(ESP_EFUSE_...);
  esp_efuse_write_block(EFUSE_BLKx, ...);
  esp_efuse_write(ESP_EFUSE_1, 3); 
esp_efuse_batch_write_commit();
```

gl-chenxiaosheng, 2023-10-31:
I was writed the  ("61cceb487a606433" , "776a3039386239", "9483C4A098B9") by one batch. Like as follows:
```c
    esp_efuse_batch_write_begin();

    // sn
    esp_efuse_write_field_blob(ESP_EFUSE_USER_DATA_SN_64, value, (value_len * 8));
    vTaskDelay(10 / portTICK_PERIOD_MS);

    // ddns
    esp_efuse_write_field_blob(ESP_EFUSE_USER_DATA_DDNS_56, value, (value_len * 8));
    vTaskDelay(10 / portTICK_PERIOD_MS);
    
    // basic mac
    esp_efuse_write_field_blob(ESP_EFUSE_USER_DATA_MAC_CUSTOM, value, (value_len * 8));
    vTaskDelay(10 / portTICK_PERIOD_MS);
    
    esp_efuse_batch_write_commit();
```
I know  RS algorithm will also burned the encoded data. But it is only on one block, right? So i try these data to other block, like block4 to block7 which can be user burned.
So i more confuse about it./(ㄒoㄒ)/~~

KonstantinKondrashov, 2023-10-31:
@gl-chenxiaosheng 
I found out why this happened. I will provide the fix for a test.

KonstantinKondrashov, 2023-10-31:
@gl-chenxiaosheng
I am able to reproduce it. The issue is in `hal_memcpy()` func, it is used inside efuse API. It left the RS coding registers = 0. You can use this path if it is urgent. I am going to update the master and v5.1 as well. Other releases do not have this issue.
```diff
diff --git a/components/hal/platform_port/include/hal/misc.h b/components/hal/platform_port/include/hal/misc.h
index c6e50ad..ab24c1e 100644
--- a/components/hal/platform_port/include/hal/misc.h
+++ b/components/hal/platform_port/include/hal/misc.h
@@ -66,11 +66,7 @@ extern "C" {
  * @param len The number of bytes to be copied
  * @return a pointer to destination
  */
-__attribute__((always_inline)) static inline void *hal_memcpy(void *dst_mem, const void *src_mem, size_t len)
-{
-    asm("" : "+r"(dst_mem), "+r"(src_mem));
-    return memcpy(dst_mem, src_mem, len);
-}
+#define hal_memcpy(dst_mem, src_mem, len) (__extension__({memcpy(dst_mem, src_mem, len);}))
 
 /**
  * @brief Sets the first num bytes of the block of memory pointed by ptr to the specified value
@@ -82,11 +78,7 @@ __attribute__((always_inline)) static inline void *hal_memcpy(void *dst_mem, con
  * @param len The number of bytes to be copied
  * @return a pointer to the memory area
  */
-__attribute__((always_inline)) static inline void *hal_memset(void *dst_mem, int value, size_t len)
-{
-    asm("" : "+r"(dst_mem));
-    return memset(dst_mem, value, len);
-}
+#define hal_memset(dst_mem, value, len) (__extension__({memset(dst_mem, value, len);}))
 
 #ifdef __cplusplus
 }
```

gl-chenxiaosheng, 2023-11-01:
It works! 
Thank you very much for your help. I will test it a while. Make sure it not other problems, and then close the issues.

KonstantinKondrashov, 2023-11-01:
If somebody else is faced with the same issue, or similar to that an efuse BLOCK has a coding error. You can try to recover it by using -> **espefuse.py  check_error --recovery**

closed 2023-11-07 by espressif-bot