espressif/esp-idf #12491: nimble compile error

AxelLin, 2023-10-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-3775-gb4268c874a

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

build success.

### What is the actual behavior?

/home/axel/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c: In function 'ble_gap_master_connect_reattempt':
/home/axel/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:1036:40: error: storing the address of local variable 'conn' in 'ble_conn_reattempt[idx].cb_arg' [-Werror=dangling-pointer=]
 1036 |         ble_conn_reattempt[idx].cb_arg = &conn;
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
/home/axel/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:992:30: note: 'conn' declared here
  992 |     struct ble_gap_conn_desc conn;
      |                              ^~~~
/home/axel/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:130:46: note: 'ble_conn_reattempt' declared here
  130 | static struct ble_gap_connect_reattempt_ctxt ble_conn_reattempt[MYNEWT_VAL(BLE_MAX_CONNECTIONS)];
      | 

### Steps to reproduce.

Just build with nimble for esp32c3.

rahult-github, 2023-10-31:
Hi @AxelLin ,

May i know which example you attempting to compile ? We tried both peripheral and central and don't see the issue. Also enabled / disabled the ENABLE_CONN_REATTEMPT flag, but don't see the issue.

The code being pointed is not modifed since a year. Is there any specific GCC version being used for this ?

AxelLin, 2023-10-31:
@rahult-github 
I compile with our application code, not example code.
However, the error message does make sense, I think this needs fix.
/home/axel/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:1036:40: error: storing the address of local variable 'conn' in 'ble_conn_reattempt[idx].cb_arg' [-Werror=dangling-pointer=]
1036 | ble_conn_reattempt[idx].cb_arg = &conn;

AxelLin, 2023-10-31:
BTW, I didn't hit this issue when I build with v5.1 branch. 
And the ble_gap.c  file is exactly the same in v5.1 branch and master.
Maybe this is toolchain issue?

KaeLL, 2023-10-31:
Seems to be an upstream code issue. `cb_arg` does become a dangling reference after `ble_gap_master_connect_reattempt` exits.

rahult-github, 2023-11-01:
Hi @AxelLin ,  

Not able to hit the issue at my end yet. Will investigate on the toolchain input provided.  

In meantime, can you check if attached patch helps ? 
[dangling_pointer_error.txt](https://github.com/espressif/esp-idf/files/13223318/dangling_pointer_error.txt)

AxelLin, 2023-11-01:
Hi @rahult-github 
It's strange that I no longer hit the compile error today (even without your path).
I'm closing the issue, sorry for the noise.

AxelLin, 2023-11-07:
> Hi @AxelLin ,
> 
> Not able to hit the issue at my end yet. Will investigate on the toolchain input provided.
> 
> In meantime, can you check if attached patch helps ? [dangling_pointer_error.txt](https://github.com/espressif/esp-idf/files/13223318/dangling_pointer_error.txt)

Hi @rahult-github 
I hit this compile error again, and confirm your patch fixed the compile issue.

rahult-github, 2023-11-07:
Hi @AxelLin , noted . Will raise fix internally.

AxelLin, 2023-11-21:
> Hi @AxelLin , noted . Will raise fix internally.

I reopen this issue because I still cannot find this fix in esp-nimble.