espressif/esp-idf #12493: open_memstream [used by esp-console] fails when CONFIG_HEAP_TASK_TRACKING: assert failed: insert_free_block tlsf.c:358 (current && "free list cannot have a null entry")

alonbl, 2023-10-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-3775-gb4268c874a

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

Provided the following program:

```c
#include <stdio.h>

void app_main(void)
{
        char *ptr = NULL;
        size_t size = 0;
        FILE *fp = open_memstream(&ptr, &size);
        if (fp != NULL) {
                fclose(fp);
        }
}
```

And the following config:

```ini
CONFIG_HEAP_TASK_TRACKING=y
```

Legit code should run properly and not crash.

### What is the actual behavior?

Crash as per the following traceback:

```
I (303) main_task: Started on CPU0
I (313) main_task: Calling app_main()

assert failed: insert_free_block tlsf.c:358 (current && "free list cannot have a null entry")
HINT: CORRUPT HEAP: heap metadata corrupted resulting in TLSF malfunction.
Make sure you are not making out of bound writing on the memory you allocate in your application.
Make sure you are not writing on freed memory.
For more information run 'idf.py docs -sp api-reference/system/heap_debug.html'.

Backtrace: 0x4037585a:0x3fc983e0 0x40379a5d:0x3fc98400 0x4037f8b9:0x3fc98420 0x4037f596:0x3fc98540 0x4037e7ee:0x3fc98570 0x403760e8:0x3fc98590 0x40376158:0x3fc985c0 0x4037618f:0x3fc985e0 0x4037f911:0x3fc98600 0x4200cee1:0x3fc98620 0x4200bf0a:0x3fc98640 0x4200bf8b:0x3fc98670 0x4200803d:0x3fc98690 0x42017f97:0x3fc986c0 0x4037a2c1:0x3fc986f0
0x4037585a: panic_abort at /home/alonbl/exodigo/esp-idf/components/esp_system/panic.c:472

0x40379a5d: esp_system_abort at /home/alonbl/exodigo/esp-idf/components/esp_system/port/esp_system_chip.c:93

0x4037f8b9: __assert_func at /home/alonbl/exodigo/esp-idf/components/newlib/assert.c:81

0x4037f596: insert_free_block at /home/alonbl/exodigo/esp-idf/components/heap/tlsf/tlsf.c:358 (discriminator 1)
 (inlined by) block_insert at /home/alonbl/exodigo/esp-idf/components/heap/tlsf/tlsf.c:388 (discriminator 1)
 (inlined by) block_trim_used at /home/alonbl/exodigo/esp-idf/components/heap/tlsf/tlsf.c:517 (discriminator 1)
 (inlined by) tlsf_realloc at /home/alonbl/exodigo/esp-idf/components/heap/tlsf/tlsf.c:1197 (discriminator 1)

0x4037e7ee: multi_heap_realloc_impl at /home/alonbl/exodigo/esp-idf/components/heap/multi_heap.c:245

0x403760e8: heap_caps_realloc_base at /home/alonbl/exodigo/esp-idf/components/heap/heap_caps.c:445

0x40376158: heap_caps_realloc at /home/alonbl/exodigo/esp-idf/components/heap/heap_caps.c:479

0x4037618f: heap_caps_realloc_default at /home/alonbl/exodigo/esp-idf/components/heap/heap_caps.c:259

0x4037f911: _realloc_r at /home/alonbl/exodigo/esp-idf/components/newlib/heap.c:54

0x4200cee1: memcloser at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/open_memstream.c:282 (discriminator 4)

0x4200bf0a: _fclose_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fclose.c:96 (discriminator 1)

0x4200bf8b: fclose at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fclose.c:125 (discriminator 1)

0x4200803d: app_main at /home/alonbl/exodigo/esp-idf/examples/system/console/basic/main/main.c:9

0x42017f97: main_task at /home/alonbl/exodigo/esp-idf/components/freertos/app_startup.c:208

0x4037a2c1: vPortTaskWrapper at /home/alonbl/exodigo/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:137
```

### Steps to reproduce.

Attached [1] a trivial example project.
Use:

```
idf.py set-target esp32s3
idf.py build
```

[1] [example.zip](https://github.com/espressif/esp-idf/files/13214405/example.zip)

### More Information.

Turning off `CONFIG_HEAP_TASK_TRACKING` makes it work, so either the problem is in the task tracing or in the `open_memstream()`, I guess it is in the task tracing as it does work properly.

It is important as `components/console/commands.c::esp_console_cmd_register()` is using `open_memstream()` so any application with console support cannot be run with `CONFIG_HEAP_TASK_TRACKING`.

igrr, 2023-10-31:
This is probably the same issue as https://github.com/espressif/esp-idf/issues/12498, given that there's a `realloc` in the backtrace.

alonbl, 2023-11-01:
Thanks!

Closed as dup of #12498

closed 2023-11-01 by alonbl