espressif/esp-idf #12494: I2C Problems while using BLE

thomasweichselbaumer, 2023-10-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

There is a problem reading out RTC values via i2c while heavily using BLE. 

It's a critical bug for our system that we can not trust i2c readouts while using BLE. We use the stored time in an external RTC to build some security features on it. On the other hand BLE is a key feature of our product.

Maybe the issue #7781 is related with this bug, but we couldn't find any helpfule solution for it.

We discovered the bug first in 4.4.1 on an ESP32 WROVER-E.

WatDavid, 2023-11-02:
With IDF Version 4.3 we could not reproduce this bug. But its no solution to make an idf downgrade to 4.3.

alutov, 2023-12-04:
Try run menuconfig and uncheck High level interrupt option in Bluetooth controller.

kscheff, 2023-12-09:
When running ESP32-C3 with framework-espidf @ 3.50101.230828 (5.1.1) I get also i2c issues during ble traffic. In my case I am utilising the i2c bus for reading out a memory device during a ble read event. There is also a timer running in background exercising i2c traffic. The i2c traffic called from within the ble read event works fine, it is always the i2c traffic generated from the timer event which does get corrupted (read or write fails). I am using the old style i2c interface i2c_master_cmd_begin().

ginkgm, 2024-02-22:
Issue closed. If you still have problem feel free to reopen it.

closed 2024-02-22 by ginkgm