espressif/esp-idf #12497: heap_caps_check_integrity_all() crashes when SPIRAM is enabled and comperhensive heap posioning is enabled

alonbl, 2023-10-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-3775-gb4268c874a

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

Provided the following trivial code:

```c
#include <stdbool.h>

#include <esp_heap_caps_init.h>

void app_main(void)
{
        heap_caps_check_integrity_all(true);
}
```

And the following configuration:

```ini
CONFIG_HEAP_POISONING_COMPREHENSIVE=y
CONFIG_SPIRAM=y
CONFIG_SPIRAM_MODE_OCT=y
```

`heap_caps_check_integrity_all()` should not crash.

### What is the actual behavior?

Crash with the the following traceback:

```
I (1542) main_task: Started on CPU0
I (1552) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1552) main_task: Calling app_main()
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x4037f3a4  PS      : 0x00060c34  A0      : 0x8037f1ac  A1      : 0x3fc99140  
0x4037f3a4: verify_fill_pattern at /home/alonbl/exodigo/esp-idf/components/heap/multi_heap_poisoning.c:151

A2      : 0x3c66f900  A3      : 0x001c06f8  A4      : 0x00000001  A5      : 0xfefefefe  
A6      : 0xcececece  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3fc99230  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000001  A13     : 0xfefefefe  
A14     : 0x80000000  A15     : 0x3fc95bb4  SAR     : 0x0000001b  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffd  
0x400556d5: strlen in ROM

0x400556e5: strlen in ROM

Backtrace: 0x4037f3a1:0x3fc99140 0x4037f1a9:0x3fc99160 0x42018663:0x3fc99180 0x42018999:0x3fc991a0 0x42018ad7:0x3fc991c0 0x42018b1b:0x3fc991e0 0x420186af:0x3fc99210 0x42002ecb:0x3fc99230 0x42002ef1:0x3fc99250 0x420086fd:0x3fc99270 0x4201855f:0x3fc99290 0x4037ac6d:0x3fc992c0
0x4037f3a1: verify_fill_pattern at /home/alonbl/exodigo/esp-idf/components/heap/multi_heap_poisoning.c:145

0x4037f1a9: multi_heap_internal_check_block_poisoning at /home/alonbl/exodigo/esp-idf/components/heap/multi_heap_poisoning.c:416

0x42018663: tlsf_check_hook at /home/alonbl/exodigo/esp-idf/components/heap/multi_heap.c:319

0x42018999: integrity_walker at /home/alonbl/exodigo/esp-idf/components/heap/tlsf/tlsf.c:689

0x42018ad7: tlsf_walk_pool at /home/alonbl/exodigo/esp-idf/components/heap/tlsf/tlsf.c:767

0x42018b1b: tlsf_check_pool at /home/alonbl/exodigo/esp-idf/components/heap/tlsf/tlsf.c:791

0x420186af: multi_heap_check at /home/alonbl/exodigo/esp-idf/components/heap/multi_heap.c:340 (discriminator 1)

0x42002ecb: heap_caps_check_integrity at /home/alonbl/exodigo/esp-idf/components/heap/heap_caps.c:617

0x42002ef1: heap_caps_check_integrity_all at /home/alonbl/exodigo/esp-idf/components/heap/heap_caps.c:626

0x420086fd: app_main at /home/alonbl/exodigo/esp-idf/examples/system/console/basic/main/main.c:8

0x4201855f: main_task at /home/alonbl/exodigo/esp-idf/components/freertos/app_startup.c:208

0x4037ac6d: vPortTaskWrapper at /home/alonbl/exodigo/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:137

Core  1 register dump:
PC      : 0x40378502  PS      : 0x00060734  A0      : 0x82002bfd  A1      : 0x3fc9a140  
0x40378502: esp_cpu_wait_for_intr at /home/alonbl/exodigo/esp-idf/components/esp_hw_support/cpu.c:145

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3fc96f90  A5      : 0x3fc96f70  
A6      : 0x40378284  A7      : 0x00000001  A8      : 0x8200adbe  A9      : 0x3fc9a100  
0x40378284: ipc_task at /home/alonbl/exodigo/esp-idf/components/esp_system/esp_ipc.c:48

A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3fc96f70  A13     : 0x3fc96f40  
A14     : 0x00000001  A15     : 0x3fc9a31c  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x403784ff:0x3fc9a140 0x42002bfa:0x3fc9a160 0x4037ba39:0x3fc9a180 0x4037ac6d:0x3fc9a1a0
0x403784ff: xt_utils_wait_for_intr at /home/alonbl/exodigo/esp-idf/components/xtensa/include/xt_utils.h:81
 (inlined by) esp_cpu_wait_for_intr at /home/alonbl/exodigo/esp-idf/components/esp_hw_support/cpu.c:132

0x42002bfa: esp_vApplicationIdleHook at /home/alonbl/exodigo/esp-idf/components/esp_system/freertos_hooks.c:59

0x4037ba39: prvIdleTask at /home/alonbl/exodigo/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4257 (discriminator 1)

0x4037ac6d: vPortTaskWrapper at /home/alonbl/exodigo/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:137
```

### Steps to reproduce.

Attached[1] is an example application.

```
idf.py set-target esp32s3
idf.py build
```

[1] [example.zip](https://github.com/espressif/esp-idf/files/13216033/example.zip)

### More Information.

Please help us to debug heap while external SPIRAM is enabled.

SoucheSouche, 2023-11-01:
Hi @alonbl,

This happens because when comprehensive poisoning is enabled, `heap_caps_check_integrity_all` function is checking all free bytes of the memory. In the case where SPIRAM is enabled, this means that all free bytes of the external memory are checked and since this memory region is fairly big, it takes quite a while to check. For this reason, the interrupt watchdog is triggered and your application crashes as you can see in your log.

To prevent this from happening, you can increase the interrupt watchdog timeout, by setting a different value of ESP_INT_WDT_TIMEOUT_MS set by default to 300ms in the menuconfig  `Component config → ESP System Settings → Interrupt watchdog timeout (ms)`.

alonbl, 2023-11-01:
Hello @SoucheSouche,
Thank you so much for the quick response.
I would expect that if we know there is an issue with timing, the watchdog will be reset by task or task will give up the CPU in each iteration, I do not think that global change of the watchdog setting is valid in this case.
Closing for now, please consider an implementation that will be compatible with watchdog as any other application that implements an algorithm which takes time.
Regards,

closed 2023-11-01 by alonbl

igrr, 2023-11-01:
> or task will give up the CPU in each iteration

In the specific case of checking heap integrity, we have to disable interrupts (and therefore preemption) to get an "atomic" snapshot of the heap state. So it's pretty difficult to "pause" the integrity check.

As for feeding the watchdog, it could be done, but then the "heap" component has to gain knowledge about the watchdog timers, which we would like to avoid as it increases coupling between components.

alonbl, 2023-11-02:
Hello @igrr,

I understand the atomic requirement I also understand the separation of concerns pattern, however, let's assume we have a long-running application logic which should be atomic, this is the super case of this discussion, what would we do?

I could argue that separation of concern is not followed by the `heap_caps_check_integrity_all()` as it is aware of the console :)

Giving up atomic - not an option, so we either disable interrupts or disable all other tasks, this is intrusive.
We will probably refresh the watchdog manually.

What do you think of adding a callback to `heap_caps_check_integrity_all()` to allow user to implement his logic?

Option#1: use idle callback between lines:

```c
void heap_caps_check_integrity_all_with_idle(bool print_errors, void (*idle)(void));
```

Option#2: interact with console via callback - aka separation of concerns of watchdog + console:

```c
void heap_caps_check_integrity_all_stream(bool print_errors, void (*puts_func)(const char *s));
```

Thanks,

o-marshmallow, 2023-11-02:
@alonbl Actually you can already override the implementation of `heap_caps_check_integrity_all` thanks to the linker and `--wrap` option. In that case, you can have something like:
```
extern bool __real_heap_caps_check_integrity_all(bool print_errors);

bool __wrap_heap_caps_check_integrity_all(bool print_errors)
{
    // Your logic here
    bool ret = __real_heap_caps_check_integrity_all(print_errors); 
    // Your logic here
    return ret;
}
```

Check this example which wraps/overrides `esp_restart`: https://github.com/espressif/esp-idf/tree/master/examples/build_system/wrappers

alonbl, 2023-11-02:
@o-marshmallow ,

Thank you for the suggestion, however, the problem is that the `heap_caps_check_integrity_all()` loops "forever", we should reset the watchdog within its loop, so wrapper will not help.

Regards,

o-marshmallow, 2023-11-02:
@alonbl One solution with the wrapper is to disable the interrupt watchdog before calling `heap_caps_check_integrity_all` and resume/restart it afterwards .

alonbl, 2023-11-02:
@o-marshmallow, @igrr wrote that interrupts are already disabled.

> In the specific case of checking heap integrity, we have to disable interrupts (and therefore preemption) to get an "atomic" snapshot of the heap state. So it's pretty difficult to "pause" the integrity check.

Usually watchdog resets the device, maybe in esp32 watchdog is a special interrupt, no need to add a wrapper for this, if it works I would just to disable the interrupts before calling it and enable them.

Thanks for the suggestion!

igrr, 2023-11-02:
> maybe in esp32 watchdog is a special interrupt

Interrupt watchdog in IDF uses a level 4 interrupt, whereas the RTOS critical sections (portENTER/EXIT_CRITICAL) disable interrupts up to and including level 3.

> I could argue that separation of concern is not followed by the heap_caps_check_integrity_all() as it is aware of the console :)

That's true. Yes, we could consider injecting the dependency on interrupt watchdog into heap component somehow. After all, `heap` and `esp_system` are at the same level/group of components — [G1](https://github.com/espressif/esp-idf/tree/master/components#g1-components).

alonbl, 2023-11-02:
@igrr Thank You!

May I open a feature request for "Disable level 4 interrupts (watchdog) during heap dump" or change the subject of this one and reopen?

igrr, 2023-11-02:
I don't think this disabling level 4 interrupts would help. The watchdog would raise the interrupt anyway, it's just that the CPU wouldn't service the interrupts until they are re-enabled. So the result would roughly the same, just the watchdog would trigger whenever heap_caps_check_integrity_all completes.

Besides, the interrupt watchdog is an important system-level check, I don't think it's safe to disable it — if heap_caps_check_integrity_all gets genuinely stuck somewhere, this shouldn't cause the system to hang.

"Feed the interrupt watchdog during heap_caps_check_integrity_all execution" might be a more reasonable request. However, since heap_caps_check_integrity_all is a function used for debugging purposes, and the solution to the problem is already very straightforward (increase CONFIG_ESP_INT_WDT_TIMEOUT_MS), I don't think implementing this feature request would be very high on the priority list...