espressif/esp-idf #12498: realloc() crashes when CONFIG_HEAP_TASK_TRACKING enabled: current && "free list cannot have a null entry"

alonbl, 2023-10-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-3775-gb4268c874a

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

Provided this trivial code:

```c
#include <stdlib.h>
#include <stdio.h>

void app_main(void)
{
        printf("Start\n");
        void *p1 = malloc(1024);
        printf("p1=%p\n", p1);
        void *p2 = malloc(1024);
        printf("p2=%p\n", p2);
        p2 = realloc(p1, 1025);
        printf("p2=%p\n", p2);
}
```

And the following configuration:

```ini
CONFIG_HEAP_TASK_TRACKING=y
```

Realloc should not crash.

### What is the actual behavior?

Realloc crashes with the following backtrace:

```
I (303) main_task: Started on CPU0
I (313) main_task: Calling app_main()
Start
p1=0x3fc99a2c
p2=0x3fc99e34

assert failed: insert_free_block tlsf.c:358 (current && "free list cannot have a null entry")
HINT: CORRUPT HEAP: heap metadata corrupted resulting in TLSF malfunction.
Make sure you are not making out of bound writing on the memory you allocate in your application.
Make sure you are not writing on freed memory.
For more information run 'idf.py docs -sp api-reference/system/heap_debug.html'.

Backtrace: 0x4037585a:0x3fc98460 0x40379a5d:0x3fc98480 0x4037f8b9:0x3fc984a0 0x4037f596:0x3fc985c0 0x4037e7ee:0x3fc985f0 0x403760e8:0x3fc98610 0x40376158:0x3fc98640 0x4037618f:0x3fc98660 0x4037f8d9:0x3fc98680 0x4200805d:0x3fc986a0 0x42017d33:0x3fc986c0 0x4037a2c1:0x3fc986f0
0x4037585a: panic_abort at /home/alonbl/esp-idf/components/esp_system/panic.c:472

0x40379a5d: esp_system_abort at /home/alonbl/esp-idf/components/esp_system/port/esp_system_chip.c:93

0x4037f8b9: __assert_func at /home/alonbl/esp-idf/components/newlib/assert.c:81

0x4037f596: insert_free_block at /home/alonbl/esp-idf/components/heap/tlsf/tlsf.c:358 (discriminator 1)
 (inlined by) block_insert at /home/alonbl/esp-idf/components/heap/tlsf/tlsf.c:388 (discriminator 1)
 (inlined by) block_trim_used at /home/alonbl/esp-idf/components/heap/tlsf/tlsf.c:517 (discriminator 1)
 (inlined by) tlsf_realloc at /home/alonbl/esp-idf/components/heap/tlsf/tlsf.c:1197 (discriminator 1)

0x4037e7ee: multi_heap_realloc_impl at /home/alonbl/esp-idf/components/heap/multi_heap.c:245

0x403760e8: heap_caps_realloc_base at /home/alonbl/esp-idf/components/heap/heap_caps.c:445

0x40376158: heap_caps_realloc at /home/alonbl/esp-idf/components/heap/heap_caps.c:479

0x4037618f: heap_caps_realloc_default at /home/alonbl/esp-idf/components/heap/heap_caps.c:259

0x4037f8d9: realloc at /home/alonbl/esp-idf/components/newlib/heap.c:34

0x4200805d: app_main at /home/alonbl/esp-idf/examples/system/console/basic/main/main.c:11

0x42017d33: main_task at /home/alonbl/esp-idf/components/freertos/app_startup.c:208

0x4037a2c1: vPortTaskWrapper at /home/alonbl/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:137
```

### Steps to reproduce.

Attached an example:

```
idf.py set-target esp32s3
idf.py build
```

[1] [example.zip](https://github.com/espressif/esp-idf/files/13216356/example.zip)

o-marshmallow, 2023-10-31:
Hello @alonbl ,

This issue is indeed due to a prior commit merged on our side, it has been introduced with commit  7492c862af90d8a8d9920f30b3f652287ef12436. The good thing is that a fix is currently being reviewed and will be merged soon.

SoucheSouche, 2023-11-10:
The fix was merged internally. It will be available with the next GitHub sync.

closed 2023-11-10 by SoucheSouche