espressif/esp-idf #12499: Please make panic_print_char weak

kriegste, 2023-10-31:
### Is your feature request related to a problem?

We use esp_log_set_vprintf to redirect log output into a (RTC) memory (ring) buffer. This memory log then can be read out using the HTTP server. However, not all the output that normally ends up in the console can be redirected. Even esp_rom_install_channel_putc does not help saving the core dump during a panic.

### Describe the solution you'd like.

By making panic_print_char weak, i.e.
> __attribute__((weak)) void panic_print_char(const char c)
in panic.c, we can implement our own panic_print_char function to resolve this problem.

### Describe alternatives you've considered.

There are several config options such as CONFIG_ESP_CONSOLE_UART, CONFIG_ESP_CONSOLE_USB_CDC, CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG and CONFIG_ESP_CONSOLE_SECONDARY_USB_SERIAL_JTAG to redirect panic log output. But all of them rely on a wired connection. There is currently no wireless solution.

o-marshmallow, 2023-11-01:
Hello @kriegste ,

Actually, it is possible to "override" the `panic_print_char` function by wrapping it with the linker, you will then be able to use your own implementation of it. This is the simplest solution, more details about it in this comment: https://github.com/espressif/esp-idf/issues/7681#issuecomment-941957783

If you don't want to wrap the function, you can still mark it as weak by forcing the inclusion of a header file at build time. More details in this comment: https://github.com/espressif/esp-idf/issues/7681#issuecomment-1168424015

kriegste, 2023-11-02:
Thanks. I'm currently trying the option using a weak.h file. Waiting for the next crash.

kriegste, 2023-11-02:
It's working. :)

o-marshmallow, 2023-11-03:
@kriegste Glad to hear it's working, closing this issue then

closed 2023-11-03 by o-marshmallow