espressif/esp-idf #12502: cc1.exe: some warnings being treated as errors

baldhead69, 2023-10-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I updated the esp-idf version from v5.1 to v5.2-dev-3775-gb4268c874a, and now i have this problem at compilation.

With version v5.1 all was ok.

How to solve this ?

NOTE:
I have a mdns component installed in my esp-idf project, espressif/mdns: "^1.0.7". 
[compilation_error.txt](https://github.com/espressif/esp-idf/files/13221982/compilation_error.txt)

Thanks.

baldhead69, 2023-10-31:
I updated the mdns component installed in my esp-idf project to espressif/mdns: "^1.2.1" and same problem.

igrr, 2023-11-01:
Could you please attach your sdkconfig file?

baldhead69, 2023-11-02:
> Could you please attach your sdkconfig file?

sdkconfig file attached

baldhead69, 2023-11-06:
@igrr ,

Some suggestion ?

igrr, 2023-11-06:
@baldhead69 I have commented here asking for the sdkconfig simply to make sure that there is enough information in the issue report for the developers to troubleshoot this issue. The issue is now assigned to @espressif-abhikroy, he will reply to you.

(Upon a brief look, I think the issue happens because you have a bunch of `CONFIG_XYZ_DEBUG` options enabled, and on our side we haven't tested building with all these options in CI. One of these options, when enabled, might be adding code which can't be compiled successfully due to the mismatch in format string specifiers. Given that the errors originate from ip6.c, my guess is that disabling CONFIG_LWIP_IP6_DEBUG could be a workaround until the issue is fixed.)

baldhead69, 2023-11-07:
> CONFIG_LWIP_XYZ_DEBUG

@igrr,

Yes, i disabled all LWIP Debug and  included #include "esp_log.h" in one c file my and it worked, ie, compiled.

Thanks.

espressif-abhikroy, 2023-11-08:
@baldhead69, it appears that there are some string formatting errors in ip6.c and ip4_napt.c. The fix for these issues is scheduled to be merged in lwip-2.1.3 in the near future. Until then, disabling LWIP_DEBUG can serve as a temporary workaround.

closed 2024-01-10 by espressif-abhikroy