espressif/esp-idf #12503: USB slave continuous restart issue

Firmware-beginner, 2023-11-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm using TI's MSP430F5XXX microcontroller(USB slave) with inbuilt USB stack and TI's custom VID PID,
USB slave is connected to ESP32S3(USB host). I'm using ESP-IDF 5.1 CMD for build, flash and monitor.
I'm facing issues opening the slave device using usb host cdc_acm_vcp example, the issue is that the
device keeps on restarting. I am not able to understand why device is restarting.

roma-jam, 2023-11-01:
@Firmware-beginner ,
Thanks for the reporting! 

To have a better chance to help, is it possible to attach the log from example? 
If you can provide the USB Descriptor of connected device that also will be very helpful. 

Does the general `usb_host_lib` example acts the same way?

tore-espressif, 2023-11-07:
Hi @Firmware-beginner , is this the same issue as you reported here https://github.com/espressif/esp-idf/issues/12492 ?

We will close that duplicate issue and continue tracking the problem in this issue

roma-jam, 2023-11-07:
Hi @Firmware-beginner, 
If the log is valid from #12492, it seems that the problem is during the `cdc_acm_transfers_allocate`.

To be able to proceed, could you please share the USB Descriptor of your device, which is based on MSP430F5XXX? 

You can use `usb_host_lib` example to get the USB Descriptor of the device.

Firmware-beginner, 2023-11-08:
The general usb_host_lib example does not act the same way

USB Descriptor of MSP430F5XXX are shared below

I (323) main_task: Calling app_main()
I (323) DAEMON: Installing USB Host Library
I (363) CLASS: Registering Client
I (843) CLASS: Opening device at address 1
I (843) CLASS: Getting device information
I (843) CLASS:  Full speed
I (843) CLASS:  bConfigurationValue 1
I (853) CLASS: Getting device descriptor
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0x2
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 8
idVendor 0x2047
idProduct 0x403
bcdDevice 2.00
iManufacturer 1
iProduct 2
iSerialNumber 3
bNumConfigurations 1
I (873) CLASS: Getting config descriptor
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 67
bNumInterfaces 2
bConfigurationValue 1
iConfiguration 4
bmAttributes 0x80
bMaxPower 100mA
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0x2
        bInterfaceSubClass 0x2
        bInterfaceProtocol 0x1
        iInterface 5
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x81   EP 1 IN
                bmAttributes 0x3        INT
                wMaxPacketSize 64
                bInterval 255
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 1
        bAlternateSetting 0
        bNumEndpoints 2
        bInterfaceClass 0xa
        bInterfaceSubClass 0x0
        bInterfaceProtocol 0x0
        iInterface 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x2    EP 2 OUT
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 255
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x82   EP 2 IN
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 255
I (973) CLASS: Getting Manufacturer string descriptor
Texas Instruments<
I (983) CLASS: Getting Product string descriptor
F5529LP simpleUsbBackchannel
I (993) CLASS: Getting Serial Number string descriptor
835F994609002000

tore-espressif, 2023-11-21:
@Firmware-beginner  the cdc_acm_vcp example is intended for CDC-like devices, that implement vendor specific class. Like the CP2102. Here we are trying to show how you can override default CDC functions with vendor specific ones.

Your device seems to be CDC compatible so you should work with the CDC-ACM driver

If you want to experiment with the original CDC-ACM driver, this example might work better for you https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/cdc/cdc_acm_host , just update the VID and PID at the beginning of main.c

tore-espressif, 2023-11-21:
We released version 2.0.2 of [usb_host_cdc_acm](https://components.espressif.com/components/espressif/usb_host_cdc_acm) that will not crash on this condition.

However the device will not be opened. Please try the [cdc_acm_host example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/cdc/cdc_acm_host). Modify VID and PID in the beginning of the main.c file, it should work OK with your USB device.

tore-espressif, 2023-12-05:
Feel free to reopen if the problem persists

closed 2023-12-05 by tore-espressif