espressif/esp-idf #12505: occasionally storage overflow when using FatFs

AliMoal, 2023-11-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-32E

### Power Supply used.

External 3.3V

### What is the expected behavior?

Read and write on FatFs filesystem without any problem.

### What is the actual behavior?

Suddenly sometimes the storage overflows and there is no more space in that. Also, there are sometimes some corrupted files written in it with format XXXXXXX.XXX which X is 0xff in binary. Plus, `esp_vfs_fat_info` fuction returns the `out_free_bytes` 0 byte.

### Steps to reproduce.

It's suddenly and does not have clear behavior, but it happens when operation on filesystem is much more than usual,
such as removing and recreating a file.

### More Information.

Tried on ESP32 and ESP32-S3 but I got the same error.

Hossein-M98, 2023-11-05:
Same problem

AliMoal, 2023-11-05:
I guess it is for wear leveling implementation or FatFs R0.15, but as we know FatFs works correctly on many devices without any problems. So, I'm suspicious of wear level API. Does anyone encounter this?

adokitkat, 2023-11-06:
Hi @AliMoal @Hossein-M98. 

- I am assuming you're using FATFS partition on flash memory and not on SD card. Is that right? 
- Could you please set log level to debug (maybe even verbose) in `idf.py menuconfig` and copy the logs from `idf.py monitor` here as text please? 
- When you say the storage suddenly overflows, what do you exactly mean by that? What is your typical used and free memory? 
- Are you sure your power supply providing enough volage and power? FATFS corruption can also be caused by cut power while performing a operation.
- Have you experienced this in earlier versions of ESP-IDF?

Also if you could provide minimal working example code (post it here in a .zip file) it would be great.

Thanks!

Hossein-M98, 2023-11-07:
Hi @adokitkat, thank you for your response.
1. Yes, I'm using it on Flash memory
2. Captured the serial monitor on file name `cap.txt` and attached on .zip file.
3. Reading and writing just one file periodically (in the production code) will cause that. And by the passage of time, the free space decreases. For this issue, I wrote a simple testing code to write and read a file. It looks like everything is ok until I reset the ESP32, but in the next run of the program, the free size of FATFS reported by `esp_vfs_fat_info` is always equal to 0!
4. Yes, I test it and measured voltage and current, everything is fine.
5. No, unfortunately not.

[Test-WearLeveling-IDF.zip](https://github.com/espressif/esp-idf/files/13277769/Test-WearLeveling-IDF.zip)

AliMoal, 2023-11-07:
@adokitkat please let me complete @Hossein-M98's answer and reply to your question.
I run @Hossein-M98's code and even in my main program the problem DOES exists. I test it on both internal and external (w25q32) flash memory and the results are the same.
I think the restart action will interrupt something about writing headers on it but still don't know this is FatFs problem or Wear leveling implementation.
It's good to see you are facing this issue except us. At this point we have to find out what make this happens.

adokitkat, 2023-11-07:
I am 90% sure this is because of restarting the device while IO operations are not done yet.

What is your typical scenario? Why and when are you restarting your devices?

AliMoal, 2023-11-07:
I can't say when it exactly happens but i know the reading/writing operation is active, specially writing. About the restarting action, I should say this is sometimes because of power outage or in my application, an option to users for applying some configurations to device for example WiFi or Bluetooth config, generally some configurations that need restart to be applied.

adokitkat, 2023-11-07:
You can try to add a bigger timeout like `vTaskDelay(1000 / portTICK_PERIOD_MS);` and a guard to not perform any restart while an IO operation is still in progress (e.g. in a different task). However FATFS is still prone to corrupt while a power outrage (old, not designed for microcontrollers). We are working on a FATFS journaling feature which should help with this but it is not done yet. In a meantime you can try to use a third-part filesystem library [LittleFS](https://github.com/littlefs-project/littlefs) which should be resilient to power-loss. We even have an example code https://github.com/espressif/esp-idf/tree/b90dfe075923870a8d1f8d85a75318781455f57a/examples/storage/littlefs in our esp-idf repo available in `master` and `release/v5.2` branches.

AliMoal, 2023-11-07:
This is not a complete answer, because I don't know when the power goes out.
I'm familiar with littlefs a bit, but I need to implement it on usb or other external devices, and these devices can connect everywhere. So we have to have a known file system format. Can we use littlefs in this situation?
Another information which I should mention is I used FatFs (not R0.15 version of IDF v5.1.1) on other devices and I have never seen this issue.

Hossein-M98, 2023-11-08:
> You can try to add a bigger timeout like `vTaskDelay(1000 / portTICK_PERIOD_MS);` and a guard to not perform any restart while an IO operation is still in progress (e.g. in a different task). However FATFS is still prone to corrupt while a power outrage (old, not designed for microcontrollers). We are working on a FATFS journaling feature which should help with this but it is not done yet.
> 
> In a meantime you can try to use a third-part filesystem library [LittleFS](https://github.com/littlefs-project/littlefs) which should be resilient to power-loss. We even have an example code https://github.com/espressif/esp-idf/tree/b90dfe075923870a8d1f8d85a75318781455f57a/examples/storage/littlefs in our esp-idf repo available in `master` and `release/v5.2` branches. On earlier IDF versions use [esp_littlefs](https://github.com/joltwallet/esp_littlefs).
> 
> Here is a comparison between [FATFS, SPIFFS and LittleFS performance](https://github.com/joltwallet/esp_littlefs#performance).

Thank you for your attention.

We have sold lots of devices with this bug, and we have to update them to fix this problem while keeping all data on FATFS partition.
When we talk about a small embedded system used by a non-professional user, unfortunately, we have no control over the user's behavior and power failure issues. So, I ask you to fix this problem in some way, please.

ermacv, 2023-11-14:
Hello! I found the same issue when I work with 1 big file (around 1.4 Mbytes) and I change the content of the middle of that file. If I shutdown device when it performs IO operations. I added logs to check storage memory and I observe this:
1 bug caught: `app:keys: free_space = 4407296, total_space = 7241728, used_space = 2834432`
2 bug caught: `app:keys: free_space = 2990080, total_space = 7241728, used_space = 4251648`
3 bug caught: `app:keys: free_space = 1572864, total_space = 7241728, used_space = 5668864`

After third attempt I can't create a new file because there is no free space.

Then I added `readdir` to check the content of the root folder (the file is written to it) and I found 512 files with `XXXXXXX.XXX` file name where `X` is binary `0xFF`. The most strange part is that type of the file is 2 which `DIR`.

I'm trying to find the root cause and what part of FatFS or Wear Levelling creates this file. I also rewrote the code to be able to work directly with wear levelling API (the file is kind of database) and I can't reproduce any problems with data structure loss but I use double buffering through special sector.

@Hossein-M98 @AliMoal have you found the root cause?

AliMoal, 2023-11-14:
@ermacv Not yet! But I saw exactly what you saw. Actually, I postponed fixing this problem for now. Please notice us if you found anything.

adokitkat, 2024-03-31:
> @adokitkat How is the status for this fix?

I am not sure if there is a fix at all because I previously only have been able to replicate it in a manner, which likely is breaking FATFS itself (which is really not designed to handle power-loss well during write operations) see the older comment below:

> Thank you for the quick reply.
> 
> So far I wasn't been able to get a corrupted file but I noticed there sometimes is a decrease in the free memory value after a manual restart via button on a ESP32 device: 995328 -> 991232 -> 987136 - all these numbers differ by 4096 (==995328-991232), which a size of a WL sector.
> 
> At what loop count / restart count (and maybe their ratio?) When did you start to experience corrupted files please?
> 
> Also are you running your devices in this fashion? E.g. restarting them while writing in production? Or did you just setup the example code and your devices are running normally (not writing and reading the same file over and over while restarting the device randomly) yet you still experience this issue?
> 
> EDIT: Ok, so at first I had done the restarts manually, then I made it loop 1000 times and then restart (not while performing a operation, still nothing) and then I changed it to 10 loops and then restart and I finally experienced the corruption as well as 0 free memory.

See [our docs](https://docs.espressif.com/projects/esp-faq/en/latest/software-framework/storage/fatfs.html#how-to-improve-the-damage-to-fatfs-file-system-caused-by-accidental-power-loss) as well.

If this issue is also present in normal operation (reboots are NOT happening during IO write operations), then please share an example project as a zip file here.

pacucha42, 2025-03-06:
No reaction for long time, closing as expired. Please, feel free to reopen or file a new tracker. Thank you

closed 2025-03-06 by pacucha42