espressif/esp-idf #12506: NVS runs out of space before expected.

amrsoll, 2023-11-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-3196-g732df630cd

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 S3 mini

### Power Supply used.

USB

### What is the expected behavior?

I try over-writing multiple blobs to the NVS. The NVS partition is 16KB, and the blob sizes are 4 KB and 6 KB respectively.

I expect the blobs to simply overwrite the existing data.

### What is the actual behavior?

Eventually, the NVS runs out of memory and creates a system crash.

### Steps to reproduce.

1. `git clone git@github.com:amrsoll/nvs_rw_blobs.git`
2. `cd nvs_rw_blobs`
3. `idf.py flash monitor`

### Debug Logs.

```plain
I (314) main_task: Started on CPU0
I (324) main_task: Calling app_main()
Nothing saved yet!
I (504) storage: Saving blob, pass 1
Writing : 6000
Writing : 4000
Blob size : 6000
I (664) storage: Saving blob, pass 2
Writing : 6000
Writing : 4000
Blob size : 6000

 [ ... ]

I (12074) storage: Saving blob, pass 78
Writing : 6000
Writing : 4000
Blob size : 6000
I (12224) storage: Saving blob, pass 79
Writing : 6000
Writing : 4000
W (12224) storage: Error (ESP_ERR_NVS_NOT_ENOUGH_SPACE) saving run time blob to NVS!
Blob size : 6000
I (12344) storage: Saving blob, pass 80
Writing : 6000
Writing : 4000
W (12344) storage: Error (ESP_ERR_NVS_NOT_ENOUGH_SPACE) saving run time blob to NVS!
Blob size : 6000

 [ ... ]

I (13544) storage: Saving blob, pass 90
Writing : 6000
Writing : 4000
W (13544) storage: Error (ESP_ERR_NVS_NOT_ENOUGH_SPACE) saving run time blob to NVS!
Blob size : 6000
I (13644) storage: Saving blob, pass 91
Writing : 6000
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40056ece  PS      : 0x00060f30  A0      : 0x8200d300  A1      : 0x3fc99d70  
0x40056ece: memcmp in ROM

A2      : 0x00000000  A3      : 0x3fc99dc4  A4      : 0x00000020  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x3fc99da4  A8      : 0x00000008  A9      : 0x3fc99d50  
A10     : 0x00000000  A11     : 0x00000080  A12     : 0x3fc99dc4  A13     : 0x00000020  
A14     : 0x3c0250c8  A15     : 0x00000000  SAR     : 0x00000002  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  
0x40056f5c: memcpy in ROM

0x40056f72: memcpy in ROM

Backtrace: 0x40056ecb:0x3fc99d70 0x4200d2fd:0x3fc99d90 0x4200b5af:0x3fc99e20 0x4200bee6:0x3fc99e90 0x4200c219:0x3fc99f00 0x4200b177:0x3fc99f20 0x4200814d:0x3fc99f50 0x420082ec:0x3fc99f80 0x4201b6ab:0x3fc99fa0 0x4037a951:0x3fc99fd0
0x40056ecb: memcmp in ROM

0x4200d2fd: nvs::Page::cmpItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int, unsigned char, nvs::VerOffset) at /home/alex/esp/esp-idf/components/nvs_flash/src/nvs_page.cpp:339

0x4200b5af: nvs::Storage::cmpMultiPageBlob(unsigned char, char const*, void const*, unsigned int) at /home/alex/esp/esp-idf/components/nvs_flash/src/nvs_storage.cpp:560

0x4200bee6: nvs::Storage::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int) at /home/alex/esp/esp-idf/components/nvs_flash/src/nvs_storage.cpp:322

0x4200c219: nvs::NVSHandleSimple::set_blob(char const*, void const*, unsigned int) at /home/alex/esp/esp-idf/components/nvs_flash/src/nvs_handle_simple.cpp:44

0x4200b177: nvs_set_blob at /home/alex/esp/esp-idf/components/nvs_flash/src/nvs_api.cpp:422

0x4200814d: save_blobs at /home/alex/workspace/nvs_rw_blob/main/nvs_blob_example_main.c:56

0x420082ec: app_main at /home/alex/workspace/nvs_rw_blob/main/nvs_blob_example_main.c:139

0x4201b6ab: main_task at /home/alex/esp/esp-idf/components/freertos/app_startup.c:208

0x4037a951: vPortTaskWrapper at /home/alex/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:137

ELF file SHA256: 36bf2bf24cc13bb5

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375800
0x40375800: esp_restart_noos at /home/alex/esp/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:158
```
```

### More Information.

See my repository for the source files to reproduce error

https://github.com/amrsoll/nvs_rw_blobs

amrsoll, 2023-11-01:
I fixed a memory leak in the original example.

However, the NVS partition does not want to allocate space for 2x 6KB blobs. I updated the example

AxelLin, 2023-12-10:
```
 * // Example of nvs_get_stats() to get the number of used entries and free entries:
 * nvs_stats_t nvs_stats;
 * nvs_get_stats(NULL, &nvs_stats);
 * printf("Count: UsedEntries = (%d), FreeEntries = (%d), AllEntries = (%d)\n",
          nvs_stats.used_entries, nvs_stats.free_entries, nvs_stats.total_entries);
```

amrsoll, 2023-12-12:
So you're saying I should check for available space before committing? How does that explain the bahaviour I am seeing?

rrtandler, 2023-12-12:
Hi @amrdoll,

Please look at the documentation of [nvs_stats_t](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html#_CPPv413nvs_get_statsPKcP11nvs_stats_t) returned by nvs_get_stats. This structure was extended by the member available_entries (not shown in @AxelLin 's example) which indicates useable space for data. Please note, that (over)writing blobs requires extra entries for blob data (one entry per 32 bytes of data) and 2 entries for overhead data (index and key). If you are overwriting existing blob (re-using same key), the data occupied by the old value will be returned to the pool of available entries first after successful writing of the new value. It effectively means, that during the write operation, you will need space for both (data as well as overhead).

rrtandler, 2025-04-07:
Hi @@amrsoll,
I am closing the issue as there was no response and no other activity for couple of recent months.

closed 2025-04-07 by rrtandler