espressif/esp-idf #12508: TWAI returns ESP_ERR_INVALID_STATE

jrmurdoch, 2023-11-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm having issues getting TWAI to work. I'm now using the TWAI self test example.

If I run the example with a loop wire between TX and RX all works fine.

```
I (280) main_task: Calling app_main()
I (280) gpio: GPIO[1]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (290) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (300) TWAI Self Test: Driver installed
I (310) TWAI Self Test: Driver started
I (310) TWAI Self Test: Msg received - Data = 0
I (320) TWAI Self Test: Msg received - Data = 1
I (330) TWAI Self Test: Msg received - Data = 2
I (340) TWAI Self Test: Msg received - Data = 3
I (350) TWAI Self Test: Msg received - Data = 4
I (360) TWAI Self Test: Msg received - Data = 5
I (370) TWAI Self Test: Msg received - Data = 6
```

If I remove the loop wire when TX (gpio 0) and RX (gpio 1) the code is left waiting indefinitely as expected with the following code:
`ESP_ERROR_CHECK(twai_receive(&rx_message, portMAX_DELAY));`

```
I (280) main_task: Calling app_main()
I (280) gpio: GPIO[1]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (290) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (300) TWAI Self Test: Driver installed
I (310) TWAI Self Test: Driver started
```

If I change to TX (gpio 4) and RX (gpio 5) with the loop wire all works as expected. If I remove the loop wire with the change in IO to 4 & 5 the response is different.

```
I (285) main_task: Calling app_main()
I (285) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (295) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (305) TWAI Self Test: Driver installed
I (315) TWAI Self Test: Driver started
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4200724e
0x4200724e: twai_transmit_task at E:/Software/esp32_examples/twai_self_test/main/twai_self_test_example_main.c:64 (discriminator 1)

file: "./main/twai_self_test_example_main.c" line 64
func: twai_transmit_task
expression: twai_transmit(&tx_msg, portMAX_DELAY)
```

The issue I have is I have is trying to track down the what is causing the "ESP_ERR_INVALID_STATE" on twai_transmit as this shouldn't happen.

wanckl, 2023-11-02:
@jrmurdoch  Nice to meet you,,,,     
you can config gpio pulldown for RX pin after twai_driver _install to resolve it: `gpio_pulldown_en(RX_GPIO_NUM);`     

It is a driver issue, we will fix it ASAP, thanks for your report, best wishes !! :heart:

wanckl, 2023-11-08:
@jrmurdoch  Hello, glad to tell you the result:     

After internal evaluation, we don't think your usage (left `TX` and `RX` pin as floting) is a valid use case. For common TWAI bus usage, we can:     
- Connected to external PHY for regular application use
- Connect TX and RX with jumper for loopback testing
- Use same pin for TX and RX (without jumper) for loopback testing

We can floating `CAN_H/L` after the PHY, but floating TX/RX don't have a defined status, we should't use as this. PLS check     

Thank you.

wanckl, 2023-11-27:
@jrmurdoch  Any other question is welcome, otherwise issue will closed

jrmurdoch, 2023-11-27:
Happy, I was debugging an issue and noticed that the response was different depending on the pins used which seemed strange. Would have thought that in the event something goes wrong and the pins are disconnected then a common error would be returned regardless of the pins used. Either way the pull-up suggestion you offered resolved this for me.