espressif/esp-idf #12512: How to reduce DRAM size

ArvinHou, 2023-11-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have many parameters stored in nvs flash and they are modifiable.I have to define a lot of  variables and then read value from flash during initialization.
Now the bss is 125k.There are some issues with the program when running。Such as emac no mem....
```
Used static DRAM:  137660 bytes (  43076 remain, 76.2% used)

      .data size:   12220 bytes

      .bss  size:  125440 bytes
```
Is there any method  to store these variables and then use them in the program without occupying additional memory ？
In STM32, I can use a pointer to point to the address of the flash。When the access pointer directly accesses the data stored in flash。
In ESP32 ,It seems impossible to do this?

ArvinHou, 2023-11-02:
I have read /api-guides/performance/ram-usage ,I have do some optimizations.But that's not enough。

Dazza0, 2023-11-02:
@ArvinHou 

> In STM32, I can use a pointer to point to the address of the flash。When the access pointer directly accesses the data stored in flash。

This won't work on the ESP32 because the CPU does not have direct write-access to Flash. Flash on the ESP32 is connected via SPI.

> I have to define a lot of variables and then read value from flash during initialization.

Will using dynamic memory be a viable alternative here?

rrtandler, 2023-11-03:
Hi @ArvinHou,
One of possible ways to reduce internal RAM would be using PSRAM provided your module has some. You can then allocate the variables using heap_caps_malloc with capability parameter MALLOC_CAP_SPIRAM.
Please see details [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/mem_alloc.html)

rrtandler, 2023-11-22:
Hi @ArvinHou,
Please let me know whether you have solved the issue or need any further help.
Thanks.

Alvin1Zhang, 2024-02-27:
Thanks for reporting, will close due to short of feedback, feel free to reopen. Thanks.

closed 2024-02-27 by Alvin1Zhang