espressif/esp-idf #12513: ESP32S3 Espressif sysview with FreeRTOS multicore does not work

TiboDevC, 2023-11-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

## General issue report

I try to use the Espressif sysview to debug a memory leak. Here is the Espressif documentation about sysview:
https://docs.espressif.com/projects/esp-idf/en/v5.1.1/esp32s3/api-reference/system/heap_debug.html#host-based-mode 

This example works well when FreeRTOS is configured to run on a single core but does not work when it runs on dual core.

## Environment 

- ESP32S3
- Espressif SDK v5.1.1
- openocd `v0.12.0-esp32-20230921 (2023-09-21-13:41)`
- xtensa-esp32s3-elf-gdb `12.1_20221002`

## Demo code

### Working example

The Espressif example works well, here is an example code: 
https://github.com/TiboDevC/esp32_heap_tracing/tree/master

### Non-working example

The only difference with the working one is `Run FreeRTOS only on first core` is unchecked in `menuconfig`
https://github.com/TiboDevC/esp32_heap_tracing/tree/freertos_multi_core

### Steps to reproduce

Flash one of the above example code, then:
```
# Run openocd
openocd -f board/esp32s3-builtin.cfg

# In another terminal, run gdb with the `gdbinit` described as above in github links
xtensa-esp32s3-elf-gdb -x gdbinit ./build/heap-trace-host-example.elf
```
### logs when it works
```
$ xtensa-esp32s3-elf-gdb -x gdbinit ./build/heap-trace-host-example.elf 
GNU gdb (esp-gdb) 12.1_20221002
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=xtensa-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./build/heap-trace-host-example.elf...
warning: multi-threaded target stopped without sending a thread-id, using first non-exited thread
[Switching to Thread 1070174560]
0x40379b46 in esp_cpu_wait_for_intr () at /home/myUserName/esp/esp-idf/components/esp_hw_support/cpu.c:121
121 }
JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
[esp32s3.cpu0] requesting target halt and executing a soft reset
[... 117 lines cut ...]

Thread 4 "main" hit Temporary breakpoint 1, heap_trace_start (mode_param=(HEAP_TRACE_LEAKS | unknown: 0x2)) at /home/myUserName/esp/esp-idf/components/app_trace/heap_trace_tohost.c:36
36	{
Total trace memory: 16384 bytes
Open file /tmp/heap_cpu0.svdat
Open file /tmp/heap_cpu1.svdat
App trace params: from 2 cores, size 4294967295 bytes, stop_tmo -1 s, poll period 0 ms, wait_rst 0, skip 0 bytes
Connect targets...
Targets connected.
```
It is stuck here. When I break with gdb:
```
^C[esp32s3.cpu0] Target halted, PC=0x4037EBE8, debug_reason=00000000
Set GDB target to 'esp32s3.cpu0'
[esp32s3.cpu1] Target halted, PC=0x4037EBFA, debug_reason=00000000

Thread 4 "main" received signal SIGINT, Interrupt.
spinlock_acquire (timeout=-1, lock=0x3fc9409c <xKernelLock>) at /home/myUserName/esp/esp-idf/components/esp_hw_support/include/spinlock.h:112
112	        lock_set = esp_cpu_compare_and_set(&lock->owner, SPINLOCK_FREE, core_id);
```
it seems to be stuck in this function.

I tried also with Espressif v5.2-dev but same result.

Is it a bug or a miss-configuration?

erhankur, 2023-11-03:
@TiboDevC It should work. I will test it and turn back. 

But one thing strange to me, you started sysview in both modes with the below command, right? 
```
mon esp sysview start file:///tmp/heap_cpu0.svdat file:///tmp/heap_cpu1.svdat
```

For the single mode it should be like `mon esp sysview start file:///tmp/heap_cpu0.svdat`

erhankur, 2023-11-03:
OK looks like we have an issue with double core sysview. Also, we have an issue in OpenOCD to understand if 2nd core is disabled or not. 

In single core mode, you can run OpenOCD like this
```
openocd -c "set ESP32_S3_ONLYCPU 1" -f board/esp32s3-builtin.cfg
```

and then no need to mention about 2nd file 
```
mon esp sysview start file:///tmp/heap_cpu.svdat
```

I will have a look both issues and let you know.

TiboDevC, 2023-11-06:
> @TiboDevC It should work. I will test it and turn back.
>
> But one thing strange to me, you started sysview in both modes with the below command, right?
>
> ```
> mon esp sysview start [...]

yes, I could have started it with a single file for the single core, right.
Thanks for having a look!