espressif/esp-idf #12515: Static ip address of IPV6

ElieSKickMaker, 2023-11-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Is it possible to have a static address on IPV6 ?  
On the example with ipv4 it's using esp_netif_set_ip_info  but the structure esp_netif_ip_info_t is only for ipv4. For IPV6 the strucutre is esp_netif_ip6_info_t; 

### What is the actual behavior?

Not working

### Steps to reproduce.

Code not working :
```c
/* stop DHCP client*/
		ESP_ERROR_CHECK(esp_netif_dhcpc_stop(eth_netif)) ;

        #if CONFIG_IPV6_STATIC
        ESP_LOGI(TAG, "Set IPV6 static address:%s", CONFIG_IPV6_STATIC_ADDR);
        /* assign a static IP to the ETH network interface */
        esp_netif_ip6_info_t static_ipv6;
        memset(&static_ipv6, 0x00, sizeof(static_ipv6)) ;
        ip6addr_aton(CONFIG_IPV6_STATIC_ADDR, &static_ipv6.ip);
        ESP_ERROR_CHECK(esp_netif_set_ip_info(eth_netif, &static_ipv6)) ;  
        /* assign a static DNS to the ETH network interface */
       
        esp_netif_dns_info_t dns_info;
        ip6addr_aton(CONFIG_IPV6_STATIC_DNS_ADDR, &static_ipv6.ip);
                      dns_info.ip.type = IPADDR_TYPE_V6;
        ESP_ERROR_CHECK(esp_netif_set_dns_info(eth_netif, ESP_NETIF_DNS_MAIN, &dns_info)); 
```

ElieSKickMaker, 2023-11-13:
Hello @espressif-abhikroy , do you need more informations to provide any help ? 
Thansk you

espressif-abhikroy, 2023-11-13:
To be able to add an ipv6 address to a esp-netif try using
```c
esp_err_t esp_netif_add_ip6_address(esp_netif_t *esp_netif, const ip_event_add_ip6_t *addr);
```

ElieSKickMaker, 2023-11-13:
Hello @espressif-abhikroy , Thanks for your help but I've also tested it and I get an error, "Invalid static ip" : 
```c
`#define CONFIG_IPV6_STATIC_ADDR "2001:861:3a00:be40::2791:fa50"`

esp_netif_ip6_info_t ip6_addr = { 0 };
        ip6addr_aton(CONFIG_IPV6_STATIC_ADDR, &ip6_addr.ip);
        ESP_ERROR_CHECK(netif_add_ip6_address(eth_netif, (ip6_addr_t *)&ip6_addr.ip, 0));

        esp_netif_dns_info_t dns_info;
        ip6addr_aton(CONFIG_IPV6_STATIC_DNS_ADDR, &dns_info.ip.u_addr.ip6.addr);
        dns_info.ip.type = IPADDR_TYPE_V6;
        ESP_ERROR_CHECK(esp_netif_set_dns_info(eth_netif, ESP_NETIF_DNS_MAIN, &dns_info)); 

```

[image]

[image]

this error is in the file esp_netif\esp_netif_handlers.c and it's the return of the function

```c
esp_netif_is_valid_static_ip
```
it's normal because it's only checking if it's an ip4 format or not ...

espressif-abhikroy, 2023-11-14:
Hi @ElieSKickMaker,
Please try replacing the following code
```c
esp_netif_ip6_info_t ip6_addr = { 0 };
        ip6addr_aton(CONFIG_IPV6_STATIC_ADDR, &ip6_addr.ip);
        ESP_ERROR_CHECK(netif_add_ip6_address(eth_netif, (ip6_addr_t *)&ip6_addr.ip, 0));
```

with

```c
    esp_netif_ip6_info_t ip6_addr = { 0 };
    ip6addr_aton(CONFIG_IPV6_STATIC_ADDR, (ip6_addr_t *)&ip6_addr.ip);
    struct netif *lwip_netif = esp_netif_get_netif_impl(eth_netif);

    ESP_ERROR_CHECK(netif_add_ip6_address(lwip_netif, (ip6_addr_t *)&ip6_addr.ip, 0));
```

`netif_add_ip6_address()` is an lwip api and doesn't accept `esp_netif`.

You will also need to include 
```c
#include "lwip/netif.h"
#include "esp_netif_net_stack.h"
```

espressif-abhikroy, 2024-02-05:
```C
esp_err_t esp_netif_add_ip6_address(esp_netif_t *esp_netif, const esp_ip6_addr_t addr, bool preferred) 
esp_err_t esp_netif_remove_ip6_address(esp_netif_t *esp_netif, const esp_ip6_addr_t *addr)
```
has beed added to master.

eriksl, 2024-06-18:
It's not in de documentation (yet) (5.2.2). That's probably because it's mentioned in a private header (esp_netif_private.h). Please amend.

AxelLin, 2024-07-31:
> It's not in de documentation (yet) (5.2.2). That's probably because it's mentioned in a private header (esp_netif_private.h). Please amend.

@espressif-abhikroy 
Any update for above comment?
I think 325a8d7a6ea67c7bea1f99bc39d2dcd38780c1bc needs backport to stable releases to fix this issue.

espressif-abhikroy, 2024-07-31:
Thanks for pointing that out. I'm currently working on backporting the changes from 325a8d7 to the stable releases. I'll keep you updated on the progress.

AxelLin, 2024-08-20:
> Thanks for pointing that out. I'm currently working on backporting the changes from [325a8d7](https://github.com/espressif/esp-idf/commit/325a8d7a6ea67c7bea1f99bc39d2dcd38780c1bc) to the stable releases. I'll keep you updated on the progress.

@espressif-abhikroy
It does not make sense taking so much time to backport the fix especially 325a8d7 can be cleanly applied to 5.2 branch.

AxelLin, 2024-11-05:
> It's not in de documentation (yet) (5.2.2). That's probably because it's mentioned in a private header (esp_netif_private.h). Please amend.

v5.2 fix: 8809589cf74a6d0f7ed27ae84dd44a834bcfc5c8

closed 2025-01-10 by espressif-abhikroy