espressif/esp-idf #12516: Impossible Deadlock triggered by multicore semaphore take

timoxd7, 2023-11-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-Wrover-E (Module)

### Power Supply used.

USB

### What is the expected behavior?

I can take a mutex from different tasks on different cores. After a task gave back the mutex, another task blocked on this semaphore gets woken up and can continue. This is not coupled to any other component or critical section. As long as i always give back a semaphore, a take should be possible next.

### What is the actual behavior?

After initializing many different components (BLE, WiFi, I2C, CAN...), it can periodically happen that taking a semaphore that is later given back will not be counted as given back, creating a deadlock. This means the semaphore is given back by a given task, but FreeRTOS does not count it as given back OR does not wake up any task blocked by this semaphore (not sure which of these).

### Steps to reproduce.

Hard to describe, we have a large codebase and i was not able to reproduce this issue in an isolated project. However, the isolated project only spawned 20 tasks with 10 on each core, making some busy wait and semaphore take & give, but without BLE, WiFi, OTA-Write or critical sections.

### More Information.

1. The issue was discovered during an OTA-Write. During normal operation, the issue was not discovered. However, the Tasks who take the semaphore/mutex are not bound in any way to the OTA-Write Task.
2. The mentioned mutex is always given back and does not take another semaphore. There is no possibility to create a deadlock caused by this mutex/tasks taking this mutex. We have implemented a deadlock detection which prints a backtrace of the last task successfully taking the semaphore and the current task blocked on this semaphore. Both can't be the issue, as the point where both tasks are/were will always give back the semaphore, with no source of blocking inbetween take & give.
3. The issue is hardly reproducable in an isolated project due to the many components to be initialized and running.
4. We initialize and actively use WiFi (ESP-NOW), BLE, I2C and CAN (while the CAN bus was silent). After starting an OTA-Write, it took between 1s and 20s to trigger the deadlock.
5. We already implemented a workaround to always do initializations on the same core the code handling interrupts and events is running on. This already resulted in some fixed bugs where also some deadlocks occured if we initialize e.g. I2C on Core 0 and running the callbacks and polls for I2C data on Core 1. This might also be a bug, but we found a working workaround.
6. The Coredumps are always broken, so we can't exactly see where the tasks are stuck, besides the tasks using the deadlock detection (all our own tasks) blocked by the given semaphore and the task who previously took the mentioned semaphore. The Coredump was broken even if we enabled the dedicated Coredump Stack, moved from UART to Flash or decoded the Base64 UART Dump manually.
7. The most important one: Moving all the Tasks that use the mentioned mutex and the Task writing the OTA image on one core solved the problem, no deadlock can be seen anymore.
8. The mentioned mutex is one of few (or at this point in time the only one) that is regularily taken and given from multiple tasks on multiple cores.
9. The task and method/function last successfully taking the semaphore differs between runs. This might also indicate to not be in relation with our sourcecode and be a deeper problem.

In know it is hard to find a bug without the sourcecode producing the error and without knowing that the issue might be the sourcecode itself, but i already researched alot and was able to pinpoint the error being not inside of our sourcecode. I hope we can cooperate on this issue as it might be a bug inside the RTOS or related to critical sections.

erhankur, 2023-11-02:
@timoxd7 Thank you for the detailed informations. I am interested in coredump part now. How do you trigger it? Your deadlock detection task can detect successfully? And what happens when coredump has been started to dump? What is the error you get? Dump is ok but parsing is not? Maybe you can send that file. 
Did you also try to get sysview trace and examine the tasks with the Segger tool?

Dazza0, 2023-11-02:
@timoxd7 When the daedlock occurs, please run GDB over JTAG and dump the contents of the mutex object. From the context, you should be able to determine what tasks are blocked on the mutex, and whether or not a task is holding the mutex.

timoxd7, 2023-11-02:
@erhankur Thank you for the fast reply. To clearify a bit, i want to describe what i did/what we have

### Deadlock Detection
We simply have a wrapper class for all semaphore access. We then have a `lock`-method that has `portMAX_DELAY` as standard argument if no other time is given. If it is `portMAX_DELAY`, than we replace internally to a given time (for us 15s) and if this time runs out, we assume a deadlock, as we never have a mutex taken by a task this long (normally <1s, in some rare cases ~10s). If than a potential "deadlock" is detected, a backtrace is printed which then is decoded from the uart console. There is also a verbose mode for debugging, then we allocate a buffer and write the current backtrace after every time we successfully took the semaphore. On a deadlock we can then print the last backtrace too to find out not only which task is holding the semaphore, but also which exact function and line made the last take call.

### Coredump trigger
After we detect the deadlock, we wait 2s to let other tasks blocked by the same semaphore report its error too. After that, we trigger a crash with "abort()". Throu this, the coredump is written to the console and directly decoded. We indeed have had multiple coredumps during development, some without a problem and sometimes without a working coredump. This time however, i was never able to obtain a coredump. For the Problem i have, the deadlock detection is always the trigger, but for the 15s timeout you can see that the system is hanging.

### Error Message
On decoding, i get this message after receiving the coredump over uart:

```
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\user\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_idf_monitor\__main__.py", line 4, in <module>
    main()
  File "C:\Users\user\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_idf_monitor\idf_monitor.py", line 389, in main
    monitor.main_loop()
  File "C:\Users\user\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_idf_monitor\idf_monitor.py", line 174, in main_loop
    self._main_loop()
  File "C:\Users\user\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_idf_monitor\idf_monitor.py", line 280, in _main_loop
    super()._main_loop()
  File "C:\Users\user\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_idf_monitor\idf_monitor.py", line 219, in _main_loop
    self.serial_handler.handle_serial_input(data, self.console_parser, self.coredump,  # type: ignore
  File "C:\Users\user\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_idf_monitor\base\serial_handler.py", line 112, in handle_serial_input
    with coredump.check(line_strip):
  File "contextlib.py", line 137, in __enter__
  File "C:\Users\user\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_idf_monitor\base\coredump.py", line 128, in check
    self._check_coredump_trigger_before_print(line)
  File "C:\Users\user\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_idf_monitor\base\coredump.py", line 106, in _check_coredump_trigger_before_print
    self._process_coredump()
  File "C:\Users\user\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_idf_monitor\base\coredump.py", line 79, in _process_coredump
    coredump.info_corefile()
  File "C:\Users\user\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_coredump\coredump.py", line 514, in info_corefile
    self.print_threads_info(task_info)
  File "C:\Users\user\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_coredump\coredump.py", line 337, in print_threads_info
    task_name = self.gdb_esp.get_freertos_task_name(tcb_addr)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_coredump\corefile\gdb.py", line 78, in get_freertos_task_name
    val = self._gdbmi_data_evaluate_expression('(char*)((TCB_t *)0x%x)->pcTaskName' % tcb_addr)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_coredump\corefile\gdb.py", line 72, in _gdbmi_data_evaluate_expression
    return self._gdbmi_run_cmd_get_one_response("-data-evaluate-expression \"%s\"" % expr,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: 'NoneType' object is not subscriptable
Traceback (most recent call last):
  File "C:\Users\user\esp\esp-idf\tools/idf_monitor.py", line 11, in <module>
    sys.exit(subprocess.run([sys.executable, '-m', 'esp_idf_monitor'] + sys.argv[1:]).returncode)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "subprocess.py", line 550, in run
  File "subprocess.py", line 1199, in communicate
  File "subprocess.py", line 1262, in wait
  File "subprocess.py", line 1543, in _wait
```

timoxd7, 2023-11-02:
@Dazza0 
> @timoxd7 When the daedlock occurs, please run GDB over JTAG and dump the contents of the mutex object. From the context, you should be able to determine what tasks are blocked on the mutex, and whether or not a task is holding the mutex.

I will try to do this with our project. However, this might take some time because our product already uses all pins and has the JTAG Pins not exposed, hence i need to use a development board and change our software to be able to run it without some things (like CAN, SD ...).

timoxd7, 2023-11-02:
The same Error for the Coredump Decode based from the flash:

<Details>
PS C:\work\company\Efoil_Code\main> idf.py -p COM7 coredump-info
C:\Users\user\esp\esp-idf\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: coredump-info
Failed to load core dump: parttool script execution failed with error 1, failed command was: 'C:\Users\user\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:\Users\user\esp\esp-idf\components\partition_table\parttool.py --port COM7 --baud 460800 read_partition --partition-type data --partition-subtype coredump --output C:\Users\TIMO~1.MEY\AppData\Local\Temp\tmp_x21pp_m'

┌────── Additional information about the error:
│
│   esptool.py v4.6.2
│   Serial port COM7
│   Connecting...................
│   Detecting chip type... Unsupported detection protocol, switching and trying again...
│   Connecting.....
│   Detecting chip type... ESP32
│   Chip is ESP32-D0WD-V3 (revision v3.0)
│   Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
│   Crystal is 40MHz
│   MAC: c4:5b:be:3d:78:38
│   Uploading stub...
│   Running stub...
│   Stub running...
│   Changing baud rate to 460800
│   Changed.
│   3072 (100 %)
│   3072 (100 %)
│   Read 3072 bytes at 0x00008000 in 0.1 seconds (327.2 kbit/s)...
│   Hard resetting via RTS pin...
│   Running C:\Users\user\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:\Users\user\esp\esp-idf\components\esptool_py\esptool\esptool.py --port COM7 --baud 460800 read_flash 32768 3072 C:\Users\TIMO~1.MEY\AppData\Local\Temp\tmpp2vj8vjw...
│   Traceback (most recent call last):
│     File "C:\Users\user\esp\esp-idf\components\partition_table\parttool.py", line 358, in <module>
│       main()
│     File "C:\Users\user\esp\esp-idf\components\partition_table\parttool.py", line 327, in main│       target = ParttoolTarget(**target_args)
│                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
│     File "C:\Users\user\esp\esp-idf\components\partition_table\parttool.py", line 105, in __init__
│       partition_table = gen.PartitionTable.from_binary(f.read())
│                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
│     File "C:\Users\user\esp\esp-idf\components\partition_table\gen_esp32part.py", line 318, in from_binary
│       result.append(PartitionDefinition.from_binary(data))
│                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
│     File "C:\Users\user\esp\esp-idf\components\partition_table\gen_esp32part.py", line 467, in from_binary
│       res.name = res.name.decode()
│                  ^^^^^^^^^^^^^^^^^
│   UnicodeDecodeError: 'utf-8' codec can't decode byte 0x94 in position 0: invalid start byte

│
└────── end of additional information about the error.
<\Details>

timoxd7, 2023-11-03:
After many tries and things i tested out, i finally managed to find the bug. It is however in our Source Code (i am really sorry for the inconvenience) and based by a full Queue, but i was only able to find it using the gdbstub (thank you really much for that!) and while removing the coredump print, as this always results in a crash of the coredump decode and thus returns prior gdb launch. With GDB i then was able to go to the stuck task (the current holder of the semaphore reported by the deadlock detection) and see exactly where it was stuck.

However, i still was not able to get a working coredump after the `abort()`. So i have a different Question @erhankur: Are there circumstances in which the coredump is not able to be decoded? Might there be a reason why my coredump is broken and could it be fixed with a bugfix or is it impossible due to special cases?

erhankur, 2023-11-03:
@timoxd7 good news. Glad you found the root cause.

Regarding coredump decode issue,  just a random thought, have you tried starting with a fresh coredump partition and still encountered the error? For instance, you could try adding `esp_core_dump_image_erase();` at the beginning of your application and give it another try. I recall that there were decoding errors in previous user issues, and we believed they had been resolved. However, it appears that the issue is still there.  

To get a clearer understanding, could you specify which of the following scenarios is causing issues for you?

1 - Dumping to UART with an extended gdb timeout.
2 - Saving to flash.
3 - Successfully saving to Flash, but crash with the transfer and decoding process on the PC.

And what is your partition size for the core dump?

Dazza0, 2023-11-13:
@timoxd7 Glad you were able to find the cause of the bug. I'm closing this issue as it's not related to deadlocking semaphores anymore. If you still have issues with your core dump, please feel free to open another GitHub issue to track it.

Since you had some debugging issues with a lack of JTAG pins, I'm leaving some links to debugging options that don't require the JTAG pins for your reference. Hope these help in the future.

- GDB Stub Panic: Launches GDB stub on panic for post mortem debugging. See `CONFIG_ESP_SYSTEM_PANIC_GDBSTUB` in [`CONFIG_ESP_SYSTEM_PANIC`](https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32/api-reference/kconfig.html?highlight=gdb%20stub#config-esp-system-panic)
- GDB Stub Runtime: Allows GDB stub to run during run time by launching a background task. Allows setting breakpoints and halting program. See `CONFIG_ESP_SYSTEM_GDBSTUB_RUNTIME` in [`CONFIG_ESP_SYSTEM_PANIC`](https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32/api-reference/kconfig.html?highlight=gdb%20stub#config-esp-system-panic)
- [Application Tracing via Sysview](https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32/api-guides/app_trace.html#system-behavior-analysis-with-segger-systemview): Sysview can be enabled via [`CONFIG_APPTRACE_SV_ENABLE`](https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32/api-reference/kconfig.html#config-apptrace-sv-enable). The tracing data can then be transmitted from ESP32 to host via UART by enabling `CONFIG_APPTRACE_SV_DEST_UART`

closed 2023-11-13 by Dazza0