espressif/esp-idf #12517: WiFi Authmodes are being shuffled on v5.1.1

bill88t, 2023-11-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32-C6 ESP32 ESP32-S3 (and probably the rest)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

WeAct ESP32-C6, M5Stack Timer Camera X, Adafruit Feather ESP32-S3 TFT

### Power Supply used.

USB

### What is the expected behavior?

A wifi scan from my trustworthy thinkpad returns:
```
[bill88t@Thinkpood | ~/git/esp-idf-v5.1.1/examples/wifi/scan]> nmcli device wifi list
IN-USE  BSSID              SSID                     MODE   CHAN  RATE        SIGNAL  BARS  SECURITY       
*       3C:58:C2:93:D4:29  KeyFalse                 Infra  1     65 Mbit/s   63      ***   WPA1 WPA2 WPA3 
        8C:3B:AD:21:3D:1E  Forthnet-5DbGR_EXT       Infra  1     270 Mbit/s  55      **    WPA2           
        D4:60:E3:A9:77:D6  COSMOTE-478503           Infra  10    270 Mbit/s  50      **    WPA2           
        74:9D:79:64:BE:B4  VODAFONE_2081            Infra  1     270 Mbit/s  47      **    WPA1 WPA2      
        E0:19:54:42:2D:EA  WIND_2.4G_422DEA         Infra  4     130 Mbit/s  44      **    WPA2           
        E0:19:54:4E:4F:3F  Feline34                 Infra  9     270 Mbit/s  44      **    WPA1 WPA2      
        34:36:54:D2:F6:CB  Agis2                    Infra  1     130 Mbit/s  37      **    WPA1 WPA2      
        90:9A:4A:B3:23:7E  Deco                     Infra  5     270 Mbit/s  35      **    WPA2           
        E0:19:54:4E:4F:40  Feline34_5               Infra  40    270 Mbit/s  35      **    WPA1 WPA2      
        D4:60:E3:A9:77:DB  COSMOTE-478503           Infra  44    270 Mbit/s  35      **    WPA2           
        E0:0E:E4:C6:B7:C6  VODAFONE_8934            Infra  48    270 Mbit/s  35      **    WPA1 WPA2      
        08:AA:89:79:1B:91  WIND_791B90              Infra  52    270 Mbit/s  35      **    WPA2           
        0A:AA:89:7B:1B:91  --                       Infra  52    270 Mbit/s  35      **    WPA2           
        50:42:89:77:A7:C4  COSMOTE-cg29sh           Infra  60    270 Mbit/s  35      **    WPA2           
        74:9D:79:64:BE:B6  VODAFONE_2081            Infra  100   270 Mbit/s  35      **    WPA1 WPA2      
        50:42:89:77:A7:C3  COSMOTE-cg29sh           Infra  5     270 Mbit/s  34      **    WPA2           
        28:EE:52:24:61:1E  TP-Link_611E             Infra  2     270 Mbit/s  30      *     WPA1 WPA2      
        A0:95:7F:ED:2D:11  COSMOTE-141495           Infra  9     270 Mbit/s  30      *     WPA2           
        78:96:82:56:D9:69  Forthnet-5DbGR           Infra  1     270 Mbit/s  29      *     WPA1 WPA2      
        D4:76:EA:07:A2:E4  Forthnet-AKzTk           Infra  1     270 Mbit/s  29      *     WPA1 WPA2      
        24:D3:F2:9D:7C:F1  Vodafone_2.4G-00311      Infra  1     130 Mbit/s  29      *     WPA1 WPA2      
        28:77:77:F3:23:E4  COSMOTE-atk447           Infra  1     130 Mbit/s  29      *     WPA2           
        98:00:6A:0E:A5:A1  VODAFONE_WIFI_108-07306  Infra  1     130 Mbit/s  29      *     WPA1 WPA2      
        D8:21:DA:21:0B:D1  COSMOTE-893955           Infra  2     270 Mbit/s  29      *     WPA2           
        28:DE:A8:F4:9A:70  VODAFONE_H268Q-0294      Infra  6     130 Mbit/s  29      *     WPA2           
        70:89:76:CE:C9:19  SmartLife-C918           Infra  6     65 Mbit/s   29      *     --             
        F0:87:56:A4:B4:21  Zyxel_A4B421             Infra  7     270 Mbit/s  29      *     WPA2           
        8C:DE:F9:03:4F:57  --                       Infra  7     270 Mbit/s  29      *     WPA1 WPA2      
        28:D1:27:F8:81:D3  kostastia_plus           Infra  13    270 Mbit/s  29      *     WPA1 WPA2
```
[... 448 lines cut ...]
proxy.pac-script:                       --
GENERAL.NAME:                           Hotspot
GENERAL.UUID:                           03f6d7cb-3162-47cb-ab51-3f2179af0336
GENERAL.DEVICES:                        wlp2s0
GENERAL.IP-IFACE:                       wlp2s0
GENERAL.STATE:                          activated
GENERAL.DEFAULT:                        no
GENERAL.DEFAULT6:                       no
GENERAL.SPEC-OBJECT:                    /org/freedesktop/NetworkManager/AccessPoint/14
GENERAL.VPN:                            no
GENERAL.DBUS-PATH:                      /org/freedesktop/NetworkManager/ActiveConnection/3
GENERAL.CON-PATH:                       /org/freedesktop/NetworkManager/Settings/4
GENERAL.ZONE:                           trusted
GENERAL.MASTER-PATH:                    --
IP4.ADDRESS[1]:                         10.42.0.1/24
IP4.GATEWAY:                            --
IP4.ROUTE[1]:                           dst = 10.42.0.0/24, nh = 0.0.0.0, mt = 600
IP6.ADDRESS[1]:                         fe80::3e58:c2ff:fe93:d429/64
IP6.GATEWAY:                            --
IP6.ROUTE[1]:                           dst = fe80::/64, nh = ::, mt = 256
```

Under random chance, during the day, other networks, frequently "Feline34" which is also mine appears also as OPEN. That network is a standard isp provided router and it's pretty old.

This issue is in fact not a dupe of https://github.com/espressif/esp-idf/issues/11202 as there are no Enterprise networks in the vicinity.

bill88t, 2023-11-09:
Retested with latest master (commit 294c2bd70d3fc6e5fefee627a7f2007f7b255709).
No changes (it's broken in master too).

AxelLin, 2023-11-16:
@sarveshb14  @nachiketkukade Any update? Will v5.1.2 fix this issue?

jack0c, 2023-11-20:
@AxelLin sorry that not fix this in v5.1.2. Currently esp-idf consider WPA/WPA2/WPA3 an invalid auth mode. Will consider how to fix this, and add it in v5.2 and v5.1.3.

sarveshb14, 2023-11-22:
Hi @bill88t @AxelLin,  Sorry for making you wait. 

As per [WPA3 Specification](https://www.wi-fi.org/file/wpa3-specification), `AP shall not enable WPA version 1 on the same BSS with WPA3-Personal modes`.

[image]

I agree that scan results should not show authentication mode of such AP as `OPEN`.

We will consider security of such AP as `WPA2 WPA3` (i.e. ignoring the weaker WPA version 1 security) and will provide an update for this soon.

bill88t, 2023-11-22:
`wpa_supplicant` is probably at fault.
I will try to file a bug to them too.

bill88t, 2023-11-23:
Also, the network `KeyFalse` is set to WPA2, not WPA3.
I don't know if NetworkManager does something on it's own, but my phone thinks it's WPA3 too.
[image]
Does WiFi4 even support WPA3???

sarveshb14, 2023-11-23:
Hi @bill88t , Can you please 
1) Check in which security mode does the esp-device gets connected with this AP and provide logs.
2) Provide wireless sniffer capture of the beacon transmitted by this AP.

ESP-device should be able to connect with WPA3-SAE security.
[image]
This is a screenshot from my local run with AP configured with WPA-WPA2-WPA3 mode.

---

>  I don't know if NetworkManager does something on it's own, but my phone thinks it's WPA3 too.

If phone sees `AKM suite SAE` and `management frame protection capabilities` in beacons of AP, phone will consider this AP as a WPA3 AP.  Packet capture will be helpful here in determining the security mode.

bill88t, 2023-11-23:
I ran the connection example from C6, S2 and C3, all of which say it's WPA2-PSK, which is correct.
[image]

The network configuration is as follows:
[image]

[image]

I also set my thinkpad in "monitor mode" and [captured](https://github.com/espressif/esp-idf/files/13450201/cap.zip) all the wifi traffic.

I do not know how to read it though.

sarveshb14, 2023-11-23:
Thank you providing the capture.
KeyFalse AP is on channel 1 while sniffer capture is of channel 10.
Can you please take sniffer capture of channel 1 or start the AP on channel 10 ?

bill88t, 2023-11-26:
> Can you please take sniffer capture of channel 1 or start the AP on channel 10 ?

Alright, sorry for the wait, I captured channel 1 this time. [ch1.zip](https://github.com/espressif/esp-idf/files/13467455/ch1.zip)
While the capture was running I connected to the network "KeyFalse" with an ESP32-C6 and the phone.

> Does this same configuration shows WPA WPA2 WPA3 security in the output of nmcli device wifi list ?

Yes.
[image]

bill88t, 2023-11-29:
Yes, it should use WPA2. I don't think WPA3 would work for these networks.

> We will consider security of such AP as WPA2 WPA3 (i.e. ignoring the weaker WPA version 1 security) and will provide an update for this soon.

Maybe it should be reported as `WPA2`, not `WPA2 WPA3`. I think it being detected as `WPA3` is a mistake.

sarveshb14, 2023-11-29:
> Maybe it should be reported as WPA2, not WPA2 WPA3

You are right. If AP does not follow WPA3, we will treat this AP as `WPA_WPA2` . If it follows WPA3, this will be treated as `WPA2_WPA3`

sarveshb14, 2024-01-11:
Hi @bill88t , we have merged above discussed changes into master branch now.
It will be available on next v5.1 tag (i.e. v5.1.3)

Thank you for your support

0xFEEDC0DE64, 2024-01-22:
is it already in master? we are also affected by this bug and I would like to know which commit in master fixed the issue to start my rebases on

bill88t, 2024-01-22:
I don't think so. I did test master 2 weeks ago with the latest commit and didn't see any changes.
I also can't find the related commit in the commit log.

It's possible the fix has been applied downstream and we will see it with the release. I don't know how espressif works.
In either case, I don't know.

sarveshb14, 2024-01-22:
Hi @0xFEEDC0DE64 , @bill88t . Sorry for making you wait. Fix is merged in internal repository. Push to github It is stalled due to internal CI. It will soon be available on github.

bill88t, 2024-01-22:
No problem, thanks for letting us know!

closed 2024-01-30 by espressif-bot

sarveshb14, 2024-01-31:
> It will soon be available on github.

It is available now. [(615d928a)](https://github.com/espressif/esp-idf/commit/615d928aa59315f81f052b34f59db7eb502e8199)

bill88t, 2024-01-31:
Can confirm, this fixed the issue. Thanks!

AxelLin, 2024-02-27:
> Hi @bill88t , we have merged above discussed changes into master branch now. It will be available on next v5.1 tag (i.e. v5.1.3)

@sarveshb14 v5.1.3 does **not** include this fix.

sarveshb14, 2024-02-27:
Hi @AxelLin @bill88t , We will  include this in next release of v5.1. Apologies for the inconvenience caused.

0xFEEDC0DE64, 2024-02-27:
I can confirm an update to latest esp-idf has fixed all problems for us, thank you