espressif/esp-idf #12518: usb_serial_jtag read/write issues

greenaddress, 2023-11-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1 and master b4268c874a4cf8fcf7c0c4153cffb76ad2ddda4e

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32s3 tdisplay s3

### Power Supply used.

USB

### What is the expected behavior?

read/write work more or less as uart driver functions

### What is the actual behavior?

uart via serial_jtag doesn't seem to work in v5.1.1 without first a new line output on stdout.

On master b4268c874a4cf8fcf7c0c4153cffb76ad2ddda4e even with that the code didn't work.

### Steps to reproduce.

With menuconfig setup with console output primary set to uart0 and secondary set to serial_jtag

```
#include <stdio.h>                                                              
#include "sdkconfig.h"                                                          
#include "freertos/FreeRTOS.h"                                                  
#include "freertos/task.h"                                                      
#include "driver/usb_serial_jtag.h"                                             
                                                                                
#include "esp_vfs_usb_serial_jtag.h"                                            
#include "esp_vfs_dev.h"                                                        
#include <fcntl.h>                                                              
                                                                                
                                                                                
/* master is b4268c874a4cf8fcf7c0c4153cffb76ad2ddda4e */                        
void app_main(void)                                                             
{                                                                               
    /*                                                                          
     * this doesn't help in master not in v5.1.1 for me                         
     *                                                                          
                                                                                
    setvbuf(stdin, NULL, _IONBF, 0);                                            
    esp_vfs_dev_usb_serial_jtag_set_rx_line_endings(ESP_LINE_ENDINGS_CR);       
    esp_vfs_dev_usb_serial_jtag_set_tx_line_endings(ESP_LINE_ENDINGS_CRLF);     
    fcntl(fileno(stdout), F_SETFL, O_NONBLOCK);                                 
    fcntl(fileno(stdin), F_SETFL, O_NONBLOCK);                                  
                                                                                
    */                                                                          
                                                                                
    usb_serial_jtag_driver_config_t config = USB_SERIAL_JTAG_DRIVER_CONFIG_DEFAULT();
    esp_err_t err = usb_serial_jtag_driver_install(&config);                    
    ESP_ERROR_CHECK(err);                                                       
                                                                                
    /*                                                                          
     * this doesn't help in master not in v5.1.1 for me                         
     *                                                                          
    esp_vfs_usb_serial_jtag_use_driver();                                       
    // this instead of esp_vfs_usb_serial_jtag_use_driver makes no difference in v5.1,1 nor master
    // esp_vfs_usb_serial_jtag_use_nonblocking();                               
    esp_vfs_dev_uart_register();                                                
                                                                                
    */                                                                          
    /* without this line the following loop doesn't do anything on v5.1.1.      
     * on master nothing happens regardless ? */                                
    putc('\n', stdout);                                                         
    /* with the line above on v5.1.1 i get echo back */                                                                                                                                                            
                                                                                
    uint8_t buf[64];                                                            
    while(1) {                                                                  
        int read = usb_serial_jtag_read_bytes(buf , sizeof(buf), 10);           
        assert(read < sizeof(buf));                                             
        if (read) {                                                             
            usb_serial_jtag_write_bytes(buf , read, 10);                        
        }                                                                       
    }                                                                           
} 

```

### More Information.

It seems a printf("something\n") with a new line is necessary for things to work on v5.1.1

A putc('\n', stdout); also works.

But writing a \n via jtag write_bytes didn't help

In master at b4268c874a4cf8fcf7c0c4153cffb76ad2ddda4e not even the new line or a printf helped.

Am I using the library wrong?

igrr, 2023-11-03:
What is the setting for primary and secondary console output you have in menuconfig?

Also, can you describe what you want to achieve? The code looks a bit odd, since first you are configuring a high-level method for accessing the console (via the VFS driver stdio) but then you are using low-level driver functions to read and write bytes. Understanding the goal behind this code might help with troubleshooting this.

greenaddress, 2023-11-03:
@igrr 

>What is the setting for primary and secondary console output you have in menuconfig?

I have the primary set to uart0 and secondary set to usb serial jtag port.

I also have tried with usb serial jtag port as primary.

> Also, can you describe what you want to achieve?

I want to use the esp32s3 as I am using an esp32 via the usb-serial converter chips, that includes using the "serial" for firmware flashing but also for application needs with binary data (i.e. not for logging).

I tried the software cdc which works great but when I use that I have to use the reset/boot button to flash which breaks in CI contexts without some additional hardware to emulate button pressing.

In the code I provided most of it is commented out - that's just to show things I tried that didn't work for me. 

If I understand correctly the vfs abstraction allows to use of the usb port as serial by reading/writing to a file unix style but I would rather use the functions available in usb_serial_jtag - but they don't work for me. It seems that in 5.1.1 without outputing first a new line, nothing work.

```putc('\n', stdout);``` or ```printf("\n");```

In master not even that works.

To show better what I mean let me remove the commented out code -I'd expect this code to work on v5.1.1 and it does not

```
#include <stdio.h>                                                              
#include "sdkconfig.h"                                                          
#include "freertos/FreeRTOS.h"                                                  
#include "freertos/task.h"                                                      
#include "driver/usb_serial_jtag.h"                                             
                                                                                
/* master is b4268c874a4cf8fcf7c0c4153cffb76ad2ddda4e */                                                                                                                                                           
void app_main(void)                                                             
{                                                                               
    usb_serial_jtag_driver_config_t config = USB_SERIAL_JTAG_DRIVER_CONFIG_DEFAULT();
    esp_err_t err = usb_serial_jtag_driver_install(&config);                    
    ESP_ERROR_CHECK(err);     

    // output here a new line with putc or printf to make things "work"                                                  
                                                                                
    uint8_t buf[64];                                                            
    while(1) {                                                                  
        int read = usb_serial_jtag_read_bytes(buf , sizeof(buf), 10);           
        assert(read < sizeof(buf));                                             
        if (read) {                                                             
            usb_serial_jtag_write_bytes(buf , read, 10);                        
        }                                                                       
    }                                                                           
} 
```

it starts working if *after* the driver_install you print a new line.

Thank you!

greenaddress, 2023-11-03:
I tested this with idf.py monitor as well as with screen

greenaddress, 2023-11-06:
I found a 'fix' for my last "minimal" example but doesn't look very pretty IMHO

adding `fflush` and `fsync` after `usb_serial_jtag_write_bytes`
```
            usb_serial_jtag_write_bytes(buf , read, 10);                        
            fflush(stdout);                                                                                                                                                                                        
            fsync(fileno(stdout));  
```

@igrr shouldn't the driver take care of this? v5.1.1

igrr, 2023-11-06:
You can get rid of fflush if you switch the stream to a non-buffered mode using `setvbuf` function.

I don't think we can get rid of fsync. The driver will flush the USB endpoint buffer every time it fills (64 bytes) or when it encounters a newline character. So for binary data, it's up to you to inform the driver that you'd like the data to be sent to the host immediately.

greenaddress, 2023-11-06:
@igrr I tried various things but it seems that calling fsync just after the install is enough and is not required to call it every time a write is done.

Do you agree? or do i need to do it after every write?