espressif/esp-idf #12519: idf.py menuconfig Get Error "Access violation"

shu7734, 2023-11-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Download idf5.1.1
decompress to D:\SysGCC\esp32\esp-idf\esp-idf-v5.1.1
and

PS D:\SysGCC\esp32\esp-idf\esp-idf-v5.1.1>install.ps1
..........
PS D:\SysGCC\esp32\esp-idf\esp-idf-v5.1.1>export.ps1
PS D:\SysGCC\esp32\esp-idf\esp-idf-v5.1.1> cd .\examples\get-started\hello_world\
PS D:\SysGCC\esp32\esp-idf\esp-idf-v5.1.1\examples\get-started\hello_world> idf.py menuconfig
Executing action: menuconfig
Running ninja in directory D:\SysGCC\esp32\esp-idf\esp-idf-v5.1.1\examples\get-started\hello_world\build
Executing "ninja menuconfig"...
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig
Compiler supported targets: xtensa-esp32-elf

-- App "hello_world" version: v5.1.1-dirty
-- Adding linker script D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/soc/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/app_trace D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/app_update D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/bootloader D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/bootloader_support D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/bt D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/cmock D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/console D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/cxx D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/driver D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/efuse D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp-tls D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_adc D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_app_format D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_coex D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_common D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_eth D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_event D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_gdbstub D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_hid D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_http_client D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_http_server D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_https_ota D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_https_server D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_hw_support D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_lcd D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_local_ctrl D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_mm D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_netif D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_netif_stack D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_partition D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_phy D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_pm D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_psram D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_ringbuf D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_rom D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_system D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_timer D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esp_wifi D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/espcoredump D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/esptool_py D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/fatfs D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/freertos D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/hal D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/heap D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/http_parser D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/idf_test D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/ieee802154 D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/json D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/log D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/lwip D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/main D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/mbedtls D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/mqtt D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/newlib D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/nvs_flash D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/openthread D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/partition_table D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/perfmon D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/protobuf-c D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/protocomm D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/pthread D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/sdmmc D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/soc D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/spi_flash D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/spiffs D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/tcp_transport D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/ulp D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/unity D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/usb D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/vfs D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/wear_levelling D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/wifi_provisioning D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/wpa_supplicant D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build
[0/1] cmd.exe /C "cd /D D:\SysGCC\esp32\esp-idf\esp-idf-v5.1.1\examples\get-started\hello_world\build && C:\Users\Admin\.espressif\pyth...DF_TARGET=esp32 --env IDF_ENV_FPGA= --output config D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig"
D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/Kconfig:16: warning: IDF_ENV_FPGA has 'option env="IDF_ENV_FPGA"', but the environment variable IDF_ENV_FPGA is not set
Loaded configuration 'D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig'
Access violation
FAILED: CMakeFiles/menuconfig D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/CMakeFiles/menuconfig
cmd.exe /C "cd /D D:\SysGCC\esp32\esp-idf\esp-idf-v5.1.1\examples\get-started\hello_world\build && C:\Users\Admin\.espressif\python_env\idf5.1_py3.12_env\Scripts\python.exe D:/SysGCC/esp32/esp-idf/esp-i [... cut ...] sp-idf-v5.1.1/Kconfig --sdkconfig-rename D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/sdkconfig.rename --config D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig --env-file D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/config.env --env IDF_TARGET=esp32 --env IDF_ENV_FPGA= --output config D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Dazza0, 2023-11-06:
@shu7734 Could you check if you have anti-virus enabled, and whether or not it can interfere with the build files?

shu7734, 2023-11-06:
I don't have any antivirus installed,or any system manage software.
And NTFS ACL is normal.

shu7734, 2023-11-06:
but , "idf.py build" is worked

jakub-kocka, 2023-11-06:
Hi @shu7734 can you please run `idf.py --version`?

shu7734, 2023-11-06:
> Hi @shu7734 can you please run `idf.py --version`?

V1.0.3

jakub-kocka, 2023-11-06:
Could you try to run  `. .\export.ps1`, after you should see version in format `ESP-IDF vX.Y...` and if so, your issue should not be there anymore.

shu7734, 2023-11-06:
PS D:\SysGCC\esp32\esp-idf\esp-idf-v5.1.1> .\export.ps1
Setting IDF_PATH: D:\SysGCC\esp32\esp-idf\esp-idf-v5.1.1
Checking Python compatibility
Adding ESP-IDF tools to PATH...
No directories added to PATH:

D:\SysGCC\esp32\esp-idf\esp-idf-v5.1.1\components\app_update
D:\SysGCC\esp32\esp-idf\esp-idf-v5.1.1\components\espcoredump
D:\SysGCC\esp32\esp-idf\esp-idf-v5.1.1\components\partition_table
C:\Users\Admin\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin
C:\Users\Admin\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin
C:\Users\Admin\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin
C:\Users\Admin\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin
C:\Users\Admin\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin
C:\Users\Admin\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin
C:\Users\Admin\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin
C:\Users\Admin\.espressif\tools\cmake\3.24.0\bin
C:\Users\Admin\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin
C:\Users\Admin\.espressif\tools\ninja\1.10.2\
C:\Users\Admin\.espressif\tools\idf-exe\1.0.3\
C:\Users\Admin\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64
C:\Users\Admin\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64
C:\Users\Admin\.espressif\python_env\idf5.1_py3.12_env\Scripts
D:\SysGCC\esp32\esp-idf\esp-idf-v5.1.1\tools
C:\Windows\system32
C:\Windows
C:\Windows\System32\Wbem
C:\Windows\System32\WindowsPowerShell\v1.0\
C:\Windows\System32\OpenSSH\
C:\Program Files (x86)\NetSarang\Xshell 7\
C:\Program Files (x86)\NetSarang\Xftp 7\
C:\Program Files\dotnet\
C:\Users\Admin\AppData\Local\Programs\Python\Python312\Scripts\
C:\Users\Admin\AppData\Local\Programs\Python\Python312\
C:\Users\Admin\AppData\Local\Programs\Python\Launcher\
D:\Program Files\PortableGit\bin
D:\Program Files\python-3.11.3-embed-amd64
C:\Users\Admin\AppData\Local\Programs\Fiddler
C:\Program Files\mitmproxy\bin
Checking if Python packages are up to date...
Constraint file: C:\Users\Admin\.espressif\espidf.constraints.v5.1.txt
Requirement files:
 - D:\SysGCC\esp32\esp-idf\esp-idf-v5.1.1\tools\requirements\requirements.core.txt
Python being checked: C:\Users\Admin\.espressif\python_env\idf5.1_py3.12_env\Scripts\python.exe
Python requirements are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:
    idf.py build

jakub-kocka, 2023-11-06:
Thank you, but the `.` before even if you are in right repository is necceasary.

dobairoland, 2023-11-06:
This is the command which has failed:
```
cmd.exe /C "cd /D D:\SysGCC\esp32\esp-idf\esp-idf-v5.1.1\examples\get-started\hello_world\build && C:\Users\Admin.espressif\python_env\idf5.1_py3.12_env\Scripts\python.exe D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/tools/kconfig_new/prepare_kconfig_files.py --list-separator=semicolon --env-file D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/config.env && C:\Users\Admin.espressif\python_env\idf5.1_py3.12_env\Scripts\python.exe -m kconfgen --list-separator=semicolon --kconfig D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/Kconfig --sdkconfig-rename D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/sdkconfig.rename --config D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig --env-file D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/config.env --env IDF_TARGET=esp32 --env IDF_ENV_FPGA= --dont-write-deprecated --output config D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig && C:\Users\Admin.espressif\python_env\idf5.1_py3.12_env\Scripts\python.exe D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/tools/check_term.py && C:\Users\Admin.espressif\tools\cmake\3.24.0\bin\cmake.exe -E env COMPONENT_KCONFIGS_SOURCE_FILE=D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/kconfigs.in COMPONENT_KCONFIGS_PROJBUILD_SOURCE_FILE=D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/kconfigs_projbuild.in KCONFIG_CONFIG=D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig IDF_TARGET=esp32 IDF_ENV_FPGA= C:/Users/Admin/.espressif/python_env/idf5.1_py3.12_env/Scripts/python.exe -m menuconfig D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/Kconfig && C:\Users\Admin.espressif\python_env\idf5.1_py3.12_env\Scripts\python.exe -m kconfgen --list-separator=semicolon --kconfig D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/Kconfig --sdkconfig-rename D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/sdkconfig.rename --config D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig --env-file D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/config.env --env IDF_TARGET=esp32 --env IDF_ENV_FPGA= --output config D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig"
```
There are actually serveral tools invoked in it. I would separate them and run them one after the other in order to receive a more informative error output than just "access violation".

```
cd /D D:\SysGCC\esp32\esp-idf\esp-idf-v5.1.1\examples\get-started\hello_world\build
```
```
C:\Users\Admin.espressif\python_env\idf5.1_py3.12_env\Scripts\python.exe D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/tools/kconfig_new/prepare_kconfig_files.py --list-separator=semicolon --env-file D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/config.env
```
```
C:\Users\Admin.espressif\python_env\idf5.1_py3.12_env\Scripts\python.exe -m kconfgen --list-separator=semicolon --kconfig D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/Kconfig --sdkconfig-rename D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/sdkconfig.rename --config D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig --env-file D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/config.env --env IDF_TARGET=esp32 --env IDF_ENV_FPGA= --dont-write-deprecated --output config D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig
```

```
C:\Users\Admin.espressif\python_env\idf5.1_py3.12_env\Scripts\python.exe D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/tools/check_term.py
```

```
C:\Users\Admin.espressif\tools\cmake\3.24.0\bin\cmake.exe -E env COMPONENT_KCONFIGS_SOURCE_FILE=D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/kconfigs.in COMPONENT_KCONFIGS_PROJBUILD_SOURCE_FILE=D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/kconfigs_projbuild.in KCONFIG_CONFIG=D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig IDF_TARGET=esp32 IDF_ENV_FPGA= C:/Users/Admin/.espressif/python_env/idf5.1_py3.12_env/Scripts/python.exe -m menuconfig D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/Kconfig && C:\Users\Admin.espressif\python_env\idf5.1_py3.12_env\Scripts\python.exe -m kconfgen --list-separator=semicolon --kconfig D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/Kconfig --sdkconfig-rename D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/sdkconfig.rename --config D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig --env-file D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/config.env --env IDF_TARGET=esp32 --env IDF_ENV_FPGA= --output config D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig
```

shu7734, 2023-11-06:
```
C:\Users\Admin.espressif\tools\cmake\3.24.0\bin\cmake.exe -E env COMPONENT_KCONFIGS_SOURCE_FILE=D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/kconfigs.in COMPONENT_KCONFIGS_PROJBUILD_SOURCE_FILE=D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/kconfigs_projbuild.in KCONFIG_CONFIG=D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig IDF_TARGET=esp32 IDF_ENV_FPGA= C:/Users/Admin/.espressif/python_env/idf5.1_py3.12_env/Scripts/python.exe -m menuconfig D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/Kconfig

```
rethrn

```
D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/Kconfig:16: warning: IDF_ENV_FPGA has 'option env="IDF_ENV_FPGA"', but the environment variable IDF_ENV_FPGA is not set
Loaded configuration 'D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig'
Access violation
```

shu7734, 2023-11-06:
I unpark idf5.1.1 to H:
H: is exfat format
cd h:\esp-idf-v5.1.1\examples\get-started\hello_world\
idf.py menuconfig
get same error

jakub-kocka, 2023-11-06:
I have been able to reproduce this issue, although when used `. .\export.ps1` and the ESP-IDF version was correct I can run `idf.py menuconfig` and every command above without issues.
I have:
- run `. .\export.ps1` -> checked that the `idf.py --version` gives back right format: **ESP-IDF v5.X.....**
- run every separated command [mentioned](https://github.com/espressif/esp-idf/issues/12519#issuecomment-1794168015)
-> with one change .. when the export finishes right the PATH should be correct and you don't need to specify Python interpreter 

So the change I have made to commands:
**prepare_kconfig_files**

      python D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/tools/kconfig_new/prepare_kconfig_files.py --list-separator=semicolon --env-file D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/config.env      

**kconfgen**

      python -m kconfgen --list-separator=semicolon --kconfig D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/Kconfig --sdkconfig-rename D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/sdkconfig.rename --config D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig --env-file D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/config.env --env IDF_TARGET=esp32 --env IDF_ENV_FPGA= --dont-write-deprecated --output config D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig

**check_term**

     python D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/tools/check_term.py

**cmake**
    
     cmake -E env COMPONENT_KCONFIGS_SOURCE_FILE=D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/kconfigs.in COMPONENT_KCONFIGS_PROJBUILD_SOURCE_FILE=D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/kconfigs_projbuild.in KCONFIG_CONFIG=D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig IDF_TARGET=esp32 IDF_ENV_FPGA=python -m menuconfig D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/Kconfig

**kconfgen**

      python -m kconfgen --list-separator=semicolon --kconfig D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/Kconfig --sdkconfig-rename D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/sdkconfig.rename --config D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig --env-file D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/config.env --env IDF_TARGET=esp32 --env IDF_ENV_FPGA= --output config D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig

One more thing I have done in addition is `git submodule update --init --recursive`

shu7734, 2023-11-06:
```
PS D:\SysGCC\esp32\esp-idf\esp-idf-v5.1.1\examples\get-started\hello_world> cmake -E env COMPONENT_KCONFIGS_SOURCE_FILE=D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/kconfigs.in COMPONENT_KCONFIGS_PROJBUILD_SOURCE_FILE=D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/kconfigs_projbuild.in KCONFIG_CONFIG=D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig IDF_TARGET=esp32 IDF_ENV_FPGA=python -m menuconfig D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/Kconfig
cmake -E env: unknown option '-m'
PS D:\SysGCC\esp32\esp-idf\esp-idf-v5.1.1\examples\get-started\hello_world> cmake --version
```

jakub-kocka, 2023-11-06:
Sorry, made a typo

       cmake -E env COMPONENT_KCONFIGS_SOURCE_FILE=D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/kconfigs.in COMPONENT_KCONFIGS_PROJBUILD_SOURCE_FILE=D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/kconfigs_projbuild.in KCONFIG_CONFIG=D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig IDF_TARGET=esp32 IDF_ENV_FPGA= python -m menuconfig D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/Kconfig

jakub-kocka, 2023-11-06:
Just a tip. If you have installed ESP-IDF with idf-installer from https://dl.espressif.com/dl/esp-idf/ you should have also prompt and PowerShell environment installed as well and if run from this, you don't need to run `export.ps1`.

If you already have installed ESP-IDF, you can select option "Use an existing ESP-IDF directory" in the process of installation  with "Universal Online Installer" then the launchers should install.

shu7734, 2023-11-06:
```
PS D:\SysGCC\esp32\esp-idf\esp-idf-v5.1.1\examples\get-started\hello_world> cmake -E env COMPONENT_KCONFIGS_SOURCE_FILE=D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/kconfigs.in COMPONENT_KCONFIGS_PROJBUILD_SOURCE_FILE=D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/build/kconfigs_projbuild.in KCONFIG_CONFIG=D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig IDF_TARGET=esp32 IDF_ENV_FPGA= python -m menuconfig D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/Kconfig
D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/Kconfig:16: warning: IDF_ENV_FPGA has 'option env="IDF_ENV_FPGA"', but the environment variable IDF_ENV_FPGA is not set
Loaded configuration 'D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/examples/get-started/hello_world/sdkconfig'
Access violation
```

shu7734, 2023-11-06:
I tried to delete folder C:\Users\Admin\.espressif and D:/SysGCC/esp32/esp-idf/esp-idf-v5.1.1/
and unpack reinstall idf
and get the same error

jakub-kocka, 2023-11-06:
That is strange, have you tried to install the ESP-IDF with idf-installer?

Maybe it could be linked with [Execution Policies](https://learn.microsoft.com/en-gb/powershell/module/microsoft.powershell.core/about/about_execution_policies?view=powershell-7.3), could you try to set it manually?

      Set-ExecutionPolicy -ExecutionPolicy RemoteSigned -Scope CurrentUser

Phasmatrix, 2023-11-06:
@jakub-kocka I'm currently receiving the same error as @shu7734. Windows 11 Home machine where I downloaded and installed ESP-IDF through the VSCode plugin.

Tried all of the above fixes, but none worked. Deleted all of the installation-related folders and files and tried re-installing but I keep getting the same error. On my other machine, this was working perfectly fine. I will investigate further and let you know.

Phasmatrix, 2023-11-06:
@jakub-kocka I tried installing the ESP-IDF framework once more using the VSCode plugin after deleting the _.espressif_ and _esp_ folders and also making sure the extension is deleted from the _.vscode_ folder. I have tried both using the Espressif server and GitHub server and both seem to be giving me the same error. Can it be that the below has something to do with our problem? On installing one of the prerequisities is that we need IDF-Python (which refers to a Python version 3.11.2 - see the attached picture below).

[image]

However, when using the install script after downloading ESP-IDF using the framework, we see that on creating the Python virtual environment, a Python version 3.12 is used. Can this be of any issues regarding the idf.py menuconfig command? All of the other commands seem to work.

[image]

shu7734, 2023-11-07:
@Phasmatrix @jakub-kocka Thanks
I tried uninstall python 3.12 , and install python 3.11.2, is worked now.
thanks for all

jakub-kocka, 2023-11-07:
I am glad, that it works @Phasmatrix @shu7734.

It looks like it will need a further investigation, @Phasmatrix. Thanks for pointing onto this.

Phasmatrix, 2023-11-07:
I am glad that everything works fine now @shu7734. One more thing I want to add to this issue that may be relevant for future issues concerning the Python version and installation - and more importantly also solved my issue.

I did some digging and found that on installing the ESP-IDF framework a custom virtual environment is being installed according to the IDF-Python pre-required version. The installation guide goes through this process no matter what, the virtual environment will always be created. However in the, probably rare, case that a [newer version of Python](https://www.python.org/downloads/release/python-3120/?ref=upstract.com) is released before the IDF-Python gets updated it is possible that during the installation process another virtual environment will be created matching the latest Python version. In my case this is what happened, I installed the latest version of Python being **3.12** whilst IDF-Python was still on version **3.11.2**.

All of the above however is not enough to reproduce our issues with the initial _Access violation_ error. On installing a certain Python version using their installer you can select the option to add the Python executable and scripts directory to your PATH. When this is checked you get the benefit of calling the Python executable directly from the command line, but this also messes up the installation (Powershell) script for ESP-IDF. I did not look for the exact lines of code in the scripts where this happens but basically the following happens:

* Installation of ESP-IDF begins
* Pre-required packages are being installed
* ESP-IDF is being installed
* Other tools are being installed
* Python virtual environment is being installed -> here it installs the environment for both the IDF-Python version and all versions listed in the PATH variables

When executing the export.ps1 script to export the Python virtual environment to you current working directory, it automatically picks the either the Python listed in PATH or the latest version - of this I am not sure.

**My fix in short:** I deleted my Python installation (deleting the PATH variables should be enough but I decided to work with MiniConda virtual environments from now on as well so I don't need this anymore). Then reinstalled ESP-IDF through the extensions menu in VSCode and everything works as expected.

Hope this solves some future issues @jakub-kocka and may be interesting for @shu7734 as well to change his configuration/PATH variables.

dobairoland, 2023-11-13:
So if I understood the situation well then the issue is cause by

1. using a system Python,
2. installing ESP-IDF which creates a virtual environment with Python from step 1,
3. updating system Python,
4. if the Python from 1 is uninstalled then of course the the virtual environment installed in step 2 will be broken.

This would happen on all operating systems and it is not limited to Windows or menuconfig. I don't see what we could do to avoid this.

The recommendation is to use the Windows tool installer to set up a working environment and use the Python provided with it. 
I you use IDE (e.g. vscode) then follow the installation steps provided there.

closed 2023-11-13 by dobairoland

Nielotz, 2024-01-11:
I am getting the same error having only Python 3.12.0 (without version change after ESP-IDF install)
win 10: 10.0.19045
esd-idf version and source: https://dl.espressif.com/github_assets/espressif/esp-idf/releases/download/v5.1.2/esp-idf-v5.1.2.zip

I installed python 3.11 WITHOUT setting the path and then I overwritten default python path only in powershell `$env:Path = 'D:\Python\3.11.7;' + $env:Path`, leaving 3.12 as default system interpreter.

It looks like its related to python 3.12

ecochard, 2024-03-15:
I got the same error but only with VSCode powershell terminal.

In vscode settings.json check that you have:

```json
"terminal.integrated.profiles.windows": {
		"Git Bash": {
			"source": "Git Bash"
		},
		"PowerShell-IDF": {
			"source": "PowerShell",
			"args": ["-ExecutionPolicy", "Bypass", "-NoExit", " -File", "YOUR_IDF_INSTALL_PATH/Initialize-Idf.ps1"]
		}
	}
```