espressif/esp-idf #12520: Linux unit test supports gets worse

higaski, 2023-11-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-3775-gb4268c874a

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

It seems that support for running [ESP-IDF applications on the host](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/host-apps.html#) gets worse instead of better. Take the [hello_world_linux_compatible](https://github.com/espressif/esp-idf/tree/master/tools/test_apps/linux_compatible/hello_world_linux_compatible) app for example.

It used to work on a x86 build that you could at least define a GPIO port by including
```c
#include <hal/gpio_types.h> 
```

And then somewhere in the code use any of the usual GPIO definitions.
```c
int a0_gpio_num = GPIO_NUM_7;
```

This no longer works. The definition of GPIO_NUM_7 is no longer found
```sh
/home/vinci/Develop/VSCode/hello_world_linux_compatible/main/hello_world_main.c:18:23: error: ‘GPIO_NUM_7’ undeclared (first use in this function)
   18 |     int a0_gpio_num = GPIO_NUM_7;
```

I'm used to some of the peripheral definitions missing (e.g. ADC stuff) but at the current state it simply became unusable. You can no longer maintain a single header file as project configuration because not even the pin-out definitions are available.

Now, don't get me wrong, I know that practically no other MCU vendor out there is offering mocks, let alone simulated peripherals. But you've got to either fix those issues or the [Mocks/Simulation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/host-apps.html#component-linux-mock-support-overview) paragraph in the documentation.

Please also let me know if you're at all dedicated to get any of this working. Because if you're not, I might need to rethink my testing strategy which currently is 100% host only.

### What is the actual behavior?

.

### Steps to reproduce.

.

0xjakob, 2023-11-03:
Hi @higaski! 

Thanks for reporting this, this is an actual issue with recent changes in IDF, which we will fix!

I'm happy to hear you are trying to test 100% on host. This has been a major goal of the Linux target in IDF. If you have any other issue with the Linux target, please let us know.

Some heads-up: in the very latest master, which will be merged soon, the [hello_world_linux_compatible](https://github.com/espressif/esp-idf/tree/master/tools/test_apps/linux_compatible/hello_world_linux_compatible) and the "normal" hello-world example have been merged. The former has been removed. You will be able to set the linux target in the original hello-world application and run it.

0xjakob, 2023-11-07:
The fix has been merged internally and will be synced to the github master branch soon. If you need it urgently, here's a preliminary patch: 
[0001-fix-driver-gpio-number-definitions-on-Linux.zip](https://github.com/espressif/esp-idf/files/13276241/0001-fix-driver-gpio-number-definitions-on-Linux.zip)

closed 2023-11-09 by espressif-bot