espressif/esp-idf #12526: I2S Read zero

ruielias, 2023-11-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp32@2.0.11

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

In the Arduino IDE, with the version "platform esp32:esp32@1.0.6" I was able to read values ​​from the INMP441 sensor.
I updated to version "esp32:esp32@2.0.11", and I can no longer obtain data from the sensor...

My CODE:

`#include <Arduino.h>
#include <driver/i2s.h>

// Defenição INMP
#define I2S_SCK 14
#define I2S_WS 15
#define I2S_SD 32
#define bufferLen 1024
int32_t sBuffer[bufferLen];
double vReal[bufferLen];

void i2s_install() {
  // Set up I2S Processor configuration
  static const i2s_config_t i2s_config = {
    .mode = i2s_mode_t(I2S_MODE_MASTER | I2S_MODE_RX),
    .sample_rate = 16000,
    .bits_per_sample = I2S_BITS_PER_SAMPLE_32BIT,
    .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,
    .communication_format = i2s_comm_format_t(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB),
    .intr_alloc_flags = 0,
    .dma_buf_count = 2,
    .dma_buf_len = bufferLen,
    .use_apll = false
  };
  if (ESP_OK == i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL))
    Serial.printf("\nDrive OK");
}

void i2s_setpin() {
  // Set I2S pin configuration
  static const i2s_pin_config_t pin_config = {
    .bck_io_num = I2S_SCK,
    .ws_io_num = I2S_WS,
    .data_out_num = -1,
    .data_in_num = I2S_SD
  };

  if (ESP_OK == i2s_set_pin(I2S_NUM_0, &pin_config))
    Serial.printf("\nPIN OK");
}

void setup() {
  // Set up Serial Monitor
  Serial.begin(115200);
  Serial.printf("\n Setup Start");
  // Set up I2S
  i2s_install();
  i2s_setpin();
  i2s_start(I2S_NUM_0);
}
void loop() {
  uint32_t pico=0;
  int16_t samples_read;

  // Get I2S data and place in data buffer
  size_t bytesIn = 0;
  esp_err_t result = i2s_read(I2S_NUM_0, &sBuffer, sizeof(sBuffer), &bytesIn, portMAX_DELAY);
    if (result == ESP_OK)
    {
      // Read I2S data buffer
      samples_read = bytesIn / 4;
      if (samples_read > 0)
      {
        for (int16_t i = 0; i < samples_read; ++i)
        {
          vReal[i]=sBuffer[i]>>8;
          if (vReal[i]>pico)
            pico=vReal[i];
        }
        Serial.printf("\nP: %ld",pico);
      }
    }
}`

### What is the actual behavior?

How do I update to a more current version? Will it resolve?

### Steps to reproduce.

1. Step
2. Step
3. Step
...

L-KAYA, 2023-11-13:
There is another similar issue might help, the configurations might need update for INMP441. 

https://github.com/espressif/esp-idf/issues/12296

And if possible, please migrate to the new I2S driver since IDF v5.0, the new driver simplifies the usage and is more stable.

Alvin1Zhang, 2025-01-13:
Thanks for reporting, feel free to reopen with more updates.

closed 2025-01-13 by Alvin1Zhang