espressif/esp-idf #12528: ESP32C6 Task watchdog got triggered when PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP=y && MBEDTLS_HARDWARE_AES=y && esp_sleep_cpu_retention_init()

AxelLin, 2023-11-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1-686-gb6a66b7d8c

### Espressif SoC revision.

ESP32C6

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

 Task watchdog should not be triggered.

### What is the actual behavior?

@esp-lis

I notice a problem with this modification: https://github.com/espressif/esp-idf/issues/11858#issuecomment-1640244734

I use a modifed code of examples/system/light_sleep to reproduce the issue.

What the modified code does is:
1. Call esp_sleep_cpu_retention_init()
2. Make device enter sleep/wake_up/do_something... join sta, disconnect and stop wifi, then enter sleep again.
3. Observe Task watchdog got triggered, see attached log.

PS. You need to change the EXAMPLE_ESP_WIFI_SSID and EXAMPLE_ESP_WIFI_PASS in light_sleep_example_main.c to meet your test settings.

Note, this issue only happens when
PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP=y && MBEDTLS_HARDWARE_AES=y && esp_sleep_cpu_retention_init() with change in https://github.com/espressif/esp-idf/issues/11858#issuecomment-1640244734 .

### Steps to reproduce.

Just build examples/system/light_sleep with the modified code.

Attached the sdkconfig and modified light_sleep_example_main.c
[light_sleep_example_main.c.log](https://github.com/espressif/esp-idf/files/13263380/light_sleep_example_main.c.log)
[sdkconfig.log](https://github.com/espressif/esp-idf/files/13263381/sdkconfig.log)

### Debug Logs.

```plain
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
[... 314 lines cut ...]
Core 6 register dump:
MEPC    : 0x4201c900  RA      : 0x4201c906  SP      : 0x408290b0  GP      : 0x40815a24  
0x4201c900: esp_aes_wait_dma_done at /home/axel/esp/esp-idf/components/mbedtls/port/aes/dma/esp_aes.c:118
 (inlined by) esp_aes_dma_wait_complete at /home/axel/esp/esp-idf/components/mbedtls/port/aes/dma/esp_aes.c:239
 (inlined by) esp_aes_process_dma at /home/axel/esp/esp-idf/components/mbedtls/port/aes/dma/esp_aes.c:454

0x4201c906: esp_aes_wait_dma_done at /home/axel/esp/esp-idf/components/mbedtls/port/aes/dma/esp_aes.c:118
 (inlined by) esp_aes_dma_wait_complete at /home/axel/esp/esp-idf/components/mbedtls/port/aes/dma/esp_aes.c:239
 (inlined by) esp_aes_process_dma at /home/axel/esp/esp-idf/components/mbedtls/port/aes/dma/esp_aes.c:454

TP      : 0x40806b84  T0      : 0xfffff000  T1      : 0xff000fff  T2      : 0x80000000  
0x40806b84: clk_ll_cpu_get_ls_divider at /home/axel/esp/esp-idf/components/hal/esp32c6/include/hal/clk_tree_ll.h:441
 (inlined by) rtc_clk_cpu_freq_get_config at /home/axel/esp/esp-idf/components/esp_hw_support/port/esp32c6/rtc_clk.c:287

S0/FP   : 0x00000010  S1      : 0x40821c18  A0      : 0x00000000  A1      : 0x00021c18  
A2      : 0x00000001  A3      : 0x60088000  A4      : 0x00000002  A5      : 0x00000002  
A6      : 0x00000010  A7      : 0x00000010  S2      : 0x40829130  S3      : 0x00000010  
S4      : 0x00000000  S5      : 0x408314d5  S6      : 0x00000000  S7      : 0x00000001  
S8      : 0x40821c0c  S9      : 0x40821c0c  S10     : 0x40821c18  S11     : 0x00000000  
T3      : 0x00000ffc  T4      : 0x00000fff  T5      : 0xbfffffff  T6      : 0xdfffffff  
MSTATUS : 0x4081d7a0  MTVEC   : 0x00000022  MCAUSE  : 0x00000000  MTVAL   : 0x40800a2c  
0x40800a2c: heap_caps_malloc_base at /home/axel/esp/esp-idf/components/heap/heap_caps.c:177

MHARTID : 0x40829120
```

esp-wzh, 2023-12-01:
Thanks for report it, this is a known problem. It is caused by the loss of GDMA configuration after peripheral powerdown lightsleep. The feature that supports GDMA sleep retention is under internal review and will be merged soon.

AxelLin, 2024-01-06:
@esp-wzh  Still hit this issue with v5.1.2-602-gdb1e54a0c5.

esp-wzh, 2024-01-07:
@AxelLin This patch supports gdma retention and you can apply it at first.

[0001-fix-esp_hw_support-support-gdma-register-context-retention.patch.zip](https://github.com/espressif/esp-idf/files/13852534/0001-fix-esp_hw_support-support-gdma-register-context-retention.patch.zip)

AxelLin, 2024-01-07:
@esp-wzh 
Thanks.
I just tested the patch and it looks ok now.
Hope to get it merged soon.

Alvin1Zhang, 2024-02-20:
Thanks for reporting, fix on master branch is available at https://github.com/espressif/esp-idf/commit/5454d37d496a8c58542eb450467471404c606501, and backport on release/5.1 is in progress, will sync up once fix on release/5.1 is synced, thanks.

closed 2024-03-13 by AxelLin