espressif/esp-idf #12530: [mesh][v4.4] DHCP Server not propagating backup DNS server

KonssnoK, 2023-11-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello @zhangyanjiaoesp ,
we were having issues with one of our internal networks, where only the root node was able to connect to the external world and all the children would have unresolved DNS.

We found out that the dhcps implementation in 4.4 does not support propagation of the backup DNS server.

Our router broadcasts the DNS only in the secondary DNS slot, because the first one is dedicated to the firewall.

We created a patch for the dhcps, but it would make sense for this to be implemented in general.

Also, we cannot find a way to define FALLBACK_DNS_SERVER_ADDRESS apart from putting it in the Cmakefile.
How is it supposed to be setup?

Thanks

xueyunfei998, 2023-11-06:
hi  @KonssnoK 
You can use this interface `esp_netif_set_dns_info`

`esp_netif_dns_type_t = ESP_NETIF_DNS_FALLBACK`

KonssnoK, 2023-11-06:
hello @xueyunfei998 
this interface would be ok for the children but not for the root. When DHCPS is enabled, it is impossible to set the fallback DNS.
```

    if (esp_netif->flags & ESP_NETIF_DHCP_SERVER) {
#if ESP_DHCPS
        // if DHCP server configured to set DNS in dhcps API
        if (type >= ESP_NETIF_DNS_FALLBACK) {
            ESP_LOGD(TAG, "set dns invalid type");
            return ESP_ERR_ESP_NETIF_INVALID_PARAMS;
        } else {
            dhcps_dns_setserver(lwip_ip, type);
        }
#else
        LOG_NETIF_DISABLED_AND_DO("DHCP Server", return ESP_ERR_NOT_SUPPORTED);
#endif
    } else {
        dns_setserver(type, lwip_ip);
    }
```

freakyxue, 2023-11-14:
hi  @KonssnoK 
Please add this patch.

[dns.txt](https://github.com/espressif/esp-idf/files/13348808/dns.txt)

KonssnoK, 2023-11-14:
im not sure that patch can be used without 
https://github.com/espressif/esp-lwip/pull/62

Anyway, 
we changed the DHCPS to support also the backup DNS server, not sure if you are planning on adding this feature to 4.4 or not.

freakyxue, 2023-11-15:
hi  @KonssnoK 
This patch can be used in 4.4. This macro is defined by the customer, not a feature.

KonssnoK, 2023-11-15:
@freakyxue problem is lwip does not compile if you use `FALLBACK_DNS_SERVER_ADDRESS` and only IPV4 (see issue above).

Anyway, let us know the plans for DHCPs backup server support, otherwise we will continue using our patch.
Thanks!

KonssnoK, 2024-01-19:
for reference, the issue in lwip has been fixed.
What remains to understand is plans on the DNS Server, which still does not propagate backup DNS.

zhangyanjiaoesp, 2024-01-22:
Will merge the fix ASAP, thanks for the reminder.

closed 2024-02-04 by espressif-bot