espressif/esp-idf #12531: ESP32S3 - 24bit PDM?

timsterc, 2023-11-06:
### Is your feature request related to a problem?

Currently using 16bit PDM RX with a Knowles MEMs PDM microphone - just wondering if there are any plans to support 24bit PDM at any point in the future w/ a software change, or if the 16bit limit is ESP32S3 hardware?

L-KAYA, 2023-11-07:
Sorry that the `PDM2PCM` module is a hardware feature and it is fixed to filter the raw PDM data into 16-bit PCM data. So 24-bit PDM RX can't  be supported

L-KAYA, 2025-01-21:
Update: now I2S PDM mode support to receive raw format without PDM2PCM converter (see the commit [9b779d8b](https://github.com/espressif/esp-idf/commit/9b779d8b3c1c7d7aff73ebe5ef36d03ae3d60141)). So you can try to convert the raw PDM data into 24-bit PCM format in the software.

closed 2025-01-21 by Alvin1Zhang

Arcitec, 2025-09-16:
@L-KAYA Thanks, that new "raw PDM" feature is interesting. But there's probably never going to be any reason for anyone to use more than 16-bit audio recordings in a microcontroller. And it would cost significant ESP32 CPU resources to do the PDM to PCM conversion in software.

16-bit is perfect for almost 100% of use-cases in a microcontroller.

24-bit is where you get into professional audio/studio audio territory, and then you'll be dealing with way better microphones than I2S/PDM garbage.

The amount of bits per sample is really just controlling how low the noise floor is. There's a reason why pretty much all music forever has been released at 16-bit. Because it has a huge dynamic range. 24-bit is only needed during music production in the workstations to extend the noise floor when doing DSP mixing.

So again: If someone thinks they need 24-bit audio in their ESP32 project, I seriously question their sanity or understanding of audio. 💃 Relax and enjoy the 16-bit life.