espressif/esp-idf #12533: esp.emac: no mem for receive buffer

hzhh110, 2023-11-07:
### Is your feature request related to a problem?

[WiFiGeneric.cpp:1438] hostByName(): DNS Failed for

When esp.emac: no mem for receive buffer is displayed, subsequent devices cannot be recovered, even if I release the stack

### Describe the solution you'd like.

As long as I start using Bluetooth communication, the network cable can not connect to the Internet, but the LAN can communicate, or connect to the Internet line communication, I go to use Bluetooth again, after a while the same, and then I try to release the Bluetooth FreeHeap Size:179336 after 120 seconds, but the network cable keeps appearing hostByName(): DNS Failed for and cannot be recovered until the device is restarted

hzhh110, 2023-11-07:
FreeHeap BLE1 :191456 
initBLEJH-P130-A0009
FreeHeap BLE2 :105288 
void initBLE(const char *name)
{
  DebugPrint("initBLE");
  DebugPrintln(name);
  BLEDevice::init(name);
  #if isESP32
   BLEDevice::setPower(ESP_PWR_LVL_P7,ESP_BLE_PWR_TYPE_DEFAULT); 
  #else
  // #if isESP32_C3
   BLEDevice::setPower(ESP_PWR_LVL_P18,ESP_BLE_PWR_TYPE_DEFAULT);
  #endif
  BLEServer *pServer = BLEDevice::createServer();
  _pServer =  pServer;
  pServer->setCallbacks(new MyServerCallbacks());
  BLEService *pService = pServer->createService(SERVICE_UUID);
  pTXCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID_TX, BLECharacteristic::PROPERTY_NOTIFY);
  pTXCharacteristic->addDescriptor(new BLE2902());
  BLECharacteristic *pRXCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID_RX, BLECharacteristic::PROPERTY_WRITE);
  pRXCharacteristic->setCallbacks(new MyCallbacks());
  pService->start();
  pServer->getAdvertising()->start();
}

hzhh110, 2023-11-08:
I wonder if it is possible that I set ETH.config(), then removed it, and then the router recorded it, causing the problem? And how do I get dns1 and dns2 in config(IPAddress local_ip, IPAddress gateway, IPAddress subnet, IPAddress dns1, IPAddress dns2)? Why am I getting Intranet ones and not something like 114.114.114.114 or 8.8.8.8?

hzhh110, 2023-11-08:
E (9267) esp.emac: no mem for receive buffer
E (9271) esp.emac: no mem for receive buffer
[WiFiGeneric.cpp:1438] hostByName(): DNS Failed

If [WiFiGeneric.cpp:1438] hostByName(): DNS Failed appears, it cannot be restored
You can take a router, and then the router is connected to the Internet at the beginning, after the router's Internet connection network cable is removed, you will find that you can not reconnect, has been DNS Failed

kostaond, 2023-11-10:
> Why am I getting Intranet ones and not something like 114.114.114.114 or 8.8.8.8

It depends on DHCP Server configuration and on your internet provider.

Could you please try to use IP address instead of domain name?

> E (9267) esp.emac: no mem for receive buffer

Unfortunately, it seems that `no mem for receive buffer` may result in Ethernet to hang under specific conditions - it needs to occur multiple times in a row and so fill Ethernet DMA Rx buffer. The issue was fixed in [ESP-IDF v5.x](https://github.com/espressif/esp-idf/commits/6fff81d9700c6f615b711f7581329eea3566f32a). I'll try to backport the fix to v4.4 but it will take some time till it gets to the ESP Arduino 2.x. In the meantime, you can try the following:
* Try to use ESP32 Arduino 3.0 Alpha version https://github.com/espressif/arduino-esp32/releases/tag/3.0.0-alpha2
* Fix the EMAC driver locally based on [this commit](https://github.com/espressif/esp-idf/commits/6fff81d9700c6f615b711f7581329eea3566f32a)
* Increase number of Ethernet DMA Rx buffers ([`ETH_DMA_RX_BUFFER_NUM`](https://github.com/espressif/esp-idf/blob/master/components/esp_eth/Kconfig#L87)) and better manage memory of your application (to have more free memory during a runtime)