espressif/esp-idf #12535: [MESH][v4.4] Node unable to connect to ROOT or AP on low ROOT RSSI

KonssnoK, 2023-11-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello @zhangyanjiaoesp .
This is a strange sidecase we found in our systems.

v4.4.6-103-g5daf5610b9

Setup
- One hotspot
- One device with WIFI and LTE (for now LTE not necessary). This device has a esp_wifi_set_max_tx_power set to 2dB (8)
- One device with esp_wifi_set_max_tx_power set to 14dB

I currently have the hotspot put next to the device with 14dB.
The device with 2dB is put at 8 meters from the phone and the other device.

Powering:
- Enable hotspot
- Enable device with 2dB
- Wait for 2dB device to connect to the hotspot. My device sees -63dB
```
network {
  mesh_layer: 1
  mesh_p_rssi: -63
  mesh_p_mac: 542
  modem_state: MODEM_ON_AT_READY
  modem_rssi: -200
}
```
- Switch on device 14dB next to the hotspot (60cm away)

What i see:
- The device starts looping between
```
W (573814) wifi:Haven't to connect to a suitable AP now!
I (574219) wifi:I (01:09:33.462) mesh_main: <MESH_EVENT_NO_PARENT_FOUND>scan times:60
mode : sta (7c:df:a1:e0:74:40)
W (01:09:33.464) mesh_hand: Perform NO PARENT FIXED ROOT handover 1 1
W (01:09:33.470) mesh_hand: Triggering FIXED ROOT handover
I (574234) wifi:mode : sta (7c:df:a1:e0:74:40) + softAP (7c:df:a1:e0:74:41)
W (574240) wifi:<MESH AP>adjust channel:11, secondary channel offset:2(40D)
I (574249) wifi:Total power save buffer number: 8
```
```
I (01:10:08.979) mesh_main: <MESH_EVENT_SCAN_DONE>number:26
W (01:10:08.981) mesh_hand: Triggering DYNAMIC MESH handover
W (609743) wifi:scan number is 0
W (01:10:08.988) mesh_hand: scan_done: found 0, fixed_root 0, has_parent 0
W (01:10:08.992) mesh_main: <MESH_EVENT_ROOT_FIXED>not fixed
W (01:10:08.998) mesh_hand: Triggering DYNAMIC MESH handover
I (623842) wifi:mode : sta (7c:df:a1:e0:74:40)
I (01:10:23.086) mesh_main: <MESH_EVENT_NO_PARENT_FOUND>scan times:60
I (01:10:27.300) mesh_main: <MESH_EVENT_FIND_NETWORK>new channel:11, router BSSID:000000000000
```

without ever connecting

[no_connection.txt](https://github.com/espressif/esp-idf/files/13278316/no_connection.txt)

Any help?
Thanks

KonssnoK, 2023-11-07:
i think we can forget about the "fixed root handover"
Once the device 14dB is back to dynamic search is not going to manage to find the hotspot, even if next to it, looping with the following logs:
```

I (112784) wifi:mode : sta (7c:df:a1:e0:74:40)I (01:01:52.028) mesh_main: <MESH_EVENT_NO_PARENT_FOUND>scan times:60, 2

I (01:01:52.067) processor: send status... (1000ms)
W (112824) wifi:Haven't to connect to a suitable AP now!
I (01:01:53.066) processor: send status... (1000ms)
W (113824) wifi:Haven't to connect to a suitable AP now!
I (01:01:54.066) processor: send status... (999ms)
W (114824) wifi:Haven't to connect to a suitable AP now!
W (01:01:55.066) monitor: 3W3P:: rule deactivated
I (116999) wifi:I (01:01:56.242) mesh_main: <MESH_EVENT_FIND_NETWORK>new channel:11, router BSSID:000000000000
mode : sta (7c:df:a1:e0:74:40) + softAP (7c:df:a1:e0:74:41)
W (117005) wifi:<MESH AP>adjust channel:11, secondary channel offset:2(40D)
I (117014) wifi:Total power save buffer number: 8
I (01:02:00.070) offline: object saved to PSRAM, free 522712
W (01:02:01.066) monitor: 3W3P:: rule activated
I (01:02:01.067) processor: send status... (7002ms)
W (121824) wifi:Haven't to connect to a suitable AP now!
W (01:02:02.067) monitor: 3W3P:: rule deactivated
I (01:02:08.066) stats: iteration_time_us=999992 Name=last% IDLE0=98 IDLE1=94 wifi=4 MNWK=1 
I (132600) wifi:mode : sta (7c:df:a1:e0:74:40)
I (01:02:11.844) mesh_main: <MESH_EVENT_NO_PARENT_FOUND>scan times:60, 3
```

zhangyanjiaoesp, 2023-11-08:
Do you mean that the device1 (2dB) can connect to hotspot, rssi is -63, and the device2 (14dB) can't scan any parent?

KonssnoK, 2023-11-08:
> Do you mean that the device1 (2dB) can connect to hotspot, rssi is -63, and the device2 (14dB) can't scan any parent?

exactly.

zhangyanjiaoesp, 2023-11-09:
I will try to test locally and get back to you.

zhangyanjiaoesp, 2023-11-09:
1. Do you use ESP32 ?
2. Can you reproduce this issue based on the mesh example? I have tested using the `internal_communication`, and set the ESP32 device tx power to 14dB, it still can scan APs and connect to router/parent.

KonssnoK, 2023-11-09:
> 1. Do you use ESP32 ?
> 
>     2. Can you reproduce this issue based on the mesh example? I have tested using the `internal_communication`, and set the ESP32 device tx power to 14dB, it still can scan APs and connect to router/parent.

i use one esp32s3 devkitC and one of our pcbs (with LTE) because i was investigating another issue.
i will try to use 2 devkitc

KonssnoK, 2023-11-13:
i saw there are new pushes on the wifi lib,
Specifically:
769d2384e47517667dbb2b47c89a005415a3dd10
a91050f77b744fca7d2ed5916c6165c5915d14ca

i think i will update and then retry.

zhangyanjiaoesp, 2023-11-13:
ok, waiting for your feedback.

KonssnoK, 2023-11-13:
for now the only strange thing that occurred is that both devices joined the access point as root and apparently dont see each other.
After how long a device should check for other parents and see if it has to move to a different layer?

zhangyanjiaoesp, 2023-11-14:
After the **non-root** node successfully connects to the parent node, it will scan the current channel from time to time, select a better parent node and switch. The interval between each scan is not fixed, but it is at least 4.5 seconds.

KonssnoK, 2023-11-14:
Uh, 4.5 seconds it's a very short time. 
I'm talking about minutes of parallel ROOT devices.
Is there a way to print more logs about this behavior?
I try to enable verbose logging for this component

zhangyanjiaoesp, 2023-11-14:
> I'm talking about minutes of parallel ROOT devices

I'm sorry, I didn't catch your point. Can you explain it in more detail?

KonssnoK, 2023-11-14:
yes, i ended up in a state where i had 2 devices, same mesh, connected both as ROOT.
I now enabled verbose logs on the mesh, i-ll try to reproduce.

Sadly with RSSI is always difficult to keep it stable :)

KonssnoK, 2023-11-14:
@zhangyanjiaoesp i'm attaching 2 logs

Configuration -> straight line with hotspot / layer2 / layer1

Files
- layer2_1 -> the device boots and it's unable to find a parent for 10 minutes. 
- layer2_2 -> the same device boots and it finds the root immediately

[layer2_2.txt](https://github.com/espressif/esp-idf/files/13346257/layer2_2.txt)
[layer2_1.txt](https://github.com/espressif/esp-idf/files/13346258/layer2_1.txt)

Still trying to prolong those 10 minutes to infinite

zhangyanjiaoesp, 2023-11-15:
@KonssnoK 
According to your logs, it seems that your rssi is too low(-90), and the mesh rssi threshold is <-72,-82,-85>:

you can call `esp_mesh_set_rssi_threshold()` to change the rssi threshold, or let the device closer to the router or other device.

KonssnoK, 2023-11-15:
@zhangyanjiaoesp
could you tell me how the settings influence the mesh algorithm?

KonssnoK, 2023-11-15:
also, i understand the device had -90dB to the Router, but it had only -78dB to the ROOT.
Why it didnt connect to the root?

zhangyanjiaoesp, 2023-11-16:
> also, i understand the device had -90dB to the Router, but it had only -78dB to the ROOT.

According to your logs, the rssi from the **device to the ROOT** is -84:

[image]

**And this rssi value is volatile, thus the scan sometimes can find root and sometimes can't**:

[image]

[image]

When the device is doing parent selection, and the parent is root, it will compare the scanned parent's rssi with the `select_rssi` and `backoff_rssi`, and decide whether to do rssi backoff.

[image]

Since the rssi from the device to the ROOT is about -85, suggest you can call `esp_mesh_set_switch_parent_paras()` API to set the  `select_rssi` and `backoff_rssi` to -85 or  lower.

KonssnoK, 2023-11-20:
I was able to see another strange behavior (that i have to try to replicate),
but here everything looks clear (more or less) now.

I will close the ticket.

closed 2023-11-20 by KonssnoK