espressif/esp-idf #12536: Mesh v1.1 - Unable to achieve Remote Provisioning

besbar, 2023-11-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-3775-gb4268c874a-dirty

### Espressif SoC revision.

ESP32 Chip Revision v3.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Adafruit HUZZAH32 – ESP32 Feather Board (3405)

### Power Supply used.

USB

### What is the expected behavior?

I am flashing 3 ESP32 with examples projets from master branch:
- rpr_client,
- rpr_server,
- unprov_dev

I expect the Remote Provisoning process to correctly end with 
the unprovisioned node device being remotely provisioned by rpr_server and corresponding information being reported from rpr_server to rpr_client.

### What is the actual behavior?

So far is correctly achieved:
- common provisioning (+ model binding) of the rpr_server by the rpr_client,
- button pressed on rpr_client => msg RPR_SCAN_GET is sent from rpr_client to rpr_server,
- rpr_server replies with a SCAN_STATUS msg,
- rpr_client sends then a SCAN_START msg,
- rpr_server replies with SCAN_STATUS, and a SCAN_REPORT msgs,
- rpr_client sends then a LINK_GET msg,
- rpr_server replies with a LINK_STATUS msg,
- rpr_client sends then a LINK_OPEN msg,
- rpr_server replies with a LINK_STATUS msg

From there, the rpr_server is expected to send a ESP_BLE_MESH_MODEL_OP_RPR_LINK_REPORT to rpr_client.
and then rpr_client should report this message, then send to rpr_server the remote provisioning start process message.
but this never happens. 

logs from unprov_dev indicated that the provisioning link is open, then process is timed-out.

### Steps to reproduce.

Executing the flow decribed in examples readme files.

### Debug Logs.

```plain
[... 524 lines cut ...]
I (973) BLE_MESH: Relay not supported
I (973) BLE_MESH: Friend not supported
I (1033) EXAMPLE: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
I (1043) EXAMPLE: BLE Mesh Node initialized
I (1053) EXAMPLE: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0
I (3043) main_task: Returned from app_main()
I (82073) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer PB-ADV
W (112263) BLE_MESH: Timeout, giving up transaction
I (112263) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_CLOSE_EVT, bearer PB-ADV
```

### More Information.

I also tried with ESP32-S3 from adafruit, result is identical.

It looks like ESP_BLE_MESH_MODEL_OP_RPR_LINK_REPORT is not sent to rpr_client.
Or at least rpr_client does not callback any RPR_LINK_REPORT from rpr_server, and consequently 
`/* Let remote provisioning server start provisioning */
err = esp_ble_mesh_rpr_client_action(ESP_BLE_MESH_RPR_CLIENT_ACT_START_RPR,
                                                &param);
if (err) {
    ESP_LOGE(TAG, "Failed to perform Remote Provisioning Client action: Start Prov");
}` is never executed.

I wonder also if it is not due to the bearer that is in use during remote provisioning process (PB-ADV,

forx157, 2023-11-10:
@besbar 
Thank you very much for pointing out this issue. We have now fixed the 
problem, and the correction will be synchronized with the GitHub 
repository shortly.

besbar, 2023-11-10:
Great. Thank you.

besbar, 2023-11-23:
@forx157, do you have an idea about when the correction will be synchronized?
Thanks a lot.

forx157, 2023-11-23:
@besbar 
I have attached the library for you. Please test it, and if the issue persists, share the logs with me.
[esp32_esp32s3.zip](https://github.com/espressif/esp-idf/files/13449796/esp32_esp32s3.zip)

besbar, 2023-11-24:
@forx157,
I have replaced files you sent in following folders:
esp-idf\components\bt\esp_ble_mesh\v1.1\lib\esp32 & esp-idf\components\bt\esp_ble_mesh\v1.1\lib\esp32s3.

Unfortunately, I get a build error when building either rpr_server or rpr_client (on ESP32 and ESP32S3)

> [1/7] Performing build step for 'bootloader'←[K
> -- Building ESP-IDF components for target esp32s3
> -- Project sdkconfig file C:/Users/neobe/Documents/ESP32/rpr_server/sdkconfig
> -- Compiler supported targets: xtensa-esp-elf
> -- Adding linker script C:/Users/neobe/esp_master/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
> -- Bootloader project name: "bootloader" version: 1
> -- Adding linker script C:/Users/neobe/esp_master/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
> -- Adding linker script C:/Users/neobe/esp_master/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
> -- Adding linker script C:/Users/neobe/esp_master/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
> -- Adding linker script C:/Users/neobe/esp_master/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
> -- Adding linker script C:/Users/neobe/esp_master/esp-idf/components/bootloader/subproject/main/ld/esp32s3/bootloader.ld
> -- Adding linker script C:/Users/neobe/esp_master/esp-idf/components/bootloader/subproject/main/ld/esp32s3/bootloader.rom.ld
> -- Components: bootloader bootloader_support efuse esp_app_format esp_bootloader_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
> -- Component paths: C:/Users/neobe/esp_master/esp-idf/components/bootloader C:/Users/neobe/esp_master/esp-idf/components/bootloader_support C:/Users/neobe/esp_master/esp-idf/components/efuse C:/Users/neobe/esp_master/esp-idf/components/esp_app_format C:/Users/neobe/esp_master/esp-idf/components/esp_bootloader_format C:/Users/neobe/esp_master/esp-idf/components/esp_common C:/Users/neobe/esp_master/esp-idf/components/esp_hw_support C:/Users/neobe/esp_master/esp-idf/components/esp_rom C:/Users/neobe/esp_master/esp-idf/components/esp_system C:/Users/neobe/esp_master/esp-idf/components/esptool_py C:/Users/neobe/esp_master/esp-idf/components/freertos C:/Users/neobe/esp_master/esp-idf/components/hal C:/Users/neobe/esp_master/esp-idf/components/log C:/Users/neobe/esp_master/esp-idf/components/bootloader/subproject/main C:/Users/neobe/esp_master/esp-idf/components/bootloader/subproject/components/micro-ecc C:/Users/neobe/esp_master/esp-idf/components/newlib C:/Users/neobe/esp_master/esp-idf/components/partition_table C:/Users/neobe/esp_master/esp-idf/components/soc C:/Users/neobe/esp_master/esp-idf/components/spi_flash C:/Users/neobe/esp_master/esp-idf/components/xtensa
> -- Configuring done
> -- Generating done
> -- Build files have been written to: C:/Users/neobe/Documents/ESP32/rpr_server/build/bootloader
> [0/1] Re-running CMake...
> [1/1] cmd.exe /C "cd /D C:\Users\neobe\Documents\ESP32\rpr_server\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe C:/Users/neobe/esp_master/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 C:/Users/neobe/Documents/ESP32/rpr_server/build/bootloader/bootloader.bin"
> Bootloader binary size 0x5240 bytes. 0x2dc0 bytes (36%) free.
> [3/5] Linking CXX executable rpr_server.elf←[K
> ←[31mFAILED: ←[0mrpr_server.elf
> cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=C:/Users/neobe/Documents/ESP32/rpr_server/build/rpr_server.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.newlib.ld -T esp32s3.rom.version.ld -T memory.ld -T sections.ld @CMakeFiles\rpr_server.elf.rsp -o rpr_server.elf  && cd ."
> C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:/Users/neobe/esp_master/esp-idf/components/bt/esp_ble_mesh/v1.1/lib/esp32s3/libmesh_v1.1.a(cm_data.c.o):(.literal+0x6c): undefined reference to `bt_mesh_ext_comp_get_elem_s'
> C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:/Users/neobe/esp_master/esp-idf/components/bt/esp_ble_mesh/v1.1/lib/esp32s3/libmesh_v1.1.a(cm_data.c.o): in function `bt_mesh_comp_128_register':
> (.text+0x6da): undefined reference to `bt_mesh_ext_comp_get_elem_s'
> C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:/Users/neobe/esp_master/esp-idf/components/bt/esp_ble_mesh/v1.1/lib/esp32s3/libmesh_v1.1.a(brc_srv.c.o):(.literal+0xb0): undefined reference to `bt_mesh_ext_rpl_get_src'
> C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:/Users/neobe/esp_master/esp-idf/components/bt/esp_ble_mesh/v1.1/lib/esp32s3/libmesh_v1.1.a(brc_srv.c.o):(.literal+0xb4): undefined reference to `bt_mesh_ext_update_rpl'
[... cut ...]
> C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: (.text+0xf33): undefined reference to `bt_mesh_ext_rpl_get_seq'
> C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: (.text+0xf4b): undefined reference to `bt_mesh_ext_update_rpl'
> collect2.exe: error: ld returned 1 exit status
> ninja: build stopped: subcommand failed.

forx157, 2023-11-24:
@besbar 
Is your ESP-IDF updated to the latest version?

forx157, 2023-11-25:
@besbar 
Please provide me with the commit information for the Mesh library.

besbar, 2023-11-27:
Ok, here are the logs.
Bluetooth Mesh v1.1 commit: 6d5649960d

**Unprov_dev**

> I (507) boot: Loaded app from partition at offset 0x10000
> I (507) boot: Disabling RNG early entropy source...
> I (519) cpu_start: Multicore app
> I (527) cpu_start: Pro cpu start user code
> I (527) cpu_start: cpu freq: 240000000 Hz
> I (527) cpu_start: Application information:
> I (530) cpu_start: Project name:     onoff_server
> I (536) cpu_start: App version:      1
> I (540) cpu_start: Compile time:     Nov 27 2023 13:05:21
> I (546) cpu_start: ELF file SHA256:  730055669...
> I (552) cpu_start: ESP-IDF:          v5.3-dev-422-ga7fbf452fa
> I (558) cpu_start: Min chip rev:     v0.0
> I (563) cpu_start: Max chip rev:     v3.99
> I (568) cpu_start: Chip rev:         v3.0
> I (573) heap_init: Initializing. RAM available for dynamic allocation:
> I (580) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
> I (586) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
> I (592) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
> I (598) heap_init: At 3FFCAF98 len 00015068 (84 KiB): DRAM
> I (604) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
> I (610) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
> I (617) heap_init: At 40098E38 len 000071C8 (28 KiB): IRAM
> I (625) spi_flash: detected chip: generic
> I (628) spi_flash: flash io: dio
> I (633) coexist: coex firmware version: b6d5e8c
> I (638) main_task: Started on CPU0
> I (648) main_task: Calling app_main()
> I (648) EXAMPLE: Initializing...
> I (648) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
> I (678) BTDM_INIT: BT controller compile version [6458728]
> I (678) BTDM_INIT: Bluetooth MAC: 94:b9:7e:6b:4f:32
> I (678) phy_init: phy_version 4780,16b31a7,Sep 22 2023,20:42:16
> I (988) dev_uuid: 55 55 94 b9 7e 6b 4f 32 00 00 00 00 00 00 00 00 
> Bluetooth Mesh v1.1 commit: [6d5649960d]
> I (988) BLE_MESH: Relay not supported
> I (998) BLE_MESH: Friend not supported
> I (1058) EXAMPLE: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
> I (1058) EXAMPLE: BLE Mesh Node initialized
> I (1078) EXAMPLE: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0
> I (3058) main_task: Returned from app_main()
> I (437668) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer PB-ADV
> W (467848) BLE_MESH: Timeout, giving up transaction
> I (467848) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_CLOSE_EVT, bearer PB-ADV

**RPR_server**

> I (29) boot: ESP-IDF v5.3-dev-422-ga7fbf452fa 2nd stage bootloader
> I (29) boot: compile time Nov 27 2023 14:32:28
> I (31) boot: Multicore bootloader
> I (35) boot: chip revision: v3.0
> I (39) boot.esp32: SPI Speed      : 40MHz
> I (44) boot.esp32: SPI Mode       : DIO
> I (48) boot.esp32: SPI Flash Size : 2MB
> I (53) boot: Enabling RNG early entropy source...
> I (58) boot: Partition Table:
> I (62) boot: ## Label            Usage          Type ST Offset   Length
> I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
> I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
> I (84) boot:  2 factory          factory app      00 00 00010000 00177000
> I (91) boot: End of partition table
> I (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=3a124h (237860) map
> I (186) esp_image: segment 1: paddr=0004a14c vaddr=3ffbdb60 size=053ach ( 21420) load
> I (194) esp_image: segment 2: paddr=0004f500 vaddr=40080000 size=00b18h (  2840) load
> I (195) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=c143ch (791612) map
> I (471) esp_image: segment 4: paddr=00111464 vaddr=40080b18 size=18320h ( 99104) load
[... 576 lines cut ...]
> W (24609) EXAMPLE: Remote Prov Client Recv RSP, opcode 0x805b, from 0x0005
> I (24609) EXAMPLE: link_status, status 0x00
> I (24619) EXAMPLE: link_status, rpr_state 0x00
> I (24619) BLE_MESH: Original timeout 4000ms, calculated timeout 2960ms
> I (24629) BLE_MESH: Client message 0x00008059 with timeout 4000ms
> I (24639) BLE_MESH: send, app_idx 0xfffe src 0x0001 dst 0x0005
> I (24639) BLE_MESH: send, len 18: 8059555594b97e6b4f320000000000000000
> I (24649) BLE_MESH: Seg, send_tag 0x01, send_szmic 0, aszmic 0
> I (24659) BLE_MESH: Use security credentials 0x00, proxy 0
> I (24659) BLE_MESH: Network PDU, count 2, interval 20
> I (24679) BLE_MESH: Use security credentials 0x00, proxy 0
> I (24679) BLE_MESH: Network PDU, count 2, interval 20
> W (24679) EXAMPLE: EVENT # 0
> W (24689) EXAMPLE: Remote Prov Client Send Comp, err_code 0
> I (24819) BLE_MESH: Seg 0/1 acked
> I (24819) BLE_MESH: Seg 1/1 acked
> I (24909) BLE_MESH: Attempts: 3
> I (25029) BLE_MESH: recv, app_idx 0xfffe src 0x0005 dst 0x0001
> I (25029) BLE_MESH: recv, len 4: 805b0001
> W (25029) EXAMPLE: EVENT # 2
> W (25029) EXAMPLE: Remote Prov Client Recv RSP, opcode 0x805b, from 0x0005
> I (25039) EXAMPLE: link_status, status 0x00
> I (25049) EXAMPLE: link_status, rpr_state 0x01
> I (25049) EXAMPLE: Remote Provisioning Server(addr: 0x0005) Recv Link Open Success
> W (25089) BLE_MESH: No outbound bearer found, inbound bearer 0

forx157, 2023-11-28:
Hi， @besbar 
Could you please try this library again?
[libs.zip](https://github.com/espressif/esp-idf/files/13483162/libs.zip)

besbar, 2023-12-01:
Hello @forx157 !

With new librairies files you send, the remote provisoning samples found on this repo finally works, either on esp32 and esp32s3. Well done.

But then I modify a little the code with some wifi and mqtt code, and other models addition (things I am used to when working with esp32, it works without any problem with ble mesh) and I am unable to achieve remote provisioning again. I tried many times, looking carefully at parameters etc... but I have no explanation.

Would you have an idea why the sample could work, but why my other code based on this sample could not be working? considering I only add wifi, mqtt and some ble mesh models?

Thanks a lot for your help.

besbar, 2023-12-05:
@forx157, 
Thank you for these infos.

Sorry to ask again, but would you have a template or piece of code that would allow me to use the Opcodes Aggregator models.
I have looked at the API to build myself, but I am struggling in forming correct message with the Opcode Aggregator client.
Any help will be very much appreciated.

Thanks a lot.

Alvin1Zhang, 2024-02-20:
Thanks for reporting, sorry for the slow turnaround, the fix is available https://github.com/espressif/esp-idf/commit/2d8fb1604f4006317ad479de6f732cc4ff9b694f, feel free to reopen with more updates, thanks.

closed 2024-02-20 by Alvin1Zhang