espressif/esp-idf #12537: ESP32 MAC not reacting to received signals

ThijsvanBovene, 2023-11-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

For a project I am using the lan8671 phy for SPE 10base-T1S in point to point mode but, I have some issues. I am using the ESP32 devkitc V4 with ESP-IDF V5.1.1.
The issue is when I am trying to ping another board I can see the rmii TX signals with the oscilloscope and can see them arrive on the other board so it looks the phy is working but the ESP doesn't react to it. I think I configure the mac the correct way following the example from the esp-idf website. the signals look like arp request is it maybey possible that there are more configurations needed for point to point?

    `
    static eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();
    mac_config.flags = ETH_MAC_FLAG_PIN_TO_CORE;
    static eth_esp32_emac_config_t esp32_emac_config = ETH_ESP32_EMAC_DEFAULT_CONFIG();
    esp32_emac_config.smi_mdc_gpio_num = 4;
    esp32_emac_config.smi_mdio_gpio_num = 5;
    esp32_emac_config.interface = EMAC_DATA_INTERFACE_RMII;
    esp32_emac_config.clock_config.rmii.clock_gpio = 0;
    esp32_emac_config.clock_config.rmii.clock_mode = EMAC_CLK_EXT_IN;

    mac->set_speed(mac, 0); //Configure mac speed to 10Mbit
    mac->set_duplex(mac, 0); //Configure mac to half duplex

    eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();
    phy_config.phy_addr = 0x00;
    phy_config.reset_gpio_num = 15;

    esp_eth_phy_t *phy = esp_eth_phy_new_lan867x(&phy_config);

    esp_eth_config_t config = ETH_DEFAULT_CONFIG(mac, phy);

    static esp_eth_handle_t eth_handle = NULL;

    esp_err_t err = esp_eth_driver_install(&config, &eth_handle);
    ESP_LOGI(TAG, "Ethernet Driver install: %s", esp_err_to_name(err));

    esp_event_loop_create_default();
    esp_netif_init();
    esp_netif_config_t cfg = ESP_NETIF_DEFAULT_ETH();
    esp_netif_t *eth_netif = esp_netif_new(&cfg);
    
    err = esp_netif_attach(eth_netif, esp_eth_new_netif_glue(eth_handle));
    ESP_LOGI(TAG, "esp_netif_attach: %s", esp_err_to_name(err));

    esp_netif_ip_info_t ip;
    const char* ip4 = "192.168.0.3";
    const char* gw = "0.0.0.0";
    const char* netmask = "255.255.255.0";
    esp_netif_dhcpc_stop(eth_netif);
    memset(&ip, 0, sizeof(esp_netif_ip_info_t));
    ipaddr_aton((const char *)ip4, &ip.ip.addr);
    ipaddr_aton((const char *)gw, &ip.gw);
    ipaddr_aton((const char *)netmask, &ip.netmask);
    esp_netif_set_ip_info(eth_netif, &ip);`

The event does trigger that it got an IP address.

Thanks in advance

kostaond, 2023-11-08:
What is the source of the REF RMII CLK for the PHY?

BTW: just a teaser, we are preparing a demo with 10base-T1S. It will be ready soon :)

ThijsvanBovene, 2023-11-08:
Oh nice. but the only difference is the phy right?

I am using a external 50mhz clock source (https://www.we-online.com/components/products/datasheet/831056249.pdf).
I also desolderd it and tried to clock it from the esp itself but no difference. the MDIO lines are working and I can write and read to the registers.

kostaond, 2023-11-08:
We use LAN8670 for the demo.

Regarding your issue,  what is you test setup? Two ESP32's with LAN8671 - one ESP as DHCP server and the other one as DHCP Client? If so, you need to start DHCP server in `ETHERNET_EVENT_CONNECTED` event. I don't see the DHCP start in your code snipped.

ThijsvanBovene, 2023-11-08:
I don't use DHCP. my test setup is indeed two esp's connected together one doing icmp and the other one programmed like above.

kostaond, 2023-11-08:
Try to increase debug level or change [this line](https://github.com/espressif/esp-idf/blob/master/components/esp_eth/src/esp_eth_mac_esp.c#L297) to see if your device receives Ethernet frames.

ThijsvanBovene, 2023-11-08:
If i change that line it doesn't print anything. I changed it to `ESP_LOGE(TAG, "receive len= %d", recv_len);`

ThijsvanBovene, 2023-11-08:
So i changed that line to `ESP_LOGI(TAG, "receive len= %d", recv_len);` but it didn't print anything so added `ESP_LOGI(TAG, "RX-task running");` to [here](https://github.com/espressif/esp-idf/blob/b90dfe075923870a8d1f8d85a75318781455f57a/components/esp_eth/src/esp_eth_mac_esp.c#L279) and it also doens't print that so does that mean the whole rx task isn't running?

kostaond, 2023-11-08:
The Rx task is blocked and it waits for interrupt form EMAC and it is released once the EMAC receives a data. It indicates you that no data is received. 

Unfortunately, I cannot help you without seeing your complete code. However, going over it would be duplication of work for me since we're working on the driver too. We're almost there, we're able to send/receive. We just need to polish it and prepare demo. Could you please wait a week?

ThijsvanBovene, 2023-11-08:
Okay thanks for the help. I can wait a week could you please inform me or something when it is ready and how i can test it out?

kostaond, 2023-11-08:
@ThijsvanBovene, I'll let you know once the driver is ready.

ThijsvanBovene, 2023-11-17:
@kostaond Hi, Just want to ask how it is going with the driver. is there maybey a beta version?

kostaond, 2023-11-20:
@ThijsvanBovene, please check https://github.com/bogdankolendovskyy/esp-eth-drivers/tree/lan867x/lan867x. However, keep in mind that it's work in progress.

adeepn, 2023-11-22:
> @ThijsvanBovene, please check https://github.com/bogdankolendovskyy/esp-eth-drivers/tree/lan867x/lan867x.
> 
> However, keep in mind that it's work in progress. If you have any questions or suggestions, they are very welcomed.

https://github.com/bogdankolendovskyy/esp-eth-drivers/pull/1

We already have devices on esp32 with lan8671 and can do any tests :)

ThijsvanBovene, 2023-11-23:
@adeepn what silicon revision do you have?

adeepn, 2023-11-23:
> @adeepn what silicon revision do you have?

B1

ThijsvanBovene, 2023-11-23:
So i used the server example for the lan867x but,  when i upload it the got ip event doesn't trigger. But it does say ethernet attached to netif.

kostaond, 2023-11-23:
> So i used the server example for the lan867x but, when i upload it the got ip event doesn't trigger. But it does say ethernet attached to netif.

You wouldn't see IP event on server... It has static IP and it is not reported. 

Once you connect the second device, the client, you should see:
```bash
I (402) esp_eth.netif.netif_glue: ethernet attached to netif
I (3102) ethernet_init: Ethernet(LAN867x[23,18]) Started
I (3102) ethernet_init: Ethernet(LAN867x[23,18]) Link Up
I (3102) ethernet_init: Ethernet(LAN867x[23,18]) HW Addr 08:3a:f2:31:20:f7
I (67852) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.1.2
I (67852) lan867x_example: Assigned ip 192.168.1.2 to a connected device.
```

ThijsvanBovene, 2023-11-23:
ohh oke I will try that thank you

ThijsvanBovene, 2023-11-24:
Hi,
So i have tried getting the examples to work but no success yet. I have a few questions.
when I run the server example I have added that it needs to blink a LED but, it doesn't do that so it doesn't come in the while loop. is that normal like is it waiting for a first client and than it continues to the while?
what K config settings are you using? I changed my mdio pins and it does come through the init so it looks like it is working but maybey there are other settings i need to change.
what version of ESP-IDF are you using? i am using 5.1.1

ThijsvanBovene, 2023-11-24:
IT WORKSSSS!!!! 
[image]

kostaond, 2023-11-24:
@ThijsvanBovene what was the issue?

ThijsvanBovene, 2023-11-24:
@kostaond I have 3 boards i think 1 is broken(it still doesn’t work). And i added a 100ohm resistor on both cable ends and uploaded and it started working. 

Although i still have a question about the driver: i see that with custom_crtl it is should be possible to set plca and other registers but those register acording to the datasheet need to be controlled via MMD. But i don’t see that in the software.

kostaond, 2023-11-27:
@hijsvanBovene very good point. I'll discuss with my colleague. We haven't tested PLCA yet because we also started with 3 boards and one is broken :D We're waiting for new one, it's slow process... Anyway, really appreciate your input!

kostaond, 2023-12-06:
@ThijsvanBovene, @adeepn, feel free to review - https://github.com/espressif/esp-eth-drivers/pull/19

kostaond, 2024-03-22:
The LAN867x is available at https://github.com/espressif/esp-eth-drivers

closed 2024-03-22 by kostaond