espressif/esp-idf #12538: Would be nice to have a WIFI_EVENT_CHANNEL_CHANGE

moefear85, 2023-11-07:
### Is your feature request related to a problem?

Espnow fails to send if the configured peer channel is different than the home channel. But the home channel can change at any moment if wifi is configured as STA or APSTA.

### Describe the solution you'd like.

It would be nice if there were a WIFI_EVENT_CHANNEL_CHANGE event, so that one can detect the channel change, and either modify peer infos, or ask the user what to do.

zhangyanjiaoesp, 2023-11-08:
Suppose that at the beginning, A and B establish ESPNOW communication on channel 1. Later, A may connect to the AP on channel 11, and the home channel change to channel 11. 
At this time, A post a `WIFI_EVENT_CHANNEL_CHANGE` event, the user may can change the peer's info of A, A can send ESPNOW data without channel error, but B is still on channel 1, so they still can't  communicating properly.

moefear85, 2023-11-08:
True, but there are cases where that isn't important, or A can otherhow signal to B that a channel change must happen, or both might be connected to same AP, but the apps want to do something special on channel change, such as blink an LED or write a log.

moefear85, 2023-11-08:
Also, I realize from the documentation, `esp_now_add_peer` takes a `wifi_interface_t` through `esp_now_peer_info_t`. If upon sending a packet from a non-matching interface, `esp_now_send` can return `ESP_ERR_ESPNOW_IF`. There is unfortunately no equivalent error code in case of channel mismatch. The printed output to the console is useful for a human reading the logs, but not for the app.

zhangyanjiaoesp, 2023-11-14:
we will consider to add the `channel change` event, and return channel error when call `esp_now_send()`

closed 2023-12-18 by espressif-bot