espressif/esp-idf #12541: Wrong order of includes in `esp-tls.h`

ivmarkov, 2023-11-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Reported error (in the ESP IDF support for Rust) here: https://github.com/esp-rs/esp-idf-svc/issues/312

Root cause: 
1. Observe the include order of `esp_tls_errors.h` inside `esp_tls.h`: [`esp_tls_errors.h` is included **before** the mbedtls/wolfssl headers](https://github.com/espressif/esp-idf/blob/master/components/esp-tls/esp_tls.h#L11)
2. Observe the content of `esp_tls_errors.h`: it uses mbedtls / wolfssl `#define`s[ **without** including the relevant mbedtls / wolfssl headers](https://github.com/espressif/esp-idf/blob/master/components/esp-tls/esp_tls_errors.h#L64). Note also that e.g. `MBEDTLS_ERR_SSL_WANT_READ` is **NOT** defined by `mbedtls/errors.h` but by `mbedtls/ssl.h`!

The above ^^^ leads to `ESP_TLS_ERR_SSL_WANT_READ` & friends defined to **nothing** as - e.g. by the time these macros are defined - `MBEDTLS_ERR_SSL_WANT_READ/WRITE/TIMEOUT` are not yet in scope.

... and since the `MBEDTLS_ERR_SSL_WANT_READ/WRITE/TIMEOUT` macros are only used in the `esp_tls_errors.h` header as values for other macros, the header even compiles, but is wrong.

Solution:
1. Option 1: Move `#include <esp_tls_errors.h>` [**after** including the mbedtls / wolfssl headers in `esp_tls.h`](https://github.com/espressif/esp-idf/blob/master/components/esp-tls/esp_tls.h#L25)
2. Option 2: In `esp_tls_errors.h`, include the relevant mbedtls / wolfssl headers that actually define these macros.

ivmarkov, 2023-11-07:
I've [implemented a workaround](https://github.com/esp-rs/esp-idf-sys/commit/2067b8cc518339694e936fdb405ecd15a5e33b1d) in our Rust `bindgen` code, but ideally we should fix the above root cause in ESP IDF as well.

hmalpani, 2023-11-21:
Hello @ivmarkov 
Thanks for reporting the issue. I have created an internal MR for fixing this.

closed 2023-12-06 by espressif-bot